Koenkk/zigbee2mqtt #367: Cannot pair new devices after removing others

glentakahashi, 2018-09-13:
I hit the 16 device limit, but even after successfully removing some devices from the network, I still cannot pair any other devices. I can re-add the same devices I just removed just fine, however new devices (xiaomi temperature sensors, switches, door sensors, etc.) cannot be paired. After re-flashing the entire stick I can re-add new devices again, however this was the only way I could get it to work.

This seems to be the exact same symptoms that @blakeblackshear was hitting in https://github.com/Koenkk/zigbee2mqtt/issues/308 as well

blakeblackshear, 2018-09-13:
That is correct. I experienced the same behavior. I reflashed, added some bulbs that serve as routers first, and then added my other devices successfully.

ryanbeaton, 2018-09-16:
@glentakahashi my understanding is that the device limit is hardware based (the memory of the cc2531) so reflashing as @blakeblackshear says is the only way

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

glentakahashi, 2018-09-16:
Is there no way to tell the cc2531 to purge those slots from its memory without doing a whole reflash?

Koenkk, 2018-09-16:
What kind of devices did you remove? 

It is a known issue that battery powered devices are not correctly removed due to sleeping.

glentakahashi, 2018-09-16:
Ah yea, they were definitely battery powered. Door sensors and buttons were what I removed.

ABeumer525, 2018-09-22:
Same issue here. Removing xiaomi battery driven sensors seems to work oke. But when trying to re-pair them they just don't show up. Whilst new device pop up almost instantly.

Is there another workaround possible other then reflashing the cc2531?

ABeumer525, 2018-09-22:
I'm sure of it now. I just added a 14th device to the cc2531 and now I get unsuccessful 17 error. So the cc2531 is full. With 2 to spare.

I reckon they are still paired, just not visible in the zigbee2mqtt config and database.

Is there anyway to do a remove failed node procedure? Because it would beat the alternative of reflashing the cc2531 (and repairing all those perfectly placed sensors).

123trs, 2018-09-23:
With the new branch today pairing works like a charm! Thank you wery much;-)

ciotlosm, 2018-09-27:
@123trs Can you elaborate for people reading this issue? Which branch are you talking about? What did you do in terms of steps to get it working?

basmeerman, 2018-09-30:
> I'm sure of it now. I just added a 14th device to the cc2531 and now I get unsuccessful 17 error. So the cc2531 is full. With 2 to spare.
>
> I reckon they are still paired, just not visible in the [...]

I have the same issue, I Think.

I have Xiaoi Aqara Temp/Hum/Pressure sensor that I removed through MQTT topic; then added again. It shows up in configuration.yaml / database.db but messages aren't received/processed through zigbee2mqtt.  When trying to re-pair the device zigbee sheppard says the device is allready known. 
Where should I start searching in the sourcecode to fix this?

basmeerman, 2018-10-01:
Thanks :-) Fixed it, see: https://github.com/Koenkk/zigbee2mqtt/issues/350#issuecomment-425838001 (for future reference).

closed 2018-12-22 by stale[bot]