Koenkk/zigbee2mqtt #369: IKEA TRADFRI LED1546G12 State problem

mredone, 2018-09-13:
Hello.
I have problem with state info in Home Assistant with this led bulb
Everything works almost fine.
I can turn on/off change brightness and color temp.
But. When i turn off lamp with this bulb Home assistant don't receive state of bulb and it remains on.
Is any method to get a real state of bulb.

I use code from Wiki to integrate it to HA

mredone, 2018-09-14:
Here log

https://pastebin.com/4K1sJeMY

mredone, 2018-09-17:
I have a big problem with debug log.
When i add `log_directory: /var/log/` bridge not start correctly.
When i delete it, everything work

Koenkk, 2018-09-17:
When you turn off the bulb using the power switch the bulbs doesn't sends it power state. 

The toggle doesn't switch in home assistant because zigbee2mqtt cannot control the bulb anymore.

mredone, 2018-09-17:
Ok. So maybe it must be change?
If zigbee2mqtt bridge lost connection to bulb, or other sensor, is any chance to send to mqtt status for this bulb/sensor `disconnected`

Koenkk, 2018-09-17:
I don't know if this is possible. Must look into it.

nismanoke, 2018-09-23:
I Would like to have this supported as well. I have physical switches in the bedrooms to turn off the ikea bulbs. Would be nice to have the status changed to unavailable or off in home assistant. Had this automated before with hue. When bulbs where unavailable for x min a automation turned off the bulb and group of those bulbs.

ciotlosm, 2018-09-27:
@Koenkk I know Tasmota implements telemetry on a state topic and HA can use that in the setup (https://www.home-assistant.io/components/switch.mqtt/#state_topic) to determine if a device becomes unavailable. Maybe something similar could be developed here as enhancement. I don't have full picture on the complete flow but I know i've seen this with my sonoff tasmota switches.

Koenkk, 2018-09-28:
@ciotlosm yes for sure! The problem for now is; how to determine if a device is online/offline?

closed 2018-12-22 by stale[bot]