Koenkk/zigbee2mqtt #370: Mi Cube (MFKZQ01LM) improvement.

niai, 2018-09-13:
Hi it would be nice if you could give what side is facing up when you rotate the Mi Cube (MFKZQ01LM), this way we can use the rotate to control different thing depending on the side.

Thanks for all the good work.

tb-killa, 2018-09-16:
Doesnt we Show the side already? I know that we show the side change.. Have to check.

AnhDuc85, 2018-09-16:
the following samples from MFKZQ01LM - Xiaomi Mi smart home cube (EndDevice)
payload: '{"action":"flip90","from_side":5,"to_side":1,"linkquality":21}'
payload: '{"action":"flip90","from_side":1,"to_side":3,"linkquality":92}'
payload: '{"action":"flip90","from_side":3,"to_side":4,"linkquality":99}'
payload: '{"action":"flip90","from_side":4,"to_side":2,"linkquality":94}'
payload: '{"action":"flip90","from_side":2,"to_side":1,"linkquality":97}'
payload: '{"action":"flip90","from_side":2,"to_side":3,"linkquality":94}'

niai, 2018-09-16:
All you get when you rotate is 

`{"battery":"100.00","voltage":3005,"linkquality":131,"action":"rotate_right","angle":31.87}`

I did look in to the "zigbee-shepherd-converters"  fromZigbee.js and it looks like the cube may not send the side when it rotates, so I am now not sure if it can be done.

tb-killa, 2018-09-16:
It´s because if you only rotate without flip side .. cube doesn´t trigger ``MFKZQ01LM_action_multistate`` event ..  only if you flip side we got the event and know the new side .. its from the cube themselve (firmware) so we actually doesn´t could change them .. maybe someone got an idea ... ?

sebastianheierhoff, 2018-09-18:
Could we just save the last side the cube was flipped to (from payload: '{"action":"flip90","from_side":2,"to_side":3,"linkquality":94}') and add it to {"battery":"100.00","voltage":3005,"linkquality":131,"action":"rotate_right","angle":31.87}?
The only problem I see is, that the side would be "null" if the cube has never been flipped after initialisation.

niai, 2018-09-18:
Aye if you are going to implement a dirty hack like that, it should probably be done in your own automation. It wouldn't be to hard to do in Node Red with retain message on.

From what i have looked in to and what has been said here, it looks like it's not really a feasible fix. I think this can be closed I will leave it up to the devs just in case they want more input. Thanks.

sebastianheierhoff, 2018-09-18:
What about using state caching (https://github.com/Koenkk/zigbee2mqtt/commit/9396bde1f3b022e0f634487d1a37d2a5127c8cb3#diff-f68567477d803b49930337bf7fe1556bR16)?

ciotlosm, 2018-09-27:
I would recommend managing as little state as possible in zigbee2mqtt. If you want state create a custom component in the "brains" which is HA to save all the state you need. 

Considering the original issue message I think this topic should be closed and if you really want to discuss further regarding saving state in zigbee2mqtt raise a new issue specific to this.

closed 2018-12-30 by stale[bot]