Koenkk/zigbee2mqtt #372: Problem with 0x84182600000ec07e): AB3257001NJ - OSRAM Smart+ plug (Router)

JLeandroFreitas, 2018-09-14:
Hey 
I cant use the device AB3257001NJ - OSRAM Smart+ plug (Router) as a router. Its not letting me add more devices i have 2 pcs OSRAM Smart+ plug
I tried to reset it by hold it down on the power button 5 sec until it click 5 times when it was connect to power and also when it was not connected to power source.
Please help me out i need to use this device as a router. 
Logs in the link

https://pastebin.com/iffi2hpq

Unsupported device
First check: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices

Bug report
- First make sure your on the latest version of zigbee2mqtt.
- Make sure you are running the latest CC253X firmware from https://github.com/koenkk/z-stack-firmware.
- Provide a clear description of the problem.
- If applicable, provide steps how to reproduce the problem.
- Provide a log with `log_level` set to `debug`, see https://github.com/Koenkk/zigbee2mqtt/wiki/Configuration
- DON'T copy logs directly here, post a link to https://hastebin.com/ or https://pastebin.com/.

lolorc, 2018-09-16:
Hi,
I'm using 4 of them. I'm not sure I can pair through them but it looks like they do their job as router.

prise_* are OSRAM Plugs, "manufacturerName":"OSRAM","modelId":"Plug 01"

[image]

sonicnkt, 2018-09-18:
@lolorc thanks for the small tutorial. There is a dev version for docker so i might try this soon to see what it reports back. 

Im also wondering what is the "official" way to using routers. i read on the hass forums that you have to pair you devices with the router instead of the coordinator but other say this is not necessary (like you).
As far as my tests go i can also confirm that pairing with the routers does not work and i tried 3 different brands.

lolorc, 2018-09-18:
I haven't been able to pair near this kind of router neither. I might be a software limitation, I don't know.

JLeandroFreitas, 2018-09-19:
It looks it works for me when i have now connected 3 plug 💯

sonicnkt, 2018-09-19:
hmm maybe i have to get more of those as well...
will try to map my network first and see what this reports back. 

its really strange that 3 my coverage with 3 routers + coordinator is no different than with only the coordinator tho.

JLeandroFreitas, 2018-09-19:
If want to expend with an cc2531 router do i only need connect it to usb power adapter or hub or do i need connect it to a linux device ?

lolorc, 2018-09-19:
@JLeandroFreitas you just need to power it, it doesn't require a computer.

closed 2018-10-03 by JLeandroFreitas

Alestrix, 2019-02-23:
I know this is an old topic, but couldn't find a newer one that discusses this.
I too am not able to use the Osram smart plug as a router. It seemed to have worked once (seen on the network map) but no message from the end device arrived at the coordinator.
The map also shows "offline" at the router/plug but it can be controlled though ZigBee and manual switching at the plug immediately shows in MQTT.

Here's the current graph, nothing connected to router:

digraph G {
node[shape=record];
  "0x00124b0014d9e265" [style="bold", label="{0x00124b0014d9e265|Coordinator|No model information available|online}"];
  "0x00124b0014d9e265" -> "0x84182600000eb19d" [label="40"]
  "0x00158d0002923513" [style="rounded, dashed", label="{window.EG.toilette|EndDevice|Xiaomi MiJia door & window contact sensor (MCCGQ01LM)|online}"];
  "0x00158d0002923513" -> "0x00124b0014d9e265" [label="31"]
  "0x00158d0002b35249" [style="rounded, dashed", label="{0x00158d0002b35249|EndDevice|Xiaomi MiJia door & window contact sensor (MCCGQ01LM)|online}"];
  "0x00158d0002b35249" -> "0x00124b0014d9e265" [label="1"]
  "0x00158d00023f1905" [style="rounded, dashed", label="{0x00158d00023f1905|EndDevice|Xiaomi MiJia door & window contact sensor (MCCGQ01LM)|online}"];
  "0x00158d00023f1905" -> "0x00124b0014d9e265" [label="1"]
  "0x00158d0002a1455a" [style="rounded, dashed", label="{0x00158d0002a1455a|EndDevice|Xiaomi MiJia door & window contact sensor (MCCGQ01LM)|online}"];
  "0x00158d0002a1455a" -> "0x00124b0014d9e265" [label="1"]
  "0x84182600000eb19d" [style="rounded", label="{plug|Router|OSRAM Smart+ plug (AB3257001NJ)|offline}"];
  "0x84182600000eb19d" -> "0x00124b0014d9e265" [label="25"]
  "0x00158d0002ec2ff5" [style="rounded, dashed", label="{motion|EndDevice|Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|online}"];
  "0x00158d0002ec2ff5" -> "0x00124b0014d9e265" [label="33"]
}

I use http://www.webgraphviz.com to visualize.