Koenkk/zigbee2mqtt #373: Issues in latest dev branch

AnhDuc85, 2018-09-14:
Hi,
just pulled the latest dev image and it seems there are some issues.

Some examples:
- Xiaomi Temp Sensor (should report battery and voltage)
-  {"battery":"NaN","voltage":"!","linkquality":57,"temperature":21.76,"humidity":67.92}

- Xiaomi Zigbee plug (should be ON & reporting consumption 
- values){"state":"OFF","power":null,"voltage":null,"consumption":null,"temperature":null,"linkquality":102}

When I switch back to the latest images, these sensors reporting normal values.

tb-killa, 2018-09-16:
Please provide full log via debug and do some samples.
I think i know the reason But need some input debug stuff first..

tb-killa, 2018-09-16:
``Recieved zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"65281":"\u0001!�\u000b\u0004!�\u0013\u0005!\u0006\u0000\u0006$\u0000\u0000\u0000\u0000\u0000d)�\be!�\u0015\n!\u0000\u0000"}}' of device 'lumi.sens' (0x00158d0001c2cf7e)``

The structure isnt parse correct.

Koenkk, 2018-09-16:
This is due to https://github.com/Koenkk/zigbee-shepherd-converters/pull/63

@evildad @lolorc did you ever experience this?

evildad, 2018-09-16:
Just grepped in my logs for the pattern voltage":"!" and found zero entries.
So the answer is no, never had that issue...

azuwis, 2018-09-17:
I have the same problem on WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor, debug log:

```
9/17/2018, 4:02:35 PM - debug: Recieved zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"65281":"\u0001!�\u000b\u0004!�\u0013\u0005!\u001c\u0000\u0006$\u0002\u0000\u0000\u0000\u0000d)�\ne!�\u001e\n!\u0000\u0000"}}' of device 'lumi.sens' (0x00158d0001e81c40)
9/17/2018, 4:02:35 PM - info: MQTT publish, topic: 'zigbee2mqtt/ht_bedroom', payload: '{"battery":"NaN","voltage":"!","linkquality":26,"temperature":26.98,"humidity":82.43}'
9/17/2018, 4:02:35 PM - debug: Recieved zigbee message of type 'devChange' with data '{"cid":"genBasic","data":{"65281":"\u0001!�\u000b\u0004!�\u0013\u0005!\u001c\u0000\u0006$\u0002\u0000\u0000\u0000\u0000d)�\ne!�\u001e\n!\u0000\u0000"}}' of device 'lumi.sens' (0x00158d0001e81c40)
```

Zigbee2mqtt git revision 6ede5a3

azuwis, 2018-09-17:
I can confirm that the problem does not happen on branch issue_373.

tb-killa, 2018-09-17:
We should check if this is correct??
``
var deviceId = zclId.device(260, 10).key;
``
Maybe there comes faulty result.?

dzungpv, 2018-09-18:
@tb-killa You are right, this PR will solve the problem https://github.com/Koenkk/zcl-packet/pull/6

hdo, 2018-09-18:
I can confirm that battery values are not read correctly with 'latest-dev':

`'zigbee2mqtt/0x00158d00022c9005', payload: '{"temperature":23.21,"linkquality":92,"humidity":46.08,"pressure":1002,"battery":"NaN","voltage":"!"}'
'zigbee2mqtt/0x00158d0001f535eb', payload: '{"temperature":25.1,"linkquality":21,"humidity":45.19,"pressure":1002,"battery":"NaN","voltage":"!"}'
'zigbee2mqtt/0x00158d0001e50c7c', payload: '{"illuminance":110,"linkquality":13,"occupancy":false,"battery":"NaN","voltage":"!"}'
`

Koenkk, 2018-09-18:
Can you check if this has been fixed in the latest dev branch?

hdo, 2018-09-18:
> Can you check if this has been fixed in the latest dev branch?

Running Docker latest-dev (commit #8376b7c). Still issues with battery reading but
some values seems to be correct. Maybe it's an issue with retained values.

I'll keep an eye on that.

`'{"temperature":13.38,"linkquality":0,"humidity":38.24,"pressure":1002,"battery":"NaN","voltage":"!"}'
'{"temperature":13.38,"linkquality":0,"humidity":32.16,"pressure":1002,"battery":"NaN","voltage":"!"}'
'{"temperature":27.55,"linkquality":63,"humidity":35.98,"pressure":1002,"battery":"98.33","voltage":2995}'
'{"temperature":27.6,"linkquality":63,"humidity":39.43,"pressure":1002.37,"battery":"98.33","voltage":2995}'
`

AnhDuc85, 2018-09-19:
can also confirm, will close the issue

lolorc, 2018-10-01:
@Koenkk yes I experiencing this issue at the moment. :-) probably because you reverted @dzungpv PR.

Koenkk, 2018-10-05:
Fixed again!

closed 2018-10-05 by Koenkk