Koenkk/zigbee2mqtt #375: trying to pair Philips LivingColors light Remote

briardalexis, 2018-09-15:
Hi,
I'm trying to pair a Philips livingColors remote, but I can't get it to join the network. I've got nothing showing in logs even in debug mode. Has anyone managed to pair it?

tb-killa, 2018-09-18:
Maybe this could help ?? https://www.youtube.com/watch?v=8611o3TKVeE
Follow the process of the remote only, the other ones like the hub actions .. leave out...
please provide content of ``database.db`` after you pair the device :)

briardalexis, 2018-09-18:
I saw that video but my remote does not pair when I reset it (I've got the same behavior as his first try). I'll work on that this weekend, I hope I'll find the solution.

tb-killa, 2018-10-04:
Hi i buy one of this remotes and hopefully got more results on debugging site to get them maybe implemented.
from what i read so far it seems to be a light link device who need the philips master key for zigbee data.

We have to see ...

tb-killa, 2018-10-09:
Now here the actually situation:

tb-killa, 2018-10-09:
So i´v done some tests : 
I'm afraid the remote can't be supported since it won't join ZLL or ZHA networks, it uses a proprietary touchlink key only known by Philips.

Actually i don´t know how to sniff them or use them in combination with zigbee-shepherd to hopefully join zha network.

closed 2018-12-20 by stale[bot]

uweng, 2020-03-13:
There are two Philips LivingColors Iris:

Philips LivingColors Iris 7199960PH, that may be supported
Philips LivingColors Iris 7099960PH, this is NOT supported
I've got the second one with no success to link to CC2531 and zigbee2mqtt.
Hardware linking does not work and "touch link" as well. I got "Failed to factory reset device through Touchlink".
Maybe you have the same lamp?!

IVI053, 2020-03-25:
Im also trying a 7099960PH with no success. But the docs state it is supported. Have no idea.

dedors, 2020-04-30:
Did anyone got the Philips Livingcolor Gen2 or Gen3 remote to work?
This is the remaining thing that is left so I can migrate all my philip lights to z2m. They remain at the moment still with the Hue bridge while everything else is z2m already.

As soon as I long press the (I) button on the remote near a light, the light just leaves the z2m network. Also a remote reset does not help here.
I'm on channel 25 which should be a ZLL channel, but I think that only matters for Gen3?

hurenkam, 2022-06-13:
I know I'm late to respond to this issue, but I do have a tip.
The hue Dimmer Switch V1 supports touchlinking in standalone mode, and can join through association. You may be able to first add such a hue dimmer switch to your zigbee2mqtt network, and subsequently use that to touchlink your living colors remote to have it join the network. The hue Dimmer Switch V1 contains both a ZLL touchlink key as well as the propriatary touchlink key used for Living Colors products.

AfromD, 2025-03-03:
I've tried adding the LC V3 remote with the Hue dimmer as an intermediate, but it does not add it to the zigbee network.