Koenkk/zigbee2mqtt #377: Help wanted choosing and flashing a sniffer device with external antenna

GoSpursGoNL, 2018-09-17:
Hi all,

I have Zigbee2MQTT on a RPi working perfectly with a CC2531, but unfortunately the range is insufficient for my house, in contrast to the Xiaomi gateway (that I'm trying to replace). So I'm looking for a replacement with an external antenna.

I have read the Supported sniffer devices page. https://github.com/Koenkk/zigbee2mqtt/wiki/Supported-sniffer-devices

I wonder what I should choose.
[This board](https://nl.aliexpress.com/item/CC2530-Zigbee-UART-Wireless-Core-Board-Development-Board-CC2530F256-Serial-Port-Wireless-Module-24MHz/32767470382.html)
Or this [usb-stick](https://nl.aliexpress.com/item/RF-TO-USB-CC2530-CC2591-RF-switch-USB-transparent-serial-data-transmission-equipment/1996354384.html)

I have the following questions:
- which one would you recommend?
- if I would choose the usb stick, do I still need to open the casing and connect a debugger? If that is still needed, then perhaps the board is better since it is cheaper?

- if I would choose the board, how do I flash it ?
I have read this page but it is a bit too technical for me: http://ptvo.info/how-to-select-and-flash-cc2530-144/
I don't understand the diagram under the 'How to connect CC2530 to the debugger' header, the image has a low resolution and also it seems that in the image more dupont cables are connected than what is shown in the diagram above.
I do have a different debugger by the way, I have the one [shown here](https://github.com/Koenkk/zigbee2mqtt/wiki/Getting-started).

- if I would choose the board, how do I connect it to the Pi (after flashing)?
I already have [this ftdi board](https://nl.aliexpress.com/item/1pcs-FT232RL-FTDI-USB-3-3V-5-5V-to-TTL-Serial-Adapter-Module-for-Arduino-Mini/32580666555.html?spm=a2g0s.9042311.0.0.27424c4dN6S9pt), if relevant:
Or would [such a cable](https://nl.aliexpress.com/item/6Pin-FTDI-FT232RL-FT232-Module-For-Arduino-USB-to-TTL-UART-Serial-Wire-Adapter-RS232-Download/32847308062.html?spm=2114.48010108.06001.35.2b7b7ea3hVfTFN&gps-id=5347592&scm=1007.13562.109986.0&scm_id=1007.13562.109986.0&scm-url=1007.13562.109986.0&pvid=8217a3af-5e10-4fd7-bd3e-940ef995c1df) be useful?

Thanks for your help!

ptvoinfo, 2018-09-17:
Hi,

1. The first one.
2. I can make a bigger image :)
3. You need to connect this board directly to RPi UART (Rx, Tx, GND). The board and RPi use the same voltage level.

GoSpursGoNL, 2018-09-17:
Thanks, I would really appreciate a clear picture and description how to connect that board to my debugger.

Do you happen to know how many directly connected devices are supported by the board? 

In case even the antenna can't provide full coverage (I hope so, since the xiaomi gateway at the same position also can), isn't it the case with Zigbee that powered devices like wall outlets and led lamps can also help to increase the range?

ben423423n32j14e, 2018-09-17:
Zigbee is a mesh network... Add routers to extend the coverage....

CC2531 should be fine.

GoSpursGoNL, 2018-09-20:
@ptvoinfo I have read the page again and it is becoming clearer already. I am now thinking to buy the 'CC2530 + CC2591' instead, so the board with the RF frontend. It seems though that on [this board](https://www.dx.com/p/webee-ti-cc2530-cc2591-zigbee-wireless-module-w-antenna-429601), the pins are not labeled on the pcb. Do you know if there is a technical spec somewhere that describes what each pin means, or is it identical to the board without the RF frontend?

And when connecting the board to the GPIO pins of the Pi, will the device also get listed in /dev/etc, like 'port: /dev/tty.usbmodem1411', since such a thing needs to be entered in Zigbee2MQTT's configuration file.

@ben423423n32j14e thanks, will any random powered led-lamp be able to function as a router for extended coverage, or does it need to be such a cc2530 board?

ben423423n32j14e, 2018-09-20:
Just buy the parts listed here, no cc2530 or cc2591 is needed.

https://github.com/Koenkk/zigbee2mqtt/wiki/Getting-started

Powered Zigbed nodes are almost always routers (repeaters).

GoSpursGoNL, 2018-09-20:
Thanks. I already ave those parts, but the range of the 2531 is insufficient. My first preference would be the 2530 as currently I have no need for any powered Zigbee devices. My led lights (non-zigbee) at strategic places (for range) are just 1 year old.

ben423423n32j14e, 2018-09-20:
What if your signal issue is limited by the transmit power of the sensors? How will adding a more powerful controller solve that? Antenna might help a bit but I think adding routers closer to the sensors is the best idea.

GoSpursGoNL, 2018-09-20:
The Xiaomi Gateway can receive all signals without any issues, so I am looking for something of similar quality.

GoSpursGoNL, 2018-10-14:
@ptvoinfo 
>     3. You need to connect this board directly to RPi UART (Rx, Tx, GND). The board and RPi use the same voltage level.

I now bought the 2530+2591 and flashed it successfully. When connecting to the Pi, is there no need to also connect 3.3v (Pi) to VCC (board)?

ptvoinfo, 2018-10-15:
@GoSpursGoNL You may connect 3.3V (Pi) to VCC on your board if VCC is 3.3V. Pi may give enough power. In my case, the board has a pin for 5V. Therefore I've connected it to power supply directly.

GoSpursGoNL, 2018-10-18:
I flashed the CC2530 + CC2591 board successfully and it is working in combination with Zigbee2MQTT. Nevertheless it is really disappointing that it still can't communicate with 2 of my window sensors whereas the Xiaomi Gateway directly next to the board has no problem with that.

* firmware
https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/CC2530_CC2591/bin/CC2530_CC2591ZNP-Prod.hex

GoSpursGoNL, 2018-11-06:
I've gone back to my Gateway (unfortunately) due to better range.

closed 2018-11-06 by GoSpursGoNL