Koenkk/zigbee2mqtt #378: Nothing works after reflashing CC2531

kaelsaes, 2018-09-17:
After reflashing my CC2531 with the latest firmware no Xiaomi device is reachable.

Do I have to install the newest zigbee2mqtt software also or is there something else I have to do?

Thx

ryanbeaton, 2018-09-17:
Reflashing clears the memory on the CC2531 so you have to repair them (pair them again)

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

kaelsaes, 2018-09-17:
Ah okay. Thx.

WagnerBeccard, 2018-09-17:
While youre repair them - can you leave the configuration untoutched and keep therefore the Friendly names?
Thinking of it. Didnt had to do this last time I upgraded the fw. Should be a reflash imo.

lolorc, 2018-09-17:
it should be ok to keep the config untouched, that's what I did last time I upgraded the firmware.

kaelsaes, 2018-09-20:
I didn't had to repair. After a few hours all device reappeared.

closed 2018-09-20 by kaelsaes