Koenkk/zigbee2mqtt #380: Subscribe to console log MQTT channel?

ben423423n32j14e, 2018-09-17:
Can we please have a MQTT channel that combines all device and log updates into one?

I'm wanting to add a health check section to the node-red Zigbee2MQTT Admin Panel that identifies devices that haven't been seen for more than 6 hours, devices with low battery or signal.

https://github.com/ben423423n32j14e/zigbee2mqttadminpanel

To be able to do this, I need a single MQTT channel with all devices reporting into it.

ben423423n32j14e, 2018-09-20:
A solution has been found, this is no longer needed. Closing issue.

ryanbeaton, 2018-09-21:
How did you do it? I don’t see any update to your code. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

ben423423n32j14e, 2018-09-21:
Still need a single topic unfortunately to be able to create the health widget. I worked out how to dynamically specify the topic in Node-Red sending to Zigbee2MQTT which allowed creation of the single Hue flow.

https://www.domoticz.com/forum/viewtopic.php?f=68&t=23607&sid=487dc0de08d7c55fe789074aa1f890b0&start=60#p192087

clockbrain, 2018-11-20:
@ben423423n32j14e maybe I'm missing something but is there any reason you can't just subscribe to a global wildcard topic that spans all topics.

Not sure about Node-Red but mosquitto_sub shows everything:

```
mosquitto_sub -t "zigbee2mqtt/#" -v -h 192.168.0.n -u mos
```

closed 2019-01-26 by stale[bot]