Koenkk/zigbee2mqtt #382: New device support for Aqare Vibration sensor

michel72, 2018-09-18:
I have a new unsupported device and since I run zigbee2mqtt as a hass.io plugin I dont think I have the luxury to change the required files.

The sensor is found as:

`  zigbee2mqtt:warn 2018-9-18 11:41:51 Device with modelID 'lumi.vibration.aq1' is not supported.`

Review: https://giztechreview.com/2018/08/19/aqara-smart-motion-sensor-review-addition-to-a-smart-home-from-xiaomi/'

What can I do to help to get this sensor supported?

tb-killa, 2018-09-18:
Duplication for this: https://github.com/Koenkk/zigbee2mqtt/issues/295
Please close this ones and view the other mentioned one.

michel72, 2018-09-18:
Looked for it, didn't find it. Sorry!

closed 2018-09-18 by michel72