Koenkk/zigbee2mqtt #384: Heiman SMOK_YDLV10 smoke detector

arteck, 2018-09-19:
new Device but...

i added  this
```
 {
        zigbeeModel: ['SMOK_YDLV10'],
        model: 'todo',
        vendor: 'Heiman',
        description: 'Heiman smoke detector',
        supports: 'smoke',
        fromZigbee: [],
        toZigbee: [],
    },    
```
int devices.js

in the log i can see 
`2018-9-19 16:53:23 - info: unknown (0x005043c903351248): todo - Heiman Heiman smoke detector (EndDevice)`

here ok

but thas all messages

> 2018-9-19 16:57:59 - info: New device with address 0x005043c903351248 connected!
> 2018-9-19 16:57:59 - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"device_connected","message":"0x005043c903351248"}'

arteck, 2018-09-19:
`{"id":42,"type":"EndDevice","ieeeAddr":"0x005043c903351248","nwkAddr":12369,"manufId":48042,"manufName":"Heiman","powerSource":"Battery","modelId":"SMOK_YDLV10","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,9,1280,1282],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":3,"stackVersion":2,"hwVersion":1,"manufacturerName":"Heiman","modelId":"SMOK_YDLV10","dateCode":"20150330","powerSource":3,"locationDesc":"","physicalEnv":0,"deviceEnabled":1,"alarmMask":0}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genAlarms":{"dir":{"value":1},"attrs":{"alarmCount":0}},"genOta":{"dir":{"value":2},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{"zoneState":0,"zoneType":40,"zoneStatus":32,"iasCieAddr":"0x0000000000000000","zoneId":0,"numZoneSensitivityLevelsSupported":2,"currentZoneSensitivityLevel":0}},"ssIasWd":{"dir":{"value":1},"attrs":{"maxDuration":240}}}}},"_id":"qmAugWFejpvKTRXz"}`

arteck, 2018-09-19:
or here
```
{"id":42,"type":"EndDevice","ieeeAddr":"0x005043c903351248","nwkAddr":12369,"manufId":48042,"manufName":"Heiman","powerSource":"Battery","modelId":"SMOK_YDLV10","epList":[1],"status":"online","joinTime":1537369046,"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,9,1280,1282],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":3,"stackVersion":2,"hwVersion":1,"manufacturerName":"Heiman","modelId":"SMOK_YDLV10","dateCode":"20150330","powerSource":3,"locationDesc":"","physicalEnv":0,"deviceEnabled":1,"alarmMask":0}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genAlarms":{"dir":{"value":1},"attrs":{"alarmCount":0}},"genOta":{"dir":{"value":2},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{"zoneState":0,"zoneType":40,"zoneStatus":32,"iasCieAddr":"0x0000000000000000","zoneId":0,"numZoneSensitivityLevelsSupported":2,"currentZoneSensitivityLevel":0}},"ssIasWd":{"dir":{"value":1},"attrs":{"maxDuration":240}}}}},"_id":"qmAugWFejpvKTRXz"}
{"id":42,"type":"EndDevice","ieeeAddr":"0x005043c903351248","nwkAddr":12369,"manufId":48042,"manufName":"Heiman","powerSource":"Battery","modelId":"SMOK_YDLV10","epList":[1],"status":"online","joinTime":1537369046,"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,9,1280,1282],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":3,"stackVersion":2,"hwVersion":1,"manufacturerName":"Heiman","modelId":"SMOK_YDLV10","dateCode":"20150330","powerSource":3,"locationDesc":"","physicalEnv":0,"deviceEnabled":1,"alarmMask":0}},"genPowerCfg":{"dir":{"value":1},"attrs":{"mainsVoltage":0,"mainsFrequency":0,"mainsAlarmMask":0,"mainsVoltMinThres":0,"mainsVoltMaxThres":65535,"mainsVoltageDwellTripPoint":0,"batteryVoltage":31,"batteryPercentageRemaining":200,"batteryManufacturer":"","batterySize":255,"batteryAHrRating":0,"batteryQuantity":0,"batteryRatedVoltage":30,"batteryAlarmMask":1,"batteryVoltMinThres":25,"batteryVoltThres1":0,"batteryVoltThres2":0,"batteryVoltThres3":0,"batteryPercentMinThres":40,"batteryPercentThres1":0,"batteryPercentThres2":0,"batteryPercentThres3":0,"batteryAlarmState":0}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genAlarms":{"dir":{"value":1},"attrs":{"alarmCount":0}},"genOta":{"dir":{"value":2},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{"zoneState":0,"zoneType":40,"zoneStatus":32,"iasCieAddr":"0x0000000000000000","zoneId":0,"numZoneSensitivityLevelsSupported":2,"currentZoneSensitivityLevel":0}},"ssIasWd":{"dir":{"value":1},"attrs":{"maxDuration":240}}}}},"_id":"qmAugWFejpvKTRXz"}
```

tb-killa, 2018-09-19:
Thanks for the Infos.
Is this device powered by battery ??

please test this ones 
```
{ 
    zigbeeModel: ['SMOK_YDLV10'], 
    model: 'Heiman SMOK_YDLV10 smoke detector',
    vendor: 'Heiman', 
    description: '',
    supports: '', 
    fromZigbee: [],
    toZigbee: [],
    configure: (ieeeAddr, shepherd, coordinator, callback) => {
         const device = shepherd.find(ieeeAddr, 1);
         const actions = [
                (cb) => device.bind('genPowerCfg', coordinator, cb),
                (cb) => device.report('genPowerCfg', 'batteryPercentageRemaining', 0, 1000, 0, cb),
                (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                (cb) => device.report('ssIasZone', 'zoneStatus', 0, 30, null, cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 1, zoneid: 1}, cb),
            ];
             execute(device, actions, callback);

        }, 
},
```

please provide log via ``DEBUG=zigbee-shepherd* npm start``

arteck, 2018-09-20:
```
zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +1ms
  zigbee2mqtt:debug 2018-9-20 08:48:06 Recieved zigbee message with data {"cid":"genPowerCfg","data":{"batteryPercentageRemaining":186}}
  zigbee2mqtt:warn 2018-9-20 08:48:06 No converter available for 'SMOK_YDLV10 smoke detector' with cid 'genPowerCfg', type 'attReport' and data '{"cid":"genPowerCfg","data":{"batteryPercentageRemaining":186}}'
  zigbee2mqtt:warn 2018-9-20 08:48:06 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
```

tb-killa, 2018-09-20:
Great thanks.
please provide full log because i need the converter states from the ``ssIasZone`` too.
Does if you trigger alarm something return as value ??

for ``batteryPercentageRemaining`` we could use 
``
fromZigbee: [fz.generic_battery],
``
for converter-type.

tb-killa, 2018-09-20:
``fz._324131092621_power]`` seems to be an older state of converter ... doesn´t the actually state named like my post before ? 

does this three 
``
(cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
(cb) => device.report('ssIasZone', 'zoneStatus', 0, 30, null, cb),
(cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 1, zoneid: 1}, cb),
``
return on logfile with ``ok`` and finally with `` Succesfully configured XXXXX`` ??

Because if this three states doesnt return succesfully you couldn´t get any messages by alarm !!!

arteck, 2018-09-20:
fz.generic_batterydone :-) and works

here i pushed the button 4x

```
zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +7ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +2ms
  zigbee2mqtt:debug 2018-9-20 09:39:50 Recieved zigbee message with data {"cid":"genPowerCfg","data":{"batteryPercentageRemaining":130}}
  zigbee2mqtt:info 2018-9-20 09:39:50 MQTT publish, topic: 'zigbee2mqtt/0x005043c903351248', payload: '{"battery":65,"linkquality":55}'
  zigbee2mqtt:debug 2018-9-20 09:39:50 Recieved zigbee message with data {"cid":"genPowerCfg","data":{"batteryPercentageRemaining":130}}
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +2s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +7ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +2ms
  zigbee2mqtt:debug 2018-9-20 09:39:51 Recieved zigbee message with data {"cid":"genPowerCfg","data":{"batteryPercentageRemaining":177}}
  zigbee2mqtt:info 2018-9-20 09:39:51 MQTT publish, topic: 'zigbee2mqtt/0x005043c903351248', payload: '{"battery":88.5,"linkquality":55}'
  zigbee2mqtt:debug 2018-9-20 09:39:51 Recieved zigbee message with data {"cid":"genPowerCfg","data":{"batteryPercentageRemaining":177}}
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +4s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +7ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +2ms
  zigbee2mqtt:debug 2018-9-20 09:39:56 Recieved zigbee message with data {"cid":"genPowerCfg","data":{"batteryPercentageRemaining":137}}
  zigbee2mqtt:info 2018-9-20 09:39:56 MQTT publish, topic: 'zigbee2mqtt/0x005043c903351248', payload: '{"battery":68.5,"linkquality":57}'
  zigbee2mqtt:debug 2018-9-20 09:39:56 Recieved zigbee message with data {"cid":"genPowerCfg","data":{"batteryPercentageRemaining":137}}
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +2s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +8ms
 
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +1ms
  zigbee2mqtt:debug 2018-9-20 09:39:57 Recieved zigbee message with data {"cid":"genPowerCfg","data":{"batteryPercentageRemaining":177}}
  zigbee2mqtt:info 2018-9-20 09:39:57 MQTT publish, topic: 'zigbee2mqtt/0x005043c903351248', payload: '{"battery":88.5,"linkquality":55}'
  zigbee2mqtt:debug 2018-9-20 09:39:57 Recieved zigbee message with data {"cid":"genPowerCfg","data":{"batteryPercentageRemaining":177}}
```

tb-killa, 2018-09-20:
This ``reporting`` shows up because its generic cluster .. no ``ssIasZone`` specific!
..........

arteck, 2018-09-20:
that means ??

tb-killa, 2018-09-20:
As posted on 
https://github.com/Koenkk/zigbee2mqtt/issues/384#issuecomment-423072116

I need the informations ...

arteck, 2018-09-20:
```
 zigbee2mqtt:info 2018-9-20 10:04:44 MQTT publish, topic: 'zigbee2mqtt/0x005043c903351248', payload: '{"battery":87.5,"linkquality":57}'
  zigbee2mqtt:debug 2018-9-20 10:04:44 Recieved zigbee message with data {"cid":"genPowerCfg","data":{"batteryPercentageRemaining":175}}
  zigbee2mqtt:warn 2018-9-20 10:04:44 No converter available for 'SMOK_YDLV10 smoke detector' with cid 'genPowerCfg', type 'devChange' and data '{"cid":"genPowerCfg","data":{"batteryPercentageRemaining":175}}'
  zigbee2mqtt:warn 2018-9-20 10:04:44 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object] +4s
```

tb-killa, 2018-09-20:
This converter message could be removed by insert ``fz.ignore_power_change`` on the ``fromZigbee`` section.

tb-killa, 2018-09-20:
``
error: Failed to configure 0x005043c903351248 0x005043c903351248 ??????? this is the device
``
okay that help ... what we now need is the detail section on which the failure happens so please start zigbee2mqtt on console manually via ``DEBUG=zigbee-shepherd* npm start`` and let them run.
after short time you should see the converter messages comes up .. we need to find the failure ones ..

maybe its the enroll request i think .. 

IASZone is special and very strange ... i don´t know if we could them complete integrate ..
Please provide link for buy .. maybe we could check for sample ..

arteck, 2018-09-20:
what should I look for ?

arteck, 2018-09-20:
maybe here
```
zigbee-shepherd:request REQ --> AF:dataRequest, transId: 24 +5ms
  zigbee-shepherd-converters:devices Configured '(cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb)' with result 'Error: request unsuccess: 135' +2s
  zigbee2mqtt:error 2018-9-20 19:33:18 Failed to configure 0x005043c903351248 0x005043c903351248
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +31ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +1s
  zigbee-shepherd:request REQ --> ZDO:activeEpReq +2ms
```

tb-killa, 2018-09-20:
Look like we got it...
Need to Figure what failure.
Maybe its already written?
Please Post actually ``database.db`` content

arteck, 2018-09-20:
[database.txt](https://github.com/Koenkk/zigbee2mqtt/files/2402428/database.txt)

arteck, 2018-09-23:
its no problem but i already did that.. press the button like xiaomi device.. nope
i have no idea .. ich check this with other stick.. maybe works ..

i  bought it on amazon its Heiman HS1SA-M..

tb-killa, 2018-09-23:
@arteck 
We could check if it could help by repeating the stuff ..

please test this and report.

```
{ 
      zigbeeModel: ['SMOK_YDLV10'], 
      model: 'SMOK_YDLV10 smoke detector',
      vendor: 'Heiman', 
      description: '',
      supports: '', 
      fromZigbee: [fz.generic_battery],
      toZigbee: [],
      configure: (ieeeAddr, shepherd, coordinator, callback) => {
           const device = shepherd.find(ieeeAddr, 1);
           const actions = [
                  (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                  (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                  (cb) => device.report('ssIasZone', 'zoneStatus', 0, 30, null, cb),
                  (cb) => device.report('ssIasZone', 'zoneStatus', 0, 30, null, cb),
                  (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 1, zoneid: 50}, cb),
                  (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 1, zoneid: 50}, cb),
              ];
               execute(device, actions, callback);
  
          }, 
    },
```

arteck, 2018-09-23:
here the message

```
zigbee2mqtt:info 2018-9-23 18:15:08 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:warn 2018-9-23 18:15:12 No converter available for 'SMOK_YDLV10 smoke detector' with cid 'ssIasZone', type 'devChange' and data '{"cid":"ssIasZone","data":{"iasCieAddr":"0x00124b0012022f35"}}'
  zigbee2mqtt:warn 2018-9-23 18:15:12 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:error 2018-9-23 18:15:14 Failed to configure 0x005043c903351248 0x005043c903351248
  zigbee2mqtt:warn 2018-9-23 18:15:14 No converter available for 'SMOK_YDLV10 smoke detector' with cid 'ssIasZone', type 'devChange' and data '{"cid":"ssIasZone","data":{"iasCieAddr":"0x00124b0012022f35"}}'
  zigbee2mqtt:warn 2018-9-23 18:15:14 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.

```

tb-killa, 2018-09-23:
Nice .. this read great because now the device got the right coordinator Addr.
What about the other log messages ?

Btw.. wir können dieses Thema auch gerne auf Deutsch bearbeiten ;) ..

arteck, 2018-09-23:
na dann 👍 

```

{"id":42,"type":"EndDevice","ieeeAddr":"0x005043c903351248","nwkAddr":60302,"manufId":48042,"manufName":"Heiman","powerSource":"Battery","modelId":"SMOK_YDLV10","epList":[1],"status":"online","joinTime":1537720042,"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,9,1280,1282],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":3,"stackVersion":2,"hwVersion":1,"manufacturerName":"Heiman","modelId":"SMOK_YDLV10","dateCode":"20150330","powerSource":3,"locationDesc":"","physicalEnv":0,"deviceEnabled":1,"alarmMask":0}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genAlarms":{"dir":{"value":1},"attrs":{"alarmCount":0}},"genOta":{"dir":{"value":2},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{"zoneState":0,"zoneType":40,"zoneStatus":32,"iasCieAddr":"0x0000000000000000","zoneId":0,"numZoneSensitivityLevelsSupported":2,"currentZoneSensitivityLevel":0}},"ssIasWd":{"dir":{"value":1},"attrs":{"maxDuration":240}}}}},"_id":"I15C1vqLNpLd5Cf2"}
{"id":42,"type":"EndDevice","ieeeAddr":"0x005043c903351248","nwkAddr":60302,"manufId":48042,"manufName":"Heiman","powerSource":"Battery","modelId":"SMOK_YDLV10","epList":[1],"status":"online","joinTime":1537720042,"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,9,1280,1282],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":3,"stackVersion":2,"hwVersion":1,"manufacturerName":"Heiman","modelId":"SMOK_YDLV10","dateCode":"20150330","powerSource":3,"locationDesc":"","physicalEnv":0,"deviceEnabled":1,"alarmMask":0}},"genPowerCfg":{"dir":{"value":1},"attrs":{"mainsVoltage":0,"mainsFrequency":0,"mainsAlarmMask":0,"mainsVoltMinThres":0,"mainsVoltMaxThres":65535,"mainsVoltageDwellTripPoint":0,"batteryVoltage":29,"batteryPercentageRemaining":168,"batteryManufacturer":"","batterySize":255,"batteryAHrRating":0,"batteryQuantity":0,"batteryRatedVoltage":30,"batteryAlarmMask":1,"batteryVoltMinThres":25,"batteryVoltThres1":0,"batteryVoltThres2":0,"batteryVoltThres3":0,"batteryPercentMinThres":40,"batteryPercentThres1":0,"batteryPercentThres2":0,"batteryPercentThres3":0,"batteryAlarmState":0}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genAlarms":{"dir":{"value":1},"attrs":{"alarmCount":0}},"genOta":{"dir":{"value":2},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{"zoneState":0,"zoneType":40,"zoneStatus":32,"iasCieAddr":"0x0000000000000000","zoneId":0,"numZoneSensitivityLevelsSupported":2,"currentZoneSensitivityLevel":0}},"ssIasWd":{"dir":{"value":1},"attrs":{"maxDuration":240}}}}},"_id":"I15C1vqLNpLd5Cf2"}

```

tb-killa, 2018-09-23:
Dein Converter-Output zeigt auf, dass ``iasCieAddr`` geändert wurde.
Aber dein Auszug aus der ``database.db`` spiegelt dies nicht wieder ..
Wichtig wäre auch die Informationen ob die anderen zwei Converter Konfigurationen auch einen positiven Log aufweisen, daher bitte einen vollständigen DEBUG-LOG posten.

arteck, 2018-09-23:
das sind alle mit id 42 aus der db

wobei ich hab den resettet und neu angelernt
hätt ich vielleicht nicht machen sollen

```
{"id":42,"type":"EndDevice","ieeeAddr":"0x005043c903351248","nwkAddr":39994,"manufId":48042,"manufName":"Heiman","powerSource":"Battery","modelId":"SMOK_YDLV10","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,9,1280,1282],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":3,"stackVersion":2,"hwVersion":1,"manufacturerName":"Heiman","modelId":"SMOK_YDLV10","dateCode":"20150330","powerSource":3,"locationDesc":"","physicalEnv":0,"deviceEnabled":1,"alarmMask":0}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genAlarms":{"dir":{"value":1},"attrs":{"alarmCount":0}},"genOta":{"dir":{"value":2},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{"zoneState":0,"zoneType":40,"zoneStatus":32,"iasCieAddr":"0x0000000000000000","zoneId":0,"numZoneSensitivityLevelsSupported":2,"currentZoneSensitivityLevel":0}},"ssIasWd":{"dir":{"value":1},"attrs":{"maxDuration":240}}}}},"_id":"I15C1vqLNpLd5Cf2"}
{"id":42,"type":"EndDevice","ieeeAddr":"0x005043c903351248","nwkAddr":39994,"manufId":48042,"manufName":"Heiman","powerSource":"Battery","modelId":"SMOK_YDLV10","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,9,1280,1282],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":3,"stackVersion":2,"hwVersion":1,"manufacturerName":"Heiman","modelId":"SMOK_YDLV10","dateCode":"20150330","powerSource":3,"locationDesc":"","physicalEnv":0,"deviceEnabled":1,"alarmMask":0}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genAlarms":{"dir":{"value":1},"attrs":{"alarmCount":0}},"genOta":{"dir":{"value":2},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{"zoneState":0,"zoneType":40,"zoneStatus":32,"iasCieAddr":"0x0000000000000000","zoneId":0,"numZoneSensitivityLevelsSupported":2,"currentZoneSensitivityLevel":0}},"ssIasWd":{"dir":{"value":1},"attrs":{"maxDuration":240}}}}},"_id":"I15C1vqLNpLd5Cf2"}
{"id":42,"type":"EndDevice","ieeeAddr":"0x005043c903351248","nwkAddr":60302,"manufId":48042,"manufName":"Heiman","powerSource":"Battery","modelId":"SMOK_YDLV10","epList":[1],"status":"online","joinTime":1537720042,"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,9,1280,1282],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":3,"stackVersion":2,"hwVersion":1,"manufacturerName":"Heiman","modelId":"SMOK_YDLV10","dateCode":"20150330","powerSource":3,"locationDesc":"","physicalEnv":0,"deviceEnabled":1,"alarmMask":0}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genAlarms":{"dir":{"value":1},"attrs":{"alarmCount":0}},"genOta":{"dir":{"value":2},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{"zoneState":0,"zoneType":40,"zoneStatus":32,"iasCieAddr":"0x0000000000000000","zoneId":0,"numZoneSensitivityLevelsSupported":2,"currentZoneSensitivityLevel":0}},"ssIasWd":{"dir":{"value":1},"attrs":{"maxDuration":240}}}}},"_id":"I15C1vqLNpLd5Cf2"}
{"id":42,"type":"EndDevice","ieeeAddr":"0x005043c903351248","nwkAddr":60302,"manufId":48042,"manufName":"Heiman","powerSource":"Battery","modelId":"SMOK_YDLV10","epList":[1],"status":"online","joinTime":1537720042,"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,9,1280,1282],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":3,"stackVersion":2,"hwVersion":1,"manufacturerName":"Heiman","modelId":"SMOK_YDLV10","dateCode":"20150330","powerSource":3,"locationDesc":"","physicalEnv":0,"deviceEnabled":1,"alarmMask":0}},"genPowerCfg":{"dir":{"value":1},"attrs":{"mainsVoltage":0,"mainsFrequency":0,"mainsAlarmMask":0,"mainsVoltMinThres":0,"mainsVoltMaxThres":65535,"mainsVoltageDwellTripPoint":0,"batteryVoltage":29,"batteryPercentageRemaining":168,"batteryManufacturer":"","batterySize":255,"batteryAHrRating":0,"batteryQuantity":0,"batteryRatedVoltage":30,"batteryAlarmMask":1,"batteryVoltMinThres":25,"batteryVoltThres1":0,"batteryVoltThres2":0,"batteryVoltThres3":0,"batteryPercentMinThres":40,"batteryPercentThres1":0,"batteryPercentThres2":0,"batteryPercentThres3":0,"batteryAlarmState":0}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genAlarms":{"dir":{"value":1},"attrs":{"alarmCount":0}},"genOta":{"dir":{"value":2},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{"zoneState":0,"zoneType":40,"zoneStatus":32,"iasCieAddr":"0x0000000000000000","zoneId":0,"numZoneSensitivityLevelsSupported":2,"currentZoneSensitivityLevel":0}},"ssIasWd":{"dir":{"value":1},"attrs":{"maxDuration":240}}}}},"_id":"I15C1vqLNpLd5Cf2"}

```

arteck, 2018-09-23:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/2408914/log.txt)

wobei

```
{"id":42,"type":"EndDevice","ieeeAddr":"0x005043c903351248","nwkAddr":60302,"manufId":48042,"manufName":"Heiman","powerSource":"Battery","modelId":"SMOK_YDLV10","epList":[1],"status":"online","joinTime":1537720042,"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,9,1280,1282],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":3,"stackVersion":2,"hwVersion":1,"manufacturerName":"Heiman","modelId":"SMOK_YDLV10","dateCode":"20150330","powerSource":3,"locationDesc":"","physicalEnv":0,"deviceEnabled":1,"alarmMask":0}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genAlarms":{"dir":{"value":1},"attrs":{"alarmCount":0}},"genOta":{"dir":{"value":2},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{"zoneState":0,"zoneType":40,"zoneStatus":32,"iasCieAddr":"0x0000000000000000","zoneId":0,"numZoneSensitivityLevelsSupported":2,"currentZoneSensitivityLevel":0}},"ssIasWd":{"dir":{"value":1},"attrs":{"maxDuration":240}}}}},"_id":"I15C1vqLNpLd5Cf2"}
{"id":42,"type":"EndDevice","ieeeAddr":"0x005043c903351248","nwkAddr":60302,"manufId":48042,"manufName":"Heiman","powerSource":"Battery","modelId":"SMOK_YDLV10","epList":[1],"status":"online","joinTime":1537720042,"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,9,1280,1282],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":3,"stackVersion":2,"hwVersion":1,"manufacturerName":"Heiman","modelId":"SMOK_YDLV10","dateCode":"20150330","powerSource":3,"locationDesc":"","physicalEnv":0,"deviceEnabled":1,"alarmMask":0}},"genPowerCfg":{"dir":{"value":1},"attrs":{"mainsVoltage":0,"mainsFrequency":0,"mainsAlarmMask":0,"mainsVoltMinThres":0,"mainsVoltMaxThres":65535,"mainsVoltageDwellTripPoint":0,"batteryVoltage":29,"batteryPercentageRemaining":168,"batteryManufacturer":"","batterySize":255,"batteryAHrRating":0,"batteryQuantity":0,"batteryRatedVoltage":30,"batteryAlarmMask":1,"batteryVoltMinThres":25,"batteryVoltThres1":0,"batteryVoltThres2":0,"batteryVoltThres3":0,"batteryPercentMinThres":40,"batteryPercentThres1":0,"batteryPercentThres2":0,"batteryPercentThres3":0,"batteryAlarmState":0}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genAlarms":{"dir":{"value":1},"attrs":{"alarmCount":0}},"genOta":{"dir":{"value":2},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{"zoneState":0,"zoneType":40,"zoneStatus":32,"iasCieAddr":"0x0000000000000000","zoneId":0,"numZoneSensitivityLevelsSupported":2,"currentZoneSensitivityLevel":0}},"ssIasWd":{"dir":{"value":1},"attrs":{"maxDuration":240}}}}},"_id":"I15C1vqLNpLd5Cf2"}

```

tb-killa, 2018-09-23:
``iasCieAddr`` ist immer noch im original-Zustand(0x0000000000000000).

Lerne noch mal das Gerät an und stelle den kompletten Log ein.
Möglicherweise funktioniert das Schreiben der ``iasCieAddr`` aber danach nicht mehr der ``enrollRsp``.

arteck, 2018-09-23:
das habe ich beretis ..das ist das was raus kam

habs sogar zuerst auch aus der db gelöscht..

tb-killa, 2018-09-23:
Hm. wir müssen es schaffen wieder den oben gezeigten Zustand zu erreichen!

arteck, 2018-09-23:
meinst du diesen [hier](https://github.com/Koenkk/zigbee2mqtt/issues/384#issuecomment-423827879)

tb-killa, 2018-09-23:
Genau dieser Zustand ist auf jeden Fall ein erster Schritt.

BillX86, 2018-09-24:
Hallo zusammen, ich habe auch Rauchmelder von Heiman... Ich habe diese auf ebay gekauft. Es scheint davon unterschiedliche Revisionen (eventuell nur Firmware) zu geben. Von außen sehen meine wie diese aushttps://www.replacedirect.de/product/P1018166/trust-zigbee-draadloze-rookdetector/pn/71197.html?utm_source=google&utm_medium=cpc&utm_campaign=shopping_adc&gclid=EAIaIQobChMIiN_63e7S3QIVj9CyCh02AwxYEAQYASABEgKUsPD_BwE. Es gibt sie wohl auch aus China.... Bei mir steht im production date inbder devices.DB 2015...Meiner meldet sich mit einem anderen Zigbee Model;  verhält sich aber deinem recht ähnlich.... Ich probiere also aktuell auch vor mich hin.

arteck, 2018-09-24:
> 
> 
> Genau dieser Zustand ist auf jeden Fall ein erster Schritt.

das ist nicht das Problem.. das bekomme ich reproduzierbar hin.. nur das wars.. 

`Failed to configure 0x005043c903351248 0x005043c903351248`

kommt dann trotzdem

arteck, 2018-09-26:
sagemal wo kommst du her ??

tb-killa, 2018-09-26:
Das sollten wir anderweitig kommunizieren.

arteck, 2018-09-27:
schreib mir mal ne email..... 
arteck@outlook.com
die Adresse ist ehh öffentlich da ich es im Adapter stehen habe
https://github.com/arteck/ioBroker.hs100

arteck, 2018-09-29:
ich habe zigbee2mqtt neu installiert.. hatte ein anderes Problem mit einem Hue Schalter
ok weiter gehts

das sieht doch schon mal anders aus als vorher

[database.txt](https://github.com/Koenkk/zigbee2mqtt/files/2430384/database.txt)

arteck, 2018-09-29:
ich habe zigbee2mqtt neu installiert.. hatte ein anderes Problem mit einem Hue Schalter
ok weiter gehts

das sieht doch schon mal anders aus als vorher

```
{
  "id": 3,
  "type": "EndDevice",
  "ieeeAddr": "0x005043c903351248",
  "nwkAddr": 26045,
  "manufId": 48042,
  "manufName": "Heiman",
  "powerSource": "Battery",
  "modelId": "SMOK_YDLV10",
  "epList": [
    1
  ],
  "status": "online",
  "joinTime": 1538205759,
  "endpoints": {
    "1": {
      "profId": 260,
      "epId": 1,
      "devId": 1026,
      "inClusterList": [
        0,
        1,
        3,
        9,
        1280,
        1282
      ],
      "outClusterList": [
        25
      ],
      "clusters": {
        "genBasic": {
          "dir": {
            "value": 1
          },
          "attrs": {
            "zclVersion": 1,
            "appVersion": 3,
            "stackVersion": 2,
            "hwVersion": 1,
            "manufacturerName": "Heiman",
            "modelId": "SMOK_YDLV10",
            "dateCode": "20150330",
            "powerSource": 3,
            "locationDesc": "",
            "physicalEnv": 0,
            "deviceEnabled": 1,
            "alarmMask": 0
          }
        },
        "genPowerCfg": {
          "dir": {
            "value": 1
          },
          "attrs": {
            "mainsVoltage": 0,
            "mainsFrequency": 0,
            "mainsAlarmMask": 0,
            "mainsVoltMinThres": 0,
            "mainsVoltMaxThres": 65535,
            "mainsVoltageDwellTripPoint": 0,
            "batteryVoltage": 29,
            "batteryPercentageRemaining": 165,
            "batteryManufacturer": "",
            "batterySize": 255,
            "batteryAHrRating": 0,
            "batteryQuantity": 0,
            "batteryRatedVoltage": 30,
            "batteryAlarmMask": 1,
            "batteryVoltMinThres": 25,
            "batteryVoltThres1": 0,
            "batteryVoltThres2": 0,
            "batteryVoltThres3": 0,
            "batteryPercentMinThres": 40,
            "batteryPercentThres1": 0,
            "batteryPercentThres2": 0,
            "batteryPercentThres3": 0,
            "batteryAlarmState": 0
          }
        },
        "genIdentify": {
          "dir": {
            "value": 1
          },
          "attrs": {
            "identifyTime": 0
          }
        },
        "genAlarms": {
          "dir": {
            "value": 1
          },
          "attrs": {
            "alarmCount": 0
          }
        },
        "genOta": {
          "dir": {
            "value": 2
          },
          "attrs": {}
        },
        "ssIasZone": {
          "dir": {
            "value": 1
          },
          "attrs": {
            "zoneState": 0,
            "zoneType": 40,
            "zoneStatus": 32,
            "iasCieAddr": "0x00124b0012022f35",
            "zoneId": 0,
            "numZoneSensitivityLevelsSupported": 2,
            "currentZoneSensitivityLevel": 0
          }
        },
        "ssIasWd": {
          "dir": {
            "value": 1
          },
          "attrs": {
            "maxDuration": 240
          }
        }
      }
    }
  },
  "_id": "hbewaw5qnaSj2j73"
}
```

tb-killa, 2018-10-03:
@arteck .. konntest du nochmals testen in der Zwischenzeit ?

arteck, 2018-10-04:
na klar

```
zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +2ms
  zigbee2mqtt:debug 2018-10-4 20:58:36 Recieved zigbee message of type 'devChange' with data '{"cid":"ssIasZone","data":{"zoneState":0,"zoneType":40,"zoneStatus":32,"iasCieAddr":"0x0000000000000000","zoneId":0,"numZoneSensitivityLevelsSupported":2,"currentZoneSensitivityLevel":0}}' of device 'SMOK_YDLV10' (0x005043c903351248)
  zigbee2mqtt:warn 2018-10-4 20:58:36 No converter available for 'SMOK_YDLV10 smoke detector' with cid 'ssIasZone', type 'devChange' and data '{"cid":"ssIasZone","data":{"zoneState":0,"zoneType":40,"zoneStatus":32,"iasCieAddr":"0x0000000000000000","zoneId":0,"numZoneSensitivityLevelsSupported":2,"currentZoneSensitivityLevel":0}}'
  zigbee2mqtt:warn 2018-10-4 20:58:36 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.

```

JLeandroFreitas, 2018-10-11:
Hey please me out i have the same problem but i dont understand dutch.

tb-killa, 2018-10-11:
hi @JLeandroFreitas 
i´m working on general implementation for ``ssiaszone`` to get them easier integrated.
if i´m on the stage for testing i will post.

JLeandroFreitas, 2018-10-11:
> hi @JLeandroFreitas
> i´m working on general implementation for `ssiaszone` to get them easier integrated.
> if i´m on the stage for testing i will post.

I got order zigbeeModel:
    // Heiman Smoke
{
        zigbeeModel: ['b5db59bfd81e4f1f95dc57fdbba17931'], 
        model: 'HS1SA smoke detector',
        vendor: 'Heiman', 
        description: '',
        supports: 'smoke',
        fromZigbee: [fz.generic_battery],
        toZigbee: [],
	configure: (ieeeAddr, shepherd, coordinator, callback) => {
	const device = shepherd.find(ieeeAddr, 1);
	const writeRecArgs_ssIasZone = {
                attrId: 16,
                dataType: 'ieeeAddr',
                attrData: coordinator.device.getIeeeAddr()
        };
	const enrollRspArgs_ssIasZone = {
                enrollrspcode: 0x00,
                zoneid : 1 // Hier bitte entsprechend der Anzahl an Endgeräten die ID erhöhen +1!
        };
	const actions = [
                (cb) => device.foundation('ssIasZone', 'write', [ writeRecArgs_ssIasZone ], { direction: 0}, cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', [ enrollRspArgs_ssIasZone ], cb),
                (cb) => device.report('ssIasZone', 'zoneStatus', 0, 30, null, cb),
            ];
             execute(device, actions, callback);
        },
    },
];

JLeandroFreitas, 2018-10-14:
> hi @tb-killa 
I got  other same product but other zigbeeModel:
Please let me know of your process
    // Heiman Smoke

        zigbeeModel: ['b5db59bfd81e4f1f95dc57fdbba17931'], 
 
Oct 11 21:26:08 hassbian npm[18686]:   zigbee2mqtt:warn 2018-10-11 21:26:08 Device with modelID 'b5db59bfd81e4f1f95dc57fdbba17931' is not supported.Oct 11 21:27:56 hassbian npm[18686]:

JLeandroFreitas, 2018-10-25:
@tb-killa Hey, how is progress? I got other modelname its oem brand Orvibo ZigBee Smoke Sensor [](https://www.mymemory.co.uk/orvibo-zigbee-smoke-sensor-white.html). Can i help out testing?

arteck, 2018-11-06:
push

JLFN, 2018-11-28:
Same issue as https://github.com/Koenkk/zigbee2mqtt/issues/279

JLFN, 2018-11-30:
Support added 619e1e4

JLFN, 2018-11-30:
@Arteck support is now merged

JLFN, 2018-12-02:
@Arteck no battery report or sensitivity settings yet@Koenkk havent implemented that yet.