Koenkk/zigbee2mqtt #385: Docker-Image for RasPi 3B

jonny0303, 2018-09-19:
Hi,

unfortunately there isn't a working docker image for raspi rev. a02082 (3 Model B).
will be there a working image in future? :-)

```
WARNING: Your kernel does not support memory swappiness capabilities, memory swappiness discarded.
Error response from daemon: Cannot start container 00810fc8286de02223dc84f782eea7b93db9899a76fea0ae79812d68d23b7462: operation not supported
```

cheers

Koenkk, 2018-09-19:
`arm32v6` should work for pi 3

jonny0303, 2018-09-19:
```
$ sudo docker run -it -v /home/homeassistant/data:/app/data --device=/dev/ttyAMA0 koenkk/zigbee2mqtt:arm32v6
WARNING: Your kernel does not support memory swappiness capabilities, memory swappiness discarded.
Error response from daemon: Cannot start container c51a533a9fcadc9932e2a5c63ad32dbde3333d5788a04c3b7da740ba5dc24de0: operation not supported
```

Koenkk, 2018-09-19:
What os are you running?

jonny0303, 2018-09-19:
Raspian Stretch
```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.4 (stretch)
Release:	9.4
Codename:	stretch
```

jonny0303, 2018-09-19:
seems to work now after a kernel update 😮 
thank you for your time!

closed 2018-09-19 by jonny0303