Koenkk/zigbee2mqtt #388: Aqara Cube doesn't work with Firmware 20180815

AnhDuc85, 2018-09-19:
Just flashed firmware 20180815 to my CC2531 and the cube doesn't work anymore. The cube worked fine under 20180705 no matter which docker images has been used. How can it be?

db entry of the paired and previous working cube under firmware 20180705: https://pastebin.com/raw/xRPQYxTc

db entry of the paired and previous working cube under firmware 20180815: https://pastebin.com/raw/EUNYTe4t

ben423423n32j14e, 2018-09-19:
You know it's necessary to re-pair a device after updating the firmware?

AnhDuc85, 2018-09-19:
sure, I repaired all my devices, that's why I'm so confused as my cube is not working anymore

AnhDuc85, 2018-09-21:
seems like the modelID can not be recognized with new firmware.
When I edit devices.js and replace modelID 'lumi.sensor_cube' with "0x656275635f726f736e65732e696d756c" the cube is working as before.

aptonline, 2018-09-22:
In relation to this I just received my Aqara cube (ordered an Xiaomi but I'm guessing they are the same) but I cannot get the tap function to work.  All other functions are being seen in terminal except this.

Any ideas?

Also how can you update the firmware of the cube without an original hub and Xiaomi app?  It is possible?

AnhDuc85, 2018-09-22:
@aptonline tap is working out of the box, some samples:
payload: '{"battery":"98.33","voltage":2995,"linkquality":105,"action":"tap","side":0}'
payload: '{"battery":"98.33","voltage":2995,"linkquality":107,"action":"tap","side":5}'

As far as I know, there is no other way to update the cube firmware except via xiaomi hub and app.

aptonline, 2018-09-23:
No for me @AnhDuc85, I've double checked and not getting any `tap` payload exposed in terminal. All other functions are there.  Running `zigbee2mqtt@0.1.5`

AnhDuc85, 2018-09-23:
do you get the "tap" via Mi Home app?

wilfredsmit, 2018-09-24:
> seems like the modelID can not be recognized with new firmware.
> When I edit devices.js and replace modelID 'lumi.sensor_cube' with "0x656275635f726f736e65732e696d756c" the cube is working as before.

Here the same. As a workaround it's fine, but I hope it will be fixed.

Koenkk, 2018-09-25:
@wilfredsmit @AnhDuc85 are you sure this is because of the firmware update and not because a zigbee2mqtt update?

AnhDuc85, 2018-09-25:
It stopped working after I updated my firmware and re-paired the cube. Also, I tried with old zigbee2mqtt image but still same.

Within the next days, I will get another CC2531, will flash old firmware and double check. Don't want to use my current stick and re-pair the devices again :-)

wilfredsmit, 2018-09-25:
The same for me. I updated my firmware and re-paired the cube. Btw I use the docker image.

Koenkk, 2018-09-26:
@AnhDuc85 could you trigger some actions when running both firwmare?

You `20180815` has some more events:

```
  zigbee2mqtt:debug 2018-9-25 19:25:06 Recieved zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"modelId":"lumi.sensor_cube.aqgl01"}}' of device '0x656275635f726f736e65732e696d756c' (0x00158d0002791e60)
  zigbee2mqtt:warn 2018-9-25 19:25:06 Device with modelID '0x656275635f726f736e65732e696d756c' is not supported.
  zigbee2mqtt:warn 2018-9-25 19:25:06 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
  zigbee2mqtt:debug 2018-9-25 19:25:07 Recieved zigbee message of type 'devInterview' with data '"0x00158d0002791e60"'
```

AnhDuc85, 2018-09-26:
same with all events (tap, rotate, flip, ...). Only the presentValues changes:

2018-09-26T10:43:07.761513174Z   zigbee2mqtt:warn 2018-9-26 10:43:07 Device with modelID '0x656275635f726f736e65732e696d756c' is not supported.
2018-09-26T10:43:07.761804293Z   zigbee2mqtt:warn 2018-9-26 10:43:07 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
2018-09-26T10:43:09.077634474Z   zigbee2mqtt:debug 2018-9-26 10:43:09 Recieved zigbee message of type 'attReport' with data '{"cid":"genMultistateInput","data":{"presentValue":74}}' of device '0x656275635f726f736e65732e696d756c' (0x00158d0002791e60)
2018-09-26T10:43:09.078687909Z   zigbee2mqtt:warn 2018-9-26 10:43:09 Device with modelID '0x656275635f726f736e65732e696d756c' is not supported.
2018-09-26T10:43:09.079663091Z   zigbee2mqtt:warn 2018-9-26 10:43:09 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
2018-09-26T10:43:10.381207260Z   zigbee2mqtt:debug 2018-9-26 10:43:10 Recieved zigbee message of type 'attReport' with data '{"cid":"genMultistateInput","data":{"presentValue":84}}' of device '0x656275635f726f736e65732e696d756c' (0x00158d0002791e60)
2018-09-26T10:43:10.382254528Z   zigbee2mqtt:warn 2018-9-26 10:43:10 Device with modelID '0x656275635f726f736e65732e696d756c' is not supported.
2018-09-26T10:43:10.383267175Z   zigbee2mqtt:warn 2018-9-26 10:43:10 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
2018-09-26T10:43:14.999862118Z   zigbee2mqtt:debug 2018-9-26 10:43:14 Recieved zigbee message of type 'attReport' with data '{"cid":"genMultistateInput","data":{"presentValue":516}}' of device '0x656275635f726f736e65732e696d756c' (0x00158d0002791e60)
2018-09-26T10:43:15.000384229Z   zigbee2mqtt:warn 2018-9-26 10:43:15 Device with modelID '0x656275635f726f736e65732e696d756c' is not supported.
2018-09-26T10:43:15.000969296Z   zigbee2mqtt:warn 2018-9-26 10:43:15 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
2018-09-26T10:43:21.218341588Z   zigbee2mqtt:debug 2018-9-26 10:43:21 Recieved zigbee message of type 'attReport' with data '{"cid":"genMultistateInput","data":{"presentValue":516}}' of device '0x656275635f726f736e65732e696d756c' (0x00158d0002791e60)

Koenkk, 2018-09-26:
And how does it look like when running with the `20180507` firmware?

AnhDuc85, 2018-09-26:
Seems like I mixed up something. Problem is not relating to Firmware.
However, it's not possible to pair the cube with stable and latest-dev BUT it's possible to pair with 0.1.3.
Once the cube is paired, switch back to latest-dev or stable and everything seems fine.
@wilfredsmit can you try that?

wilfredsmit, 2018-09-27:
@AnhDuc85 I'm afraid I don't have the time to test this the coming days, as I'm not at home.

apeeters, 2018-09-27:
I just paired 2 brand new cubes (with newer firmware, I guess). These are not recognised: "Device with modelID '0x656275635f726f736e65732e696d756c' is not supported.". Older cubes still work, so I'm not sure the problem is caused by zigbee2mqtt.

As @JaapBraam mentioned before, when decoding this modelID hex string, we get the normal model id but reversed...?

apeeters, 2018-09-27:
For the record: the old (working) cubes are Mi branded whereas the new ones are Aqara branded.

Koenkk, 2018-09-28:
Probably the issue is due to a recent change in https://github.com/Koenkk/zcl-packet or https://github.com/Koenkk/zigbee-shepherd. Will investigate...

Koenkk, 2018-09-29:
Fixed in the dev branch, thanks for reporting (don't forget to repair)

closed 2018-09-29 by Koenkk

Koenkk, 2018-10-14:
@ABeumer525 can you check the dev branch?

wilfredsmit, 2018-10-16:
It is still a problem in 0.1.7

michel72, 2018-10-16:
yup, having the same issue here. The cube was working fine before.

Koenkk, 2018-10-16:
@wilfredsmit have you done a `rm -rf node_modules && npm install`?

wilfredsmit, 2018-10-17:
@Koenkk No. I use docker. So as far as I know this should not be necessary. But correct me if I'm wrong.

Koenkk, 2018-10-17:
@wilfredsmit before pairing it again, did you remove all entries of this device from `database.db`? (and then restart zigbee2mqtt)

wilfredsmit, 2018-10-17:
No, I have not done this. Do you mean I have to remove the cube entry by entering the MQTT command `zigbee2mqtt/bridge/config/remove` with the cube device address as the payload? And then reset the cube and pair it again?

Koenkk, 2018-10-17:
Try:
- Stop zigbee2mqtt
- Remove all entries of the cube (by searching on ieeAddress) from `database.db`
- Start zigbee2mqtt
- Pair cube

maartendamen, 2018-10-18:
I am also experiencing this issue with a brand new wxkg12lm straight from the box. Paired the same model succesfully several times before. I get these errors in the logs (I am on dev branch):

Device with modelID '0x746977735f726f736e65732e696d756c' is not supported.

wilfredsmit, 2018-10-19:
@Koenkk Hi Koen,
Still no success. This is my log.

```
  zigbee2mqtt:info 2018-10-19 16:50:30 Zigbee: allowing new devices to join.
  zigbee2mqtt:info 2018-10-19 16:50:54 New device with address 0x00158d0001041de0 connected!
  zigbee2mqtt:info 2018-10-19 16:50:54 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"device_connected","message":"0x00158d0001041de0"}'
  zigbee2mqtt:warn 2018-10-19 16:50:57 Device with modelID 'lumi.sensor_cube' is not supported.
  zigbee2mqtt:warn 2018-10-19 16:50:57 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
  zigbee2mqtt:warn 2018-10-19 16:50:58 Device with modelID 'lumi.sensor_cube' is not supported.
  zigbee2mqtt:warn 2018-10-19 16:50:58 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
  zigbee2mqtt:warn 2018-10-19 16:51:00 Device with modelID 'lumi.sensor_cube' is not supported.

```

Ciubas, 2020-05-19:
Whre is file database.db in Home Assistant with Supervisor in Docker and Zigbee2MQTT as Supervised addon?