Koenkk/zigbee2mqtt #389: Auto update

jwillmer, 2018-09-19:
How about using something like: https://github.com/juampi92/auto-updater to update the application to a new version?

Koenkk, 2018-09-20:
Although I don't like auto-updating, I like the new version/easier updating process this library could introduce.

jwillmer, 2018-09-20:
The update process in the wiki works well but I think this could easily added to a script to automate the update to some extend.
In the future I think there could be a flag to allow automatic updates of non breaking changes?

jwillmer, 2018-09-23:
After the latest version increases in just a few days I think a simple update process would be in order. We can postpone the auto update functionality but a script that stops the service, updates it and restarts the service would be very nice.

jwillmer, 2018-10-20:
How about adding the following script to the project?

update.sh
```
#!/bin/bash
# Source: https://github.com/Koenkk/zigbee2mqtt/wiki/Running-the-bridge#6-for-later-update-zigbee2mqtt-to-the-latest-version

# Stop zigbee2mqtt and go to directory
sudo systemctl stop zigbee2mqtt
cd /opt/zigbee2mqtt

# Backup configuration
cp -R data data-backup

# Update
git checkout HEAD -- npm-shrinkwrap.json
git pull
rm -rf node_modules
npm install

# Restore configuration
cp -R data-backup/* data
rm -rf data-backup

# Start zigbee2mqtt
sudo systemctl start zigbee2mqtt
```

Koenkk, 2018-10-21:
Good idea! can you make a PR?

PR #525 (2018-10-22): Adding update script
Fix #389
commit e54de4be (2018-10-22): Adding update script (#525)

* Adding update script

Fix #389

* Update update.sh :)

* Make executable & update update.sh
closed 2018-10-22 by jwillmer