Koenkk/zigbee2mqtt #391: xiaomi-honeywell-smoke-detector improvement for (Testing-Status)

tb-killa, 2018-09-20:
We got the xiaomi-honeywell-smoke-detector basic implemented: 
[zigbeeModel: ['lumi.sensor_smoke'],](https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L283)

What i think we need ti improve or implement is the "testing" state.

If you install your smoke detector you could test this ones by pressing the button 4 seconds should trigger short alarm and the ``msg.data.zoneStatus`` should be triggered to value ``0x0002``.

We should use this status and save with actually timestamp to show that its tested.

Does somebody who is in touch with this type of device could please do the test via ``DEBUG=zigbee-shepherd* npm start`` and provide full log after push and hold 4 seconds (maybe more ??) until the sound beeps ... we should see the new zoneStatus hopefully.

ben423423n32j14e, 2018-09-21:
Test results from 0.1.5

  zigbee2mqtt:debug 2018-9-21 15:30:32 Recieved zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":2}' of device 'lumi.sensor_smoke' (0x00158d000153b9ff)
  zigbee2mqtt:info 2018-9-21 15:30:32 MQTT publish, topic: 'zigbee2mqtt/livingroomsmokedetector', payload: '{"battery":"100.00","voltage":3055,"linkquality":60,"smoke":false}'
  zigbee2mqtt:debug 2018-9-21 15:30:35 Recieved zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":0}' of device 'lumi.sensor_smoke' (0x00158d000153b9ff)
  zigbee2mqtt:info 2018-9-21 15:30:35 MQTT publish, topic: 'zigbee2mqtt/livingroomsmokedetector', payload: '{"battery":"100.00","voltage":3055,"linkquality":63,"smoke":false}'

Koenkk, 2018-09-24:
This message is send over mqtt, right?

ben423423n32j14e, 2018-09-24:
Koenkk, I just checked that, no it is not being sent over MQTT, only displayed in the Zigbee2MQTT console.

tb-killa, 2018-09-25:
The message isnt parse in our converter. I could edit the output to allow new state parse by mqtt.
But after that.. See my question above. We need a valid way to store it.

ben423423n32j14e, 2018-09-25:
Isn't storing it up to the application connected? Eg Home Assistant

Koenkk, 2018-09-25:
I prefer to store it on the homeassistant side and not on the zigbee2mqtt side.

closed 2018-10-22 by tb-killa