Koenkk/zigbee2mqtt #393: Schwaiger ZHS-15 - Outlet with Power Measurement

ThomasKra, 2018-09-20:
I recently succcessfully paired a Schwaiger ZHS-15 Outlet

I got the following information in my _database.db_

`{"id":5,"type":"Router","ieeeAddr":"0x00124b00071970ca","nwkAddr":34503,"manufId":0,"manufName":"RT-RK","powerSource":"Mains (single phase)","modelId":"SPW35Z-D0","epList":[1,2,3,14],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,3,4,5,6,1794,2820],"outClusterList":[3],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"hwVersion":4,"manufacturerName":"RT-RK","modelId":"SPW35Z-D0","dateCode":"2016042100000105","powerSource":1,"locationDesc":"                ","physicalEnv":0}},"genIdentify":{"dir":{"value":3},"attrs":{"identifyTime":0}},"genGroups":{"dir":{"value":1},"attrs":{"nameSupport":0}},"genScenes":{"dir":{"value":1},"attrs":{"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":0}},"seMetering":{"dir":{"value":1},"attrs":{"currentSummDelivered":{"0":0,"1":0},"status":0,"unitOfMeasure":0,"summaFormatting":96,"meteringDeviceType":0}},"haElectricalMeasurement":{"dir":{"value":1},"attrs":{"apparentPower":0,"acPowerMultiplier":1,"acPowerDivisor":1}}}},"2":{"profId":260,"epId":2,"devId":0,"inClusterList":[0,3,2821],"outClusterList":[3,6],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"hwVersion":4,"manufacturerName":"RT-RK","modelId":"SPW35Z-D0","dateCode":"2016042100000105","powerSource":1,"locationDesc":"                ","physicalEnv":0}},"genIdentify":{"dir":{"value":3},"attrs":{"identifyTime":0}},"genOnOff":{"dir":{"value":2},"attrs":{"onOff":0}},"haDiagnostic":{"dir":{"value":1},"attrs":{"numberOfResets":5,"macTxUcastRetry":0,"macTxUcastFail":0,"aPSTxBcast":32,"aPSTxUcastSuccess":377,"aPSTxUcastRetry":0,"aPSTxUcastFail":0,"apsDecryptFailures":0,"packetValidateDropCount":0,"lastMessageLqi":70,"lastMessageRssi":-43}}}},"3":{"profId":50690,"epId":3,"devId":3,"inClusterList":[0,2,3],"outClusterList":[0,2,3,4],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genDeviceTempCfg":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}}}},"14":{"profId":260,"epId":14,"devId":0,"inClusterList":[],"outClusterList":[25],"clusters":{"genOta":{"dir":{"value":2},"attrs":{"upgradeServerId":"0xffffffffffffffff","fileOffset":4294967295,"currentFileVersion":1,"currentZigbeeStackVersion":2,"downloadedFileVersion":4294967295,"downloadedZigbeeStackVersion":65535,"imageUpgradeStatus":0,"manufacturerId":4477,"minimumBlockReqDelay":0,"imageStamp":4294967295}}}}},"_id":"7aNCEDtycH7f6a91"}
{"id":5,"type":"Router","ieeeAddr":"0x00124b00071970ca","nwkAddr":34503,"manufId":0,"manufName":"RT-RK","powerSource":"Mains (single phase)","modelId":"SPW35Z-D0","epList":[1,2,3,14],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,3,4,5,6,1794,2820],"outClusterList":[3],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"hwVersion":4,"manufacturerName":"RT-RK","modelId":"SPW35Z-D0","dateCode":"2016042100000105","powerSource":1,"locationDesc":"                ","physicalEnv":0}},"genIdentify":{"dir":{"value":3},"attrs":{"identifyTime":0}},"genGroups":{"dir":{"value":1},"attrs":{"nameSupport":0}},"genScenes":{"dir":{"value":1},"attrs":{"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":0}},"seMetering":{"dir":{"value":1},"attrs":{"currentSummDelivered":{"0":0,"1":0},"status":0,"unitOfMeasure":0,"summaFormatting":96,"meteringDeviceType":0}},"haElectricalMeasurement":{"dir":{"value":1},"attrs":{"apparentPower":0,"acPowerMultiplier":1,"acPowerDivisor":1}}}},"2":{"profId":260,"epId":2,"devId":0,"inClusterList":[0,3,2821],"outClusterList":[3,6],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"hwVersion":4,"manufacturerName":"RT-RK","modelId":"SPW35Z-D0","dateCode":"2016042100000105","powerSource":1,"locationDesc":"                ","physicalEnv":0}},"genIdentify":{"dir":{"value":3},"attrs":{"identifyTime":0}},"genOnOff":{"dir":{"value":2},"attrs":{"onOff":0}},"haDiagnostic":{"dir":{"value":1},"attrs":{"numberOfResets":5,"macTxUcastRetry":0,"macTxUcastFail":0,"aPSTxBcast":32,"aPSTxUcastSuccess":377,"aPSTxUcastRetry":0,"aPSTxUcastFail":0,"apsDecryptFailures":0,"packetValidateDropCount":0,"lastMessageLqi":70,"lastMessageRssi":-43}}}},"3":{"profId":50690,"epId":3,"devId":3,"inClusterList":[0,2,3],"outClusterList":[0,2,3,4],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genDeviceTempCfg":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}}}},"14":{"profId":260,"epId":14,"devId":0,"inClusterList":[],"outClusterList":[25],"clusters":{"genOta":{"dir":{"value":2},"attrs":{"upgradeServerId":"0xffffffffffffffff","fileOffset":4294967295,"currentFileVersion":1,"currentZigbeeStackVersion":2,"downloadedFileVersion":4294967295,"downloadedZigbeeStackVersion":65535,"imageUpgradeStatus":0,"manufacturerId":4477,"minimumBlockReqDelay":0,"imageStamp":4294967295}}}}},"_id":"7aNCEDtycH7f6a91"}

{"id":5,"type":"Router","ieeeAddr":"0x00124b00071970ca","nwkAddr":34503,"manufId":0,"manufName":"RT-RK","powerSource":"Mains (single phase)","modelId":"SPW35Z-D0","epList":[1,2,3,14],"status":"online","joinTime":1537385498,"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,3,4,5,6,1794,2820],"outClusterList":[3],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"hwVersion":4,"manufacturerName":"RT-RK","modelId":"SPW35Z-D0","dateCode":"2016042100000105","powerSource":1,"locationDesc":"                ","physicalEnv":0}},"genIdentify":{"dir":{"value":3},"attrs":{"identifyTime":0}},"genGroups":{"dir":{"value":1},"attrs":{"nameSupport":0}},"genScenes":{"dir":{"value":1},"attrs":{"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":0}},"seMetering":{"dir":{"value":1},"attrs":{"currentSummDelivered":{"0":0,"1":0},"status":0,"unitOfMeasure":0,"summaFormatting":96,"meteringDeviceType":0}},"haElectricalMeasurement":{"dir":{"value":1},"attrs":{"apparentPower":0,"acPowerMultiplier":1,"acPowerDivisor":1}}}},"2":{"profId":260,"epId":2,"devId":0,"inClusterList":[0,3,2821],"outClusterList":[3,6],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"hwVersion":4,"manufacturerName":"RT-RK","modelId":"SPW35Z-D0","dateCode":"2016042100000105","powerSource":1,"locationDesc":"                ","physicalEnv":0}},"genIdentify":{"dir":{"value":3},"attrs":{"identifyTime":0}},"genOnOff":{"dir":{"value":2},"attrs":{"onOff":0}},"haDiagnostic":{"dir":{"value":1},"attrs":{"numberOfResets":5,"macTxUcastRetry":0,"macTxUcastFail":0,"aPSTxBcast":32,"aPSTxUcastSuccess":377,"aPSTxUcastRetry":0,"aPSTxUcastFail":0,"apsDecryptFailures":0,"packetValidateDropCount":0,"lastMessageLqi":70,"lastMessageRssi":-43}}}},"3":{"profId":50690,"epId":3,"devId":3,"inClusterList":[0,2,3],"outClusterList":[0,2,3,4],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genDeviceTempCfg":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}}}},"14":{"profId":260,"epId":14,"devId":0,"inClusterList":[],"outClusterList":[25],"clusters":{"genOta":{"dir":{"value":2},"attrs":{"upgradeServerId":"0xffffffffffffffff","fileOffset":4294967295,"currentFileVersion":1,"currentZigbeeStackVersion":2,"downloadedFileVersion":4294967295,"downloadedZigbeeStackVersion":65535,"imageUpgradeStatus":0,"manufacturerId":4477,"minimumBlockReqDelay":0,"imageStamp":4294967295}}}}},"_id":"7aNCEDtycH7f6a91"}
{"id":5,"type":"Router","ieeeAddr":"0x00124b00071970ca","nwkAddr":34503,"manufId":0,"manufName":"RT-RK","powerSource":"Mains (single phase)","modelId":"SPW35Z-D0","epList":[1,2,3,14],"status":"online","joinTime":1537385498,"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,3,4,5,6,1794,2820],"outClusterList":[3],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"hwVersion":4,"manufacturerName":"RT-RK","modelId":"SPW35Z-D0","dateCode":"2016042100000105","powerSource":1,"locationDesc":"                ","physicalEnv":0}},"genIdentify":{"dir":{"value":3},"attrs":{"identifyTime":0}},"genGroups":{"dir":{"value":1},"attrs":{"nameSupport":0}},"genScenes":{"dir":{"value":1},"attrs":{"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":1}},"seMetering":{"d [... cut ...]
        convert: (model, msg, publish, options) => {
            console.log(msg);
        },
    },
`

What works right now is reading the current power (is published as 'apparentPower') and the switch State. But when I look at the database entries, I do find some other values like 'acPowerMultiplier' and 'acPowerDivisor'. How do I get those?

I also found, that some devices use a _configure_ function within _devices.js_ . How does that work and what is the benefit from this function? I didn't find any documentation on that...

tb-killa, 2018-09-20:
Hi please test this ones for first testing.. we could bring them in general if we got the working status.
First.. please remove all your own converter sections and replace your ``devices.js`` with this ones.

After that .. please restart your zigbee2mqtt and provide full log via ``DEBUG=zigbee-shepherd* npm start`` via pastebin or other service.

```
{
        zigbeeModel: ['SPW35Z-D0'],
        model: 'ZHS-15',
        vendor: 'Schwaiger',
        description: 'Power socket on/off with power consumption monitoring',
        supports: 'on/off, power measurement',
        fromZigbee: [fz.generic_state, fz.ignore_onoff_change, fz.ignore_electrical_change, fz.Z809A_power],
        toZigbee: [tz.onoff],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const cfg = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 1000, repChange: 0};
            const actions = [
		(cb) => device.bind('genOnOff', coordinator, cb),
                (cb) => device.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb),
                (cb) => device.report('haElectricalMeasurement', 'rmsVoltage', 10, 1000, 1, cb),
                (cb) => device.report('haElectricalMeasurement', 'rmsCurrent', 10, 1000, 1, cb),
                (cb) => device.report('haElectricalMeasurement', 'activePower', 10, 1000, 1, cb),
                (cb) => device.report('haElectricalMeasurement', 'powerFactor', 10, 1000, 1, cb),
            ];

            execute(device, actions, callback);
        },
    },
```

ThomasKra, 2018-09-20:
Wow, this works!
What exactly does the **configure** part do?

ThomasKra, 2018-09-20:
Here is the log: [Debug output](https://pastebin.com/CL4x6YKg)

But there is a new problem now: I paired another outlet of the same type, but it fails to configure

See line 696 in the logfile:

```
2018-09-20T19:30:20.921Z zigbee-shepherd-converters:devices Configured '(cb) => device.bind('genOnOff', coordinator, cb)' with result 'Error: Timed out after 10000 ms'
[31m  zigbee2mqtt:error[39m 2018-9-20 19:30:20 Failed to configure schwaiger_plug2 0x00124b0007196d0b
```

But I receive messages from this device... Any ideas?

closed 2018-11-28 by stale[bot]