Koenkk/zigbee2mqtt #395: Missing temperature for lumi.weather with both 0.1.5 and dev branch

dzungpv, 2018-09-21:
Firmware cc2631 from Koenkk.
I see the temperature is read but after the mqtt publish missing temperature, only with humidity and pressure. Seem this affect many device not only with this sensor like https://github.com/Koenkk/zigbee2mqtt/issues/387
```
'{"cid":"msTemperatureMeasurement","data":{"measuredValue":3166}}' of device 'lumi.weather' (0x00xxxxd000201b4xx)
  zigbee2mqtt:info 2018-9-21 03:22:04 MQTT publish, topic: 'zigbee2mqtt/0x00xxxxd000201b4xx', payload: '{"pressure":1010,"linkquality":99,"temperature":31.66}'
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +9ms
  zigbee2mqtt:debug 2018-9-21 03:22:04 Recieved zigbee message of type 'attReport' with data '{"cid":"msRelativeHumidity","data":{"measuredValue":5931}}' of device 'lumi.weather' (0x00xxxxd000201b4xx)
  zigbee2mqtt:info 2018-9-21 03:22:04 MQTT publish, topic: 'zigbee2mqtt/0x00xxxxd000201b4xx', payload: '{"temperature":31.66,"linkquality":99,"humidity":59.31}'
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 1027, srcaddr: 18442, srcendpoint: 1, dstendpoint: 1, wasbroadcast: 0, linkquality: 99, securityuse: 0, timestamp: 1369314, transseqnumber: 0, len: 17, data: <Buffer 18 43 0a 00 00 29 f2 03 14 00 28 ff 10 00 29 76 27> } +9ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +1ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +12ms
  zigbee2mqtt:debug 2018-9-21 03:22:04 Recieved zigbee message of type 'devChange' with data '{"cid":"msTemperatureMeasurement","data":{"measuredValue":3166}}' of device 'lumi.weather' (0x00xxxxd000201b4xx)
  zigbee2mqtt:debug 2018-9-21 03:22:04 Recieved zigbee message of type 'devChange' with data '{"cid":"msRelativeHumidity","data":{"measuredValue":5931}}' of device 'lumi.weather' (0x00xxxxd000201b4xx)
  serialport:unixRead waiting for readable because of code: EAGAIN +42ms
  serialport:poller Polling for "readable" +85ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +12ms
  zigbee2mqtt:debug 2018-9-21 03:22:04 Recieved zigbee message of type 'attReport' with data '{"cid":"msPressureMeasurement","data":{"16":10102,"20":-1,"measuredValue":1010}}' of device 'lumi.weather' (0x00xxxxd000201b4xx)
  zigbee2mqtt:info 2018-9-21 03:22:04 MQTT publish, topic: 'zigbee2mqtt/0x00xxxxd000201b4xx', payload: '{"humidity":59.31,"linkquality":99,"pressure":1010}'

```

Robertops, 2018-09-21:
Is it the Xiaomi sensor WSDCGQ11LM? I have one and it is woking fine with 0.1.5.

stas-demydiuk, 2018-09-21:
For me it reproducible too, sometimes it shows just humidity, sometimes just pressure, sometimes all values

hoangtuit, 2018-09-21:
My sensor sometines not update for a long time. Temperature, humidity definitely changes during that time (AC turn on)

[image]

dzungpv, 2018-09-21:
Yes, humidity and pressure update but not the temperature. This problem not occurs before

Koenkk, 2018-09-21:
@dzungpv have you set `cache_state` to disabled in `configuration.yaml`?

dzungpv, 2018-09-21:
@Koenkk No, it is enable by default

AnhDuc85, 2018-09-21:
I'm on latest (0.1.5) commit 7d49e13, see my log: https://pastebin.com/raw/cGbHamWx
Seems fine to me.

closed 2018-11-27 by stale[bot]