Koenkk/zigbee2mqtt #396: Removing router, what happens to child devices?

ben423423n32j14e, 2018-09-21:
I understand adding routers increases the maximum number of devices that can ne added to the Zigbee network.

Question is what happens if a router gets turned off or removed? Do the child devices automatically move to another router or does the network break?

ptvoinfo, 2018-09-21:
Devices will migrate to another router or coordinator. Devices always search for a better path to a coordinator. Devices may migrate between near routers.

ben423423n32j14e, 2018-09-21:
What if coordinator and routers are already at max capacity? Just trying to understand how this works. Thanks

ptvoinfo, 2018-09-21:
A device will constantly try to find a path to a coordinator. But unsuccessfully. So, in your case you should add more routers (alternative paths)

ben423423n32j14e, 2018-09-21:
Thanks ptvoinfo, this should be added to the Wiki.

closed 2018-09-21 by ben423423n32j14e

Stefan300381, 2020-05-26:
Hi,

I'd like to bring this old thread back to live. Maybe I wasn't patient enough but I just ran a test and my end devices did NOT move automatically to a different router. What I did:

- one coordinator (has assigned a handful of end devices and one router, router A)
- router A (see last line) that has another handful of end devices connected, xiaomi body sensors.
- router B connected to coordinator, now end devices connected

1. I unplugged router B, which didn't lead to any changes of course
2. I unplugged router A and I could see in the network map, that end devices are now disconnected 
3. I plugged in router B into the same power socket that was used by router A. So I assumed that the lost end devices will now reconnect to router B, which wasn't the case. I triggered the buttons on the body sensors and they showed they were disconnected (LED flashed blue). Even after 30min and more button presses, the situation was still the same.
4. I swapped back router B with router A and now all of the lost devices were back online immediately after I pressed the button on the body sensor.

best regards

Stefan

Stefan300381, 2020-05-26:
@ben423423n32j14e not sure if someone will see my question, while the topic has been closed

ptvoinfo, 2020-05-27:
@Stefan300381 End-devices discover a new path to a coordinator for up to 24 hours. The path is logical and does not depends on a physical location directly. Therefore your short time experiment didn't get you desired results.

Stefan300381, 2020-05-27:
thanks for the explanation. not sure what's shocking me more. a zigbee device like smoke or gas detector that never moves from one router to another, or a devices that might move only after +20hours and stays disconnected during this time :-(

ptvoinfo, 2020-05-27:
If a device sends critical data, it may find a path quickly (it depends on firmware). All alarm systems should avoid wireless data channels as much as possible. Or you should not rely on these sensors.

Gio-dot, 2021-01-10:
> If a device sends critical data, it may find a path quickly (it depends on firmware). All alarm systems should avoid wireless data channels as much as possible. Or you should not rely on these sensors.

Hi, even battery powered xiaomi sensor can search for new paths if original path is broken?

ptvoinfo, 2021-01-10:
Yes, but it takes some time.

rpallai, 2021-07-29:
Seems like Xiaomi Aquara (temperature & humidity / door & window) sensors don't search for new path even after days but when its button has been pressed once. Then the blue led flashes 3 times and the next push will report values with 2 quick blue flashes.

I have to do this from time to time if a smart plug has been unplugged near to an Aqura sensor. This time I waited for 5 days before I pushed the buttons.

rpallai, 2021-07-30:
> Hi, even battery powered xiaomi sensors can search for new paths if original path is broken?

No; seems like battery powered Xiaomi sensors stick they parent until you press the button once on them. Some devices even need a factory reset (looong press).

See:
- https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2527#issuecomment-594165921
- https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2527#issuecomment-594197826

michtam, 2021-08-06:
In my case (WSDCGQ11LM) one short button doesn't help nor power cycle.
Only repairing.