Koenkk/zigbee2mqtt #397: Update complete with 3 moderate severity vulnerabilities

hoangtuit, 2018-09-21:
Hi, I update zigbee2mqtt to last version. I run 'npm install' and it's complete with  3 moderate severity vulnerabilities. I tried 'npm audit fix' but it's not work.

```
753 packages from 815 contributors and audited 1612 packages in 82.267s
found 3 moderate severity vulnerabilities
  run `npm audit fix` to fix them, or `npm audit` for details
```

[image]

Koenkk, 2018-09-21:
These issues are caused by dependencies of zigbee2mqtt -> when dependencies update it's dependencies this will be solved -> not within the scope of zigbee2mqtt.

closed 2018-09-21 by Koenkk

dirkclemens, 2018-09-27:
downgrade to node v v8.11.2, if your version is newer by running nvm (https://github.com/creationix/nvm)