Koenkk/zigbee2mqtt #398: Help setting up a router

tatunts, 2018-09-21:
Hello everyone!

I am having some difficulties setting a secondary CC2531 as a router.

I have a couple of devices (Xiaomi buttons and a cube) a bit far away from my router so I get link quality 0.

I've flashed it successfully wit this [firmware](http://ptvo.info/wp-content/uploads/2018/05/cc2531_1.2.2a.44539_firmware.zip) and managed to pair my coordinator and plugged in between both coordinator and the devices.

The problem is that nothing has changed: still get 0 link quality and the only thing that shows on the log is something related to the cube: 0x00158D00010B2E3E.

Here is my log: https://pastebin.com/iRasDgmJ

I am using it with Hassio and the oficial addon with the latest CC2531 coordinator FW.

Do I need to pair those far away devices directly with the router?

The main reason to extend the range is because sometimes when I press the button a single time, for example, many messages are received (long click), very annoying:

```
2018-9-19 23:33:44 - info: MQTT publish, topic: 'zigbee2mqtt/casa_xiaomi_botao_suite_01', payload: '{"battery":"100.00","voltage":3012,"linkquality":0,"click":"long"}'
2018-9-19 23:33:44 - info: MQTT publish, topic: 'zigbee2mqtt/casa_xiaomi_botao_suite_01', payload: '{"battery":"100.00","voltage":3012,"linkquality":0,"click":"long"}'
2018-9-19 23:33:44 - info: MQTT publish, topic: 'zigbee2mqtt/casa_xiaomi_botao_suite_01', payload: '{"battery":"100.00","voltage":3012,"linkquality":0,"click":"long"}'
2018-9-19 23:33:44 - info: MQTT publish, topic: 'zigbee2mqtt/casa_xiaomi_botao_suite_01', payload: '{"battery":"100.00","voltage":3012,"linkquality":0,"click":"long"}'
2018-9-19 23:33:44 - info: MQTT publish, topic: 'zigbee2mqtt/casa_xiaomi_botao_suite_01', payload: '{"battery":"100.00","voltage":3012,"linkquality":0,"click":"long_release","duration":85}'
2018-9-19 23:33:44 - info: MQTT publish, topic: 'zigbee2mqtt/casa_xiaomi_botao_suite_01', payload: '{"battery":"100.00","voltage":3012,"linkquality":0,"click":"long"}'
2018-9-19 23:33:44 - info: MQTT publish, topic: 'zigbee2mqtt/casa_xiaomi_botao_suite_01', payload: '{"battery":"100.00","voltage":3012,"linkquality":0,"click":"long_release","duration":47}'
```

You can see they are all sent at the same second...

Thanks a lot!

nickrout, 2018-09-22:
This post https://community.home-assistant.io/t/zigbee2mqtt-getting-rid-of-your-proprietary-zigbee-bridges-xiaomi-hue-tradfri/52108/572 suggests that you need to re-pair your device to the router. But I have also seen somewhere that this shouldn't be necessary. Worth a try.

ptvoinfo, 2018-09-22:
Also, an end device need some time to find a new router. The device may sleep most time. So, you need to wake up your device (press a button on it).

tatunts, 2018-09-23:
Thanks @nickrout and @ptvoinfo!

Someone said this on the router instructions:

[image]

I am not sure what to do here... I left the router plugged in for the last couple of days, Here is log: https://pastebin.com/FDnhRcMA

There are still no enhancements on link quality. The router itself shows as 0 quality.

Thanks!

nickrout, 2018-09-23:
Yeah that was the contrary advice I had seen but couldn't find it. Hard to know what is right, but why don't you try pairing to the router and see?

nickrout, 2018-09-23:
See also https://github.com/Koenkk/zigbee2mqtt/issues/402

kaelsaes, 2018-09-24:
I pair all my new devices near the CC2531 coordinator and then I bring them to their final position. They connect automatically to the nearest CC2531 router.

tatunts, 2018-09-25:
Hello again! Can someone that have successfully managed to flash both coordinator and router and made they work together post the links for the correct FW you had used? Thanks a lot!

basmeerman, 2018-09-30:
I used: http://ptvo.info/cc2531-based-router-firmware-136/ specifically CC2531-router-diag-usb.hex version (others didn't seem to work). Currently running a 32 node mesh with 3 CC2531 Routers and 1 CC2531 Coordinator. Seems to be running ok.

basmeerman, 2018-09-30:
Addition: Use the latest version for flashing see: https://github.com/Koenkk/zigbee2mqtt/issues/252#issuecomment-425373408

davekr8, 2018-10-23:
> I solved the tiresome topic "mesh network" with 2 Osram ZigBee Plugs.
> 
> Plug and play out of the box. The plugs paired independently and expanded my mesh network par excellence.

Would you please point out, how you managed to use the Osrams as routers. Do I need some speciall config, a refined ordering in the pairing steps or ...? Which firmware do you use on your coordinator?

I am facing a star of Osrams but end nodes only attaches to the coordinator in my environment. See...

[image]

I apprechiate any assistance.

Koenkk, 2018-10-24:
@davekr8 Xiaomi devices do not connect to the osram router (has something to do with manufacturer ID which Xiaomi forces). You can use the CC2531/CC2530 router from the wiki instead.

kaelsaes, 2018-10-24:
@koenkk

I replaced all (3) my CC2531 routers with 3 Osram plugs and my network (about 20 Xiaomi devices) works perfect - in contrast to earlier with the CC2531.

The Osram plugs and the CC2531 coordinator are distributed over 2 floors - where the Xiaomi devices never ever could reach the coordinator if your statement would be right.

I first plugged in a Osram plug near the CC2531 coordinator. It paired automatically within 20 seconds and a new device was created in FHEM.

Then I plugged in the second Osram plug far away from coordinator and within 20 seconds I had a second device in FHEM.

The same with the third Osram plug far far away from coordinator.

After approximately an hour all my Xiaomi devices worked like a charm with the 3 Osram plugs. I only had to wait and do nothing else.

The Osram plugs build up a ZigBee mesh network which works without any problems with Xiaomi devices.

kaelsaes, 2018-10-24:
@michel72 

Which Osram plug are you using?

I use this one: 
https://www.amazon.de/s/ref=nb_sb_noss_1?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&url=search-alias%3Daps&field-keywords=osram+plug

michel72, 2018-10-24:
Yup, I have the same one. Just plugged it in, pair it to the coordinator and verified if I can switch in on/off. It just doesn't route the zigbee signal to the other devices.

davekr8, 2018-10-24:
> @michel72
> 
> Which Osram plug are you using?
> 
> I use this one:
> 
> https://www.amazon.de/Osram-schaltbare-fernbedienbar-Lichtsteuerung-kompatibel/dp/B074PZLX2P/ref=sr_1_1?ie=UTF8&qid=1540375548&sr=8-1&keywords=osram+plug

yes, all the same.

davekr8, 2018-10-24:
@kaelsaes 
> The Osram plugs build up a ZigBee mesh network which works without any problems with Xiaomi devices.

it would be interesting to know, which firmware you are running on the coordinator.

michel72, 2018-10-24:
From the log : Coordinator firmware version: 'undefined'

I have this coordinator:
https://s.click.aliexpress.com/e/Lm8b86G

and i have the equipment to flash a different firmware if required.

kaelsaes, 2018-10-24:
The firmware version could be the explanation why my ZigBee network works well with the Osram plugs.

A few weeks ago ( only CC2531 coordinator and routers in use) I changed back to an older firmware version after a newer version didn't work well.

kaelsaes, 2018-10-24:
With this firmware version I don't have to repair all my Xiaomi devices after reflashing the coordinator. The devices reappear automatically within an hour.

michel72, 2018-10-24:
Thx, I have a 2530, but i’ll try a few

michel72, 2018-10-24:
Does anybody know what file to use exactly for this hardware as a coordinator? 
http://s.aliexpress.com/AzqiqYj2

davekr8, 2018-10-24:
> Coordinator firmware version: '20180815'

@kaelsaes, i am using the same ... 

```
> zigbee2mqtt@0.1.8 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 2018-10-24 19:36:12 Logging to directory: '/opt/zigbee2mqtt/data/log/2018-10-24.19-36-12'
  zigbee2mqtt:info 2018-10-24 19:36:12 Starting zigbee2mqtt version 0.1.8 (commit #97cc56b)
  zigbee2mqtt:info 2018-10-24 19:36:12 Starting zigbee-shepherd
  zigbee2mqtt:info 2018-10-24 19:36:13 zigbee-shepherd started
  zigbee2mqtt:info 2018-10-24 19:36:13 Coordinator firmware version: '20180815'
  zigbee2mqtt:info 2018-10-24 19:36:13 Currently 0 devices are joined:
```

michel72, 2018-10-24:
Just tried: http://ptvo.info/wp-content/uploads/2018/09/cc2530_1.2.2a.44539_firmware.zip
CC2530ZNP-Pro-Secure_LinkKeyJoin.hex

and in the log:
  zigbee2mqtt:info 2018-10-24 19:59:21 Coordinator firmware version: 'undefined'

michel72, 2018-10-24:
I now flashed this one: https://github.com/Koenkk/Z-Stack-firmware.git
log:
  zigbee2mqtt:info 2018-10-24 20:06:41 Coordinator firmware version: '20182308'

Where can I get this '20180815' version?

davekr8, 2018-10-24:
@kaelsaes, good news, started with a clean database and set up the osrams one after another I got it to work.
[image]
but, pairing many sensors seems to become a very unstable process and takes a lot of time... maybe the interference of pakets in the shared medium is the reason for long pairing process...

davekr8, 2018-10-24:
> Where can I get this '20180815' version?

the one for the cc2530 was built on the 23., see https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/CC2530/bin/CC2530ZNP-Prod.hex

kaelsaes, 2018-10-25:
> For me this is the main reason not to use zigbee2mqtt for "production" It's just not reliable enough for me.

I totally agree.

I would very much welcome that koenkk and friends professionalize ZigBee2MQTT to a reliable system.

In my opinion in the moment there are too many construction sites - new devices, network graphic, bugs, enhancements, ...

I would like to use my Xiaomi devices for security - smoke detector, water sensor, ... but in the moment the system is not really reliable.

The Osram plugs are a great progress for a reliable network but it's more a lucky coincidence because I use an old firmware version of the coordinator.

lolorc, 2018-10-25:
@davekr8 I'm using firmwares from https://github.com/Koenkk/Z-Stack-firmware

I've done nothing special to have my xiaomi/aqara devices being routed by my osram plugs.

I'm using a fixed channel and a fixed panid, but I don't think it has anything to do with my mesh networks apart from the channel that should help with my wireless networks.

I'm not a zigbee expert, when I was using less than 15 devices I found that all devices would communicate through the coordinator directly even with routers physically in between.

But I suspect there could be some issues with the network map feature, sometimes it reports devices connected to my coordinator while they should be routed through one of my cc2531 router. (the cc2531 router does report its childs)

if you look at my network map, osram plug are marked as offline... (prise_*)...
[image]

michel72, 2018-10-25:
I have deleted the configuration and database and started over. It now works. 👍 

How do you create those network maps?

lolorc, 2018-10-25:
@michel72 see https://github.com/Koenkk/zigbee2mqtt/issues/417 and https://github.com/Koenkk/zigbee2mqtt/issues/252

kaelsaes, 2018-10-25:
@lolorc 

In my graph all the devices are lined up in one row.

What am I doing wrong?

davekr8, 2018-10-25:
Usa a different layout, e.g. sfdp. Have a look at the graphviz tool docu (google it) there you find the different methods.

closed 2019-01-01 by stale[bot]

adrianmihalko, 2019-02-04:
Hi,

I flashed my CC2531 with router firmware linked on the docu page:

https://koenkk.github.io/zigbee2mqtt/information/zigbee_network.html
https://github.com/Koenkk/Z-Stack-firmware/tree/master/router/CC2531/bin

router-cc253x-std.hex.

Router is connecting to my network, but end-device doesn't seem to be using it?

[image]

nickrout, 2019-02-04:
Where are these three devices relative to each other?

How long have you left it between putting the router in place and creating this graph?

adrianmihalko, 2019-02-04:
After I click SW1 I see the end device is connecting to the router and router to the coordinator:

[image]

  zigbee2mqtt:info 2019-2-4 21:42:12 MQTT publish: topic 'zigbee2mqtt/0x00124b001936af49', payload '{"state":true,"linkquality":31}'

Before I click on SW1, red is blinking slowly, every 4 sec.:

zigbee2mqtt:info 2019-2-4 21:29:12 MQTT publish: topic 'zigbee2mqtt/0x00124b001936af49', payload '{"state":false,"linkquality":55}'

...and end point device is not connected to the router.

Which is interesting, because in #277 I am reading, that even in this state should be retransmit data:

`Sorry for my late response. I think this is not "Bug". This virtual router state "On/Off" does not matter at all. The router operates and retransmits packets even if the green light is on/off. By default the green light is off because it is very bright. But you may press the button and turn on/off this light if you want. I may try to add a possibility to save the green light state and restore it after power loss. Vote.`

adrianmihalko, 2019-02-04:
```
  zigbee2mqtt:info 2019-2-4 22:40:30 MQTT publish: topic 'zigbee2mqtt/0x00158d0001ab4402', payload '{"temperature":20.85,"linkquality":0,"humidity":59.66,"pressure":1014.5,"battery":74,"voltage":2955}'
  zigbee2mqtt:info 2019-2-4 22:41:15 MQTT publish: topic 'zigbee2mqtt/0x00124b001936af49', payload '{"state":true,"linkquality":23}'
```

After pressing SW1 in CC2531 I see state:true + led is green, I don't know if this matters.

I waited half hour, but EndDevice is still not connecting to router?

[image]