Koenkk/zigbee2mqtt #399: help to supporting Philips Hue Iris

D33pWat3r, 2018-09-22:
I saw in the supported list, that the Iris Light ist not supported.

I tried to connect my Iris and get the message: "Device with modelID 'LLC010' is not supported."

Okay, I fix the devices.js with the values of the Hue Bloom devices [LLC012] 
```
{
        zigbeeModel: ['LLC010'],
        model: '7199960PH',
        vendor: 'Philips',
        description: 'Hue Iris',
        supports: generic.light_onoff_brightness_colorxy().supports,
        fromZigbee: generic.light_onoff_brightness_colorxy().fromZigbee,
        toZigbee: generic.light_onoff_brightness_colorxy().toZigbee,
    },
```

Or in the second try with the values of Hue Go [LLC020]

```
{
        zigbeeModel: ['LLC010'],
        model: '7199960PH',
        vendor: 'Philips',
        description: 'Hue Iris',
        supports: generic.light_onoff_brightness_colortemp_colorxy().supports,
        fromZigbee: generic.light_onoff_brightness_colortemp_colorxy().fromZigbee,
        toZigbee: generic.light_onoff_brightness_colortemp_colorxy().toZigbee,
    },
```
In both cases I connected my Iris with my zigbee2MQTT and see a instance in FHEM. But I can't see the status or change some values... 
I have no idea, how I can control the Iris Ligth

D33pWat3r, 2018-09-22:
thanks for your quick reply. here is my log:
https://pastebin.com/keEimeUf

and I add the following entry in my devices.js:
	{
        zigbeeModel: ['LLC010'],
        model: '7199960PH',
        vendor: 'Philips',
        description: 'Hue Iris',
        supports: 'Light',
        fromZigbee: generic.light_onoff_brightness_colorxy().fromZigbee,
        toZigbee: [],
    },

D33pWat3r, 2018-09-22:
okay, I found in the wiki the entry, how I can control devices with MQTT messages.
with the Adding of:
```
	{
        zigbeeModel: ['LLC010'],
        model: '7199960PH',
        vendor: 'Philips',
        description: 'Hue Iris',
        supports: generic.light_onoff_brightness_colorxy().supports,
        fromZigbee: generic.light_onoff_brightness_colorxy().fromZigbee,
        toZigbee: generic.light_onoff_brightness_colorxy().toZigbee,
    },
```
I can do magic stuff :D .... so it works
only my Plugin for FHEM can't handle this modul :(

closed 2018-09-22 by D33pWat3r

Koenkk, 2018-09-25:
@D33pWat3r thanks, will be supported in the next version of zigbee2mqtt.