Koenkk/zigbee2mqtt #402: "Big" mesh network with Routers + Endpoints and pairing problems

alexbarcelo, 2018-09-22:
I have seen a miscellaneous of related issues in GitHub, but I am missing some information and I fail to see the big picture. Sorry if you consider that it is a dupe, but there is no information on the wiki; #26 seems to contradict the 15 direct children devices limit on the [wiki](../wiki/ZigBee-network), unless there are both limits: 15 direct children, 48 total devices in single zigbee network (is it?).

I have more than 15 total devices, half of them are router-capable devices, and the total of devices in my home may be near the 50 or 60 (I am still designing and installing them atm). I started pairing stuff and could not go any further when I reached the 15 limit. So far, I believe that I can achieve a pairing through the router devices, but I cannot find comprehensive information on how to activate pairing in a router, or if it is a general mechanism allowed by all routers. Questions in my head right now:

 - Can I pair a Xiaomi Aqara button (end device) to a Xiaomi Aqara wall-switch (router)?
 - Can I pair a Xiaomi Aqara Wall-switch (router) to a Xiaomi Aqara wall-switch (router)?
 - If I am able to pair devices through routers, then will I be able to pair up to 15 devices to each router? Up to a total maximum of... [number of devices / number of routers / number of hops]?

And --hoping that I am on the right track-- which are the main steps to achieve all that? I assume that I have to start at the coordinator and "grow" from there, but activating pairing mode seems to only affect the coordinator (CC2531 in my scenario) and not routers (wall-switches in my current tests). Do I need special routers? I would love to minimize (preferably avoid altogether) having multiple coordinators (I have seen that proposal somewhere else). I have no problem in adding routers to my home, if that means achieving a single network (single coordinator) resilient-ish configuration.

ryanbeaton, 2018-09-22:
I think you’ve got a valid question here. I’m still nervous about the limit.

I was hoping for more of an answer in https://github.com/Koenkk/zigbee2mqtt/issues/396

The whole limit in memory of the CC2531 and the scenario where a end device that’s usually through a router has to connect to the coordinator because the router goes offline but the coordinator is full freaks me out 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

evildad, 2018-09-22:
If you have for example a tradfri bulb it acts as a router and you have to do nothing special, just pair the device.
But I have a problem that if the bulb which acts as a router for the Xiaomi humidity/temperature sensor goes down (turned of via the wall switch) it doesn't automatically take an alternate route and therefore no humidity/temperature updates are received.
Has anyone had success with rerouting?

alexbarcelo, 2018-09-23:
The problem was related to Xiaomi Aqara "router" behaviour, which is not a real router. And pairing requests are not routed (neither signals).

Closing in favor of a more specific #404.

Sorry for the noise. It was not easy (as a n00b) to diagnose my initial problem :S

closed 2018-09-23 by alexbarcelo