Koenkk/zigbee2mqtt #403: No battery values reported from Xiaomi door and temperature sensors

rcarmo, 2018-09-22:
I've just installed from git master (and flashed a new dongle), and am trying to figure out why I can't get any battery info off the Xiaomi Aqara devices I just paired. Logs only report contact and link quality for some reason:

```
2018-9-22 16:55:08 - debug: Recieved zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'lumi.sensor_magnet.aq2' (0x00158d0002370064)
2018-9-22 16:55:08 - info: MQTT publish, topic: 'zigbee2mqtt/door0', payload: '{"contact":true,"linkquality":15}'
2018-9-22 16:55:08 - debug: Recieved zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'lumi.sensor_magnet.aq2' (0x00158d0002370064)
2018-9-22 16:55:09 - debug: Recieved zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'lumi.sensor_magnet.aq2' (0x00158d0002370064)
2018-9-22 16:55:09 - info: MQTT publish, topic: 'zigbee2mqtt/door0', payload: '{"contact":false,"linkquality":0}'
2018-9-22 16:55:09 - debug: Recieved zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'lumi.sensor_magnet.aq2' (0x00158d0002370064)
2018-9-22 16:55:09 - debug: Recieved zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'lumi.sensor_magnet.aq2' (0x00158d0002370064)
2018-9-22 16:55:09 - info: MQTT publish, topic: 'zigbee2mqtt/door0', payload: '{"contact":false,"linkquality":0}'
2018-9-22 16:55:11 - debug: Recieved mqtt message on topic 'zigbee2mqtt/bridge/config/devices' with data '1537635311809'
2018-9-22 16:55:11 - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"devices","message":[{"ieeeAddr":"0x00158d0002325aa2","type":"EndDevice","model":"WSDCGQ11LM","friendly_name":"temperature0"},{"ieeeAddr":"0x00158d0002370064","type":"EndDevice","model":"MCCGQ11LM","friendly_name":"door0"}]}'
```

evildad, 2018-09-22:
see here, those values should appear after a few hours runtime:
https://github.com/Koenkk/zigbee2mqtt/issues/392

rcarmo, 2018-09-23:
It did on one sensor. Can’t see it in another even after a day... but will check again.

ciotlosm, 2018-09-27:
@rcarmo Check signal strength. I usually blow hot air once in the temperature sensor to trigger an update and for the door I open and close a few times with a 2 sec pause in between. 

Usually the signal strength might be an issue so have a look a potentially getting 1 router somewhere closer (see CC2531 router in the docs).

rcarmo, 2018-10-01:
Hmmm. Meanwhile one of my door sensors (the one closest to the bridge and which was reporting battery values) has stopped sending open/close updates. I suspect the bridge too.

lolorc, 2018-10-05:
also having issues with battery reporting (the last 24h I only got 6 sensors (of 26) reporting their battery level no more than 3 times). other attributes are reported far more often.

simonvanderveldt, 2018-10-09:
FYI I currently only have a single aqara temperature/humidity/pressure sensor and it started reporting it's battery and voltage values +-50 minutes after it was connected to zigbee2mqtt.

lolorc, 2018-10-09:
my issue was caused by a regression in zigbee-sheperd that has been fixed in dev.
still I have at least 2 sensors that don't update as often as the other xiaomi/aqara ones (a door sensor and a round switch)

bbrendon, 2018-11-17:
I have a xiaomi water leak sensor where the battery status isn't getting into hass. I've noticed the battery value appears in the mqtt logs, but it never makes it into HASS. The temperature sensor battery I have is getting into hass perfectly fine.  I'm running zigbee2mqtt 0.2.

closed 2019-01-25 by stale[bot]