Koenkk/zigbee2mqtt #404: Xiaomi Aqara key wired wall switch (Router) seems to skip router duties

alexbarcelo, 2018-09-23:
My scenario is a small but long apartment in which all wall light switches are

> Xiaomi Aqara single|double key wired wall switch (Router)

with identifier being QBKG04LM and QBKG03LM

https://www.aqara.com/en/ctrl-neutral.html

(although I may have an older revision, the main features and design remain).

Supposedly, those are "Router" devices. But I started pairing devices and was not able to pair anything beyond distance X from the CC2531. Given that it is a long apartment, that was a problem. After a lot of confusion and a little troubleshooting, I tried to hack a temporal tradfri in a place, and verified that by doing this things reach was extended.

Given that I am using switches (no smart lightbulbs involved, nor any chance to use them), I don't have a realistic way to use "natural routers". If Xiaomi routers do not behave as they are expected to, I will end up adding CC253X devices to work as routers. I would love to avoid the hassle. Am I doing something wrong? Is there a way to do that? Is this behaviour of Xiaomi Aqara routers known?

ben423423n32j14e, 2018-09-25:
I know this is an open-source project but I really wish the devs would finish Z-Stack 3 implementation, so many people building networks, it would be good to have them starting out on Z-Stack 3.0

https://github.com/Koenkk/zigbee2mqtt/issues/211

Device limit is greatly improved with Z-Stack 3.0.

ben423423n32j14e, 2018-09-25:
Because you will need to re-add all your devices upon upgrading to Z-Stack 3.0 firmware...

Robertops, 2018-10-04:
I have the same problem. I am planning to build a mesh with about 20 wired switches, and I can't get my QBKG03LM to route anything. 

Found this: http://en.miui.com/thread-2058422-1-1.html

I have one QBKG12LM (with neutral) and will test it soon.

It would be nice if someone with the Xiaomi gateway could test if any wired switch repeats with the original gateway.

Robertops, 2018-10-05:
Just tested and looks like QBKG12LM is routing.

I placed the devices in a line like this:
Raspberry <-> QBKG12LM <-> LED Light

Network map was showing LED light connected to the coordinator with 0 signal, but worked. As soon as I unplugged QBKG12LM, the light stopped working. Plugged it again and the light started working immediately. Did it several times and the light worked just when QBKG12LM was powered. I am almost sure it was routing.

My QBKG03LM is currently in use, but I will try to test it in the same way in some days.

pataha, 2018-11-14:
I can confirm that the QBKG12LM works as a router

alexbarcelo, 2018-11-14:
So it seems that *03* and *04* are not behaving as a router, but *12* is.

I will propose to add this information in the wiki.

closed 2019-01-20 by stale[bot]