Koenkk/zigbee2mqtt #405: Unsupported device - LTW001 \neq LTW010

WagnerBeccard, 2018-09-23:
I've got the Luck to have only unsuported devices laying around. 
on it for a file change Proposal in zigbee-shepherd-converters

while googleing i fund also this issue from another repo
https://github.com/mariusmotea/diyHue/issues/171 

> LTW001 - Hue White Ambiance bulb (color temperature only bulb).
> LTW004 - Another Hue White Ambiance bulb (color temperature only bulb).

maybe add them already like 
`['LTW010', 'LTW001', 'LTW004'],`
I unfortunaly need you for the Home Assistant part 🙈

Koenkk, 2018-09-26:
`LTW004` and `LTW001` are supported now. thanks!

closed 2018-09-26 by Koenkk

WagnerBeccard, 2018-09-26:
Awesome - Fast! a new release 🏭 
looking forward to get rid of the hue Hub