Koenkk/zigbee2mqtt #406: How  ton"remove" previously paired device not working

hanross2323, 2018-09-24:
Hi

I'm trying to pair XIAOMI's single button no-neutral wall switch and I'm getting too frustrated with that. Some days ago, the zigbee2mqtt kind of tried to pair with the device, but something went wrong and the device cannot be controlled nor the bridge can receive any messages.

The more I try to pair it again, the more frustrated I am since the bridge seems not to receive anything. BUT, every time I restart the bridge I get this:

>   zigbee2mqtt:info 2018-9-24 10:41:26 Currently 8 devices are joined:
>   zigbee2mqtt:info 2018-9-24 10:41:26 buto_xiaomi (0x00158d00022b89be): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
>   zigbee2mqtt:info 2018-9-24 10:41:26 sensor_moviment (0x00158d0002246abb): RTCGQ01LM - Xiaomi MiJia human body movement sensor (EndDevice)
>   zigbee2mqtt:info 2018-9-24 10:41:26 Clima (0x00158d00023e50ce): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor  (EndDevice)
>   zigbee2mqtt:info 2018-9-24 10:41:26 porta_principal (0x00158d0001fa789d): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
>   zigbee2mqtt:info 2018-9-24 10:41:26 endoll_xiaomi (0x00158d000211564b): ZNCZ02LM - Xiaomi Mi power plug ZigBee (Router)
>   zigbee2mqtt:info 2018-9-24 10:41:26 hue_remote (0x0017880103cab961): 324131092621 - Philips Hue dimmer switch (EndDevice)
>   zigbee2mqtt:info 2018-9-24 10:41:26 HUE1 (0x0017880102932fd9): 9290012573A - Philips Hue white and color ambiance E26/E27 (Router)
**>   zigbee2mqtt:info 2018-9-24 10:41:26 unknown (0x00158d00023ad922): QBKG04LM - Xiaomi Aqara single key wired wall switch (Router)**
>   zigbee2mqtt:info 2018-9-24 10:41:26 Connecting to MQTT server at mqtt://homeassistant

So my question is, how can I remove this device from db and start it over again? I'm using a hassio addon so I'm not able to ssh-it (unless I could???)

Thanks

57k, 2018-09-24:
Hi,

**zigbee2mqtt/bridge/config/remove** 

via MQTT

Allows you to remove devices from the network. Payload should be the friendly_name, e.g. 0x00158d0001b79111. On successful remove a device_removed message is send.

https://github.com/Koenkk/zigbee2mqtt/wiki/MQTT-topics-and-message-structure

Greetings, Jens

hanross2323, 2018-09-24:
Thank you. I’m using Node-red to build mqtt messages, I send this msg.payload:

> msg.payload = {
> “friendly_name”: “unknown”,
> }
> return msg;
> (or friendly_name: 0x00158d00023ad922)

And I always receive success:

> zigbee2mqtt:info 2018-9-24 14:16:41 Successfully removed {“friendly_name”:“unknown”}
> zigbee2mqtt:info 2018-9-24 14:16:41 MQTT publish, topic: ‘zigbee2mqtt/bridge/log’, payload: ‘{“type”:“device_removed”,“message”:"{“friendly_name”:“unknown”}"}’

But at the end, when I restart the service, device is still there… any ideas??

WagnerBeccard, 2018-09-25:
Yes. Your payload is wrong. 

The topic:
zigbee2mqtt/bridge/config/remove
The payload:
unknown

You try to remove a device which is called "friendly_name : unknown" not unknown

basmeerman, 2018-09-30:
My understanding is there are about 3 places to 'remove' a device:
1) in configuration.yaml
2) in database.db
3) in the coordinator/router eeprom (on chip).

1 and 2 can be managed manually or through the MQTT subject (last test of mine didn't seem to work though). So proceeded to remove items manually.

3 can only be done by reflashing the CC2531 (if you use one).

Kind regards.