Koenkk/zigbee2mqtt #407: Setting scene or multiple lights from HA fails

sknsean, 2018-09-24:
Setting lights one by one goes fine, but all off or setting a  scene, some of the commands does not set a light.

I have attached some logs. (Hope that helps)
https://paste.pound-python.org/show/1CoXwOFcmnbAhy3HlFhJ/

Koenkk, 2018-09-25:
I think this is caused because of too many messages at one time.

sknsean, 2018-09-25:
Yes exactly my thoughts :+1: 
I haven't looked at the code, but is there a FIFO we could and a small delay between the transmissions?
I'm setting 5 bulbs in a scene

Koenkk, 2018-09-25:
Yes, that's the thing that should be added.

keigezellig, 2018-10-01:
I have (maybe) a related problem. I have 10 IKEA bulbs and sometimes when i try to switch them all on/off at the same time through HA, not all bulbs react correctly (some stay on or some stay off)
This sometimes also happens when switching on/off an individual bulb. When i look at the zigbee2mqtt log  i see that the command is sent, but there is no reply that it's turned on/off.
This happens randomly.

kiucheng, 2018-11-16:
I have the same issue with grouping 3 trådfri bulb together. I would say 80% of the time it works but sometime it leaves one bulb untouched. I got around with not using group in HA but adding 1 second delay in the automation. I hope there will be some kind of queuing mechanisms in zigbee2mqtt

Koenkk, 2018-11-16:
@kiucheng this is one of the features of 0.2.0: https://github.com/Koenkk/zigbee2mqtt/releases/tag/0.2.0 (https://github.com/Koenkk/zigbee2mqtt/issues/529)

kiucheng, 2018-11-16:
Thanks! Does it mean this issue can be close now?

Koenkk, 2018-11-16:
Yes, thanks @kiucheng

closed 2018-11-16 by Koenkk

sknsean, 2018-11-16:
Thanks will test