Koenkk/zigbee2mqtt #409: Request: Implement use of ZigPy to support its Zigbee USB dongles

Gamester17, 2018-09-25:
Would it be a good idea for zigbee2mqtt to also implement support for using the ZigPy library too as an option in order to via it support the Zigbee USB dongles (as well as GPIO adapters) it supports?

https://github.com/zigpy/zigpy

ZigPy (Zigbee for Python) is an open source library that currently can communicate with the more comercially available Zigbee dongles such as XBee  based devices, EmberZNet based devices, and Conbee/RaspBee by Dresden-Elektronik. It does this by depending on Python modules which in turn communicates directly with different USB radios native serial protocol (UART) for zigpy. For more on those modules see:

https://github.com/zigpy/bellows
https://github.com/zigpy/zigpy-xbee
https://github.com/Equidamoid/pyconz

Koenkk, 2018-09-25:
The underlying library does not support this because this module is not Z-Stack based.

Adding support for this would require a complete refactor of the [zigbee-shepherd] library and it's underlying libraries (https://github.com/koenkk/zigbee-shepherd) library.

This is not within the scope of zigbee2mqtt and therefore I close it.

closed 2018-09-25 by Koenkk

BuhtigithuB, 2020-01-05:
Hi,

Would it be possible to implemented it at zigbee-shepherd level??

Gamester17, 2020-04-22:
@BuhtigithuB Zigbee2mqtt has migrated to zigbee-herdsman instead of zigbee-shepherd, see:

https://github.com/Koenkk/zigbee-herdsman/issues/150