Koenkk/zigbee2mqtt #410: Yale Lock making trouble

todfox, 2018-09-25:
When I pair an unsupported device, Yale Door Lock, I now get this error.

{"message":"request unsuccess: 16","stack":"Error: request unsuccess: 16\n    at /opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/zdo.js:46:21\n    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:258:13)\n    at Object.onceWrapper (events.js:315:30)\n    at emitOne (events.js:116:13)\n    at CcZnp.emit (events.js:211:7)\n    at CcZnp._mtIncomingDataHdlr (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:360:14)\n    at Immediate.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:336:22)\n    at runCallback (timers.js:810:20)\n    at tryOnImmediate (timers.js:768:5)\n    at processImmediate [as _immediateCallback] (timers.js:745:5)"}

tb-killa, 2018-09-26:
Hi ..
which versions of node and npm are you using? 
Need to get the actual version (including hash) from your zigbee2mqtt instance.
which firmware for C22531 are you using?

todfox, 2018-09-26:
I using the 2530 firmware from here
https://raw.githubusercontent.com/dzungpv/Z-Stack-firmware/master/coordinator/CC2530/bin/CC2530ZNP-Prod.hex

Node version = v8.11.4
Npm = 5.6.0

tb-killa, 2018-09-26:
You use as Coordinator the type CC2530 ?
Please use the official supported firmware from wiki:
https://github.com/Koenkk/zigbee2mqtt/wiki/Supported-sniffer-devices
https://github.com/kirovilya/files/blob/master/CC2530ZNP-Pro-Secure_LinkKeyJoin.hex

Finally test again and report if status is working.

todfox, 2018-09-26:
Yes I have a CC2530 mounted on a development board. When I use the
supported firmware, I get this error when trying to start zigbee2mqtt.

{"message":"request timeout","stack":"Error: request timeout\n    at
CcZnp.<anonymous>
(/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:255:22)\n    at
Object.onceWrapper (events.js:315:30)\n    at emitOne (events.js:116:13)\n
  at CcZnp.emit (events.js:211:7)\n    at Timeout.<anonymous>
(/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:234:18)\n    at
ontimeout (timers.js:498:11)\n    at tryOnTimeout (timers.js:323:5)\n    at
Timer.listOnTimeout (timers.js:290:5)"}

tb-killa, 2018-09-26:
Does this failure only apply if you add this type of device or is the error Generally?

todfox, 2018-09-26:
Generally. I have rolled back my database to at state before I added the
Yale Lock. The program doesn't boot with the supported firmware.

Here is the hardware I am using.
https://www.aliexpress.com/item/ZigBee-CC2530-Sensor-Node-Baseboard-Functional-Module-Expansion-Board-USB-Port-24MHz-256KB/32767498002.html
and
https://www.aliexpress.com/item/CC2530-Zigbee-UART-Wireless-Core-Board-Development-Board-CC2530F256-Serial-Port-Wireless-Module-2-4GHz/32767470382.html

tb-killa, 2018-09-26:
@Koenkk do you could help?

Koenkk, 2018-09-27:
Could you reflash your device?

Koenkk, 2018-09-28:
So do you still have this problem with a fresh CC2530 and a clean database.db file?

todfox, 2018-10-01:
Start with a clean database? That's painful. I won't be able to try that
until I am back from traveling in three weeks.

closed 2018-12-07 by stale[bot]

CanuckianOz, 2019-02-06:
> When I pair an unsupported device, Yale Door Lock, I now get this error.
> 
> ```
> {"message":"request unsuccess: 16","stack":"Error: request unsuccess: 16
>     at /opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/zdo.js:46:21
>     at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:258:13)
>     at Object.onceWrapper (events.js:315:30)
>     at emitOne (events.js:116:13)
>     at CcZnp.emit (events.js:211:7)
>     at CcZnp._mtIncomingDataHdlr (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:360:14)
>     at Immediate.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:336:22)
>     at runCallback (timers.js:810:20)
>     at tryOnImmediate (timers.js:768:5)
>     at processImmediate [as _immediateCallback] (timers.js:745:5)"}
> ```

Hi todfox, 

Did you get anywhere with the Yale Lock? I can't get it to post any activity in the Zigbee2MQTT log when I lock or unlock it, so I can't begin to troubleshoot. Would be interested if you managed to get it to work.

todfox, 2019-02-06:
Hi @CanukianOz. Sadly I haven't had the time to look into it. It is on my
long list of things to do. If you figure it out please let me know.