Koenkk/zigbee2mqtt #411: Repairing a device with the old Hue hub brakes Zigbee2mqtt

WagnerBeccard, 2018-09-26:
The hue motion sensor is getting mad without proper configuration. 
so I thourght to repair it to the Hue hub wich is still in use because of some unsupported lights. 
[image] 
Waay to many false positives.

So removed them via MQTT and tryed to repair them.. 
Zigbee2mqtt always rejoind them to their network even "permit_join: false" is set 
So turned the Pi off and reconfigured the sensors with the Hub 
Getting power back and.. Deam - The Lights arnt controlable anymore - 
Go to the logs and 

` zigbee2mqtt:info 2018-9-26 05:26:57 MQTT publish, topic: 'zigbee2mqtt/0x00178801020015ab
`
Wich is the hue motion sensor wich I thrown out of the network... 
https://pastebin.com/8j3XAAS6

Deam.. 
I cant remove the devices via MQTT because the service failes to start..

Koenkk, 2018-09-26:
This is indeed a bug.

Can you try adding to your `configuration.yaml` -> `devices`:

```yaml
"0x00178801020015ab":
  retain: false
  friendly_name: "0x00178801020015ab"
```

WagnerBeccard, 2018-09-26:
Oh I'm happy that you are here =D - Kind of stupid to make every device "smart" in this situation
Its already in the configuration. 

https://pastebin.com/iunnE898

Removed the 2 before shutting the pi down - and configured the hue bridge - At startup theyre were back in the configuration

lolorc, 2018-10-17:
```
a=$(date +"%Y%m%d%H%M%S")
cp data/database.db data/database.db-$a
grep -i -v  0x00178801020015ab data/database.db-$a > data/database.db
```

closed 2018-12-23 by stale[bot]