Koenkk/zigbee2mqtt #414: Cannot reset CC2531 remotely

rcarmo, 2018-09-27:
I have been having some weird crashes I'm assuming are the result of the soft reset mechanism triggering. However, when I restart the daemon, I get this:
```
2018-9-27 20:21:57 - error: Press the reset button on the stick (the one closest to the USB) and start again
2018-9-27 20:21:57 - error: Failed to start Error: request timeout
    at CcZnp.<anonymous> (/home/user/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:255:22)
    at Object.onceWrapper (events.js:315:30)
    at emitOne (events.js:116:13)
    at CcZnp.emit (events.js:211:7)
    at Timeout.<anonymous> (/home/user/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:234:18)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5)
```

Since I have the dongle inside a case, the most effective way to fix it is walk up to the machine and unplug it. Is there any documented way to force a reset?

Koenkk, 2018-09-28:
This soft reset brings the CC2531 in a strange situation. 

For now I would recommend the fix from: https://github.com/Koenkk/zigbee2mqtt/issues/274#issuecomment-412591653

Soon I will provide an option to complete disable this soft reset behavior.

rcarmo, 2018-10-01:
I'm having some erratic behaviour (sensors dropping out of the network and failing to update until re-paired), and keep wondering if this has something to do with it.

lolorc, 2018-10-02:
@rcarmo could you check your logs and see if the panId is still the same. I'm having that kind of issues here, and this is because the panId has changed.

rcarmo, 2018-12-01:
Logs show no changes... it keeps happening.

Morphy99, 2019-01-12:
I'm having this issue after rebooting my RPi 3 running HASS.IO. The dongle lights up after boot then goes off. I've tried pressing the reset button which lights the LED again however it still won't connect even after restarting the zigbee2mqtt addon.  In fact it then goes to another state where even pressing the reset button does nothing.

To resolve I had to power off

`[Info] Configuration file found. Will overwrite configurable               fields with values from add-on configuration
[Info] Configuration written to /share/zigbee2mqtt/configuration.yaml
2019-01-12T10:53:25: PM2 log: Launching in no daemon mode
2019-01-12T10:53:26: PM2 log: App [npm:0] starting in -fork mode-
2019-01-12T10:53:26: PM2 log: App [npm:0] online
> zigbee2mqtt@1.0.1 start /app
> node index.js
  zigbee2mqtt:info 2019-1-12 10:53:36 Logging to directory: '/share/zigbee2mqtt/log/2019-01-12.10-53-36'
  zigbee2mqtt:info 2019-1-12 10:53:38 Starting zigbee2mqtt version 1.0.1 (commit #9191788)
  zigbee2mqtt:info 2019-1-12 10:53:38 Starting zigbee-shepherd
  zigbee2mqtt:info 2019-1-12 10:53:41 Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
  zigbee2mqtt:info 2019-1-12 10:54:42 Starting zigbee-shepherd
  zigbee2mqtt:error 2019-1-12 10:54:45 Error while starting zigbee-shepherd!
  zigbee2mqtt:error 2019-1-12 10:54:45 Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 2019-1-12 10:54:45 Failed to start
	{"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/app/node_modules/cc-znp/lib/ccznp.js:255:22)\n    at Object.onceWrapper (events.js:315:30)\n    at emitOne (events.js:116:13)\n    at CcZnp.emit (events.js:211:7)\n    at Timeout.<anonymous> (/app/node_modules/cc-znp/lib/ccznp.js:234:18)\n    at ontimeout (timers.js:498:11)\n    at tryOnTimeout (timers.js:323:5)\n    at Timer.listOnTimeout (timers.js:290:5)"}`

rcarmo, 2019-01-13:
FYI, I'm still having the same issue as initially reported.

Koenkk, 2019-01-13:
@rcarmo when does this happen?

peel, 2019-01-15:
Facing the same behaviour both with `npm start` as well as via docker whenever trying to start the process.
Flashed with the board with the latest available firmware, tried unplugging, replugging, reflashing, reset button, yet the failure persists. 
I'm running NixOS:

Full log
```
[nix-shell:/tmp/zigbee2mqtt]# DEBUG=* npm start

> zigbee2mqtt@1.0.1 start /tmp/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 1/15/2019, 4:52:50 PM Logging to directory: '/tmp/zigbee2mqtt/data/log/2019-01-15.16-52-50'
  serialport:binding:auto-detect loading LinuxBinding +0ms
  zigbee2mqtt:debug 1/15/2019, 4:52:50 PM Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11]},"dbPath":"/tmp/zigbee2mqtt/data/database.db","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee2mqtt:debug 1/15/2019, 4:52:50 PM Loaded state from file /tmp/zigbee2mqtt/data/state.json
  zigbee2mqtt:info 1/15/2019, 4:52:50 PM Starting zigbee2mqtt version 1.0.1 (commit #9380bbc)
  zigbee2mqtt:info 1/15/2019, 4:52:50 PM Starting zigbee-shepherd
  zigbee-shepherd:init zigbee-shepherd booting... +0ms
  serialport:main opening path: /dev/ttyACM0 +0ms
  serialport:bindings open +0ms
  serialport:main _read queueing _read for after open +0ms
  serialport:poller Creating poller +0ms
  serialport:main opened path: /dev/ttyACM0 +35s
  cc-znp The serialport /dev/ttyACM0 is opened. +35s
  serialport:main _read reading +2ms
  serialport:bindings read +35s
  serialport:unixRead Starting read +0ms
  zigbee-shepherd:request REQ --> SYS:osalNvRead +4ms
  cc-znp:SREQ --> SYS:osalNvRead, { id: 3840, offset: 0, len: 1, value: [ 85 ] } +0ms
  serialport:main _write 8 bytes of data +5ms
  serialport:bindings write 8 bytes +4ms
  serialport:unixWrite Starting write 8 bytes offset 0 bytesToWrite 8 +0ms
  serialport:unixRead waiting for readable because of code: EAGAIN +3ms
  serialport:poller Polling for "readable" +8ms
  serialport:unixWrite write returned null 8 +0ms
  serialport:unixWrite wrote 8 bytes +0ms
  serialport:unixWrite Finished writing 8 bytes +0ms
  serialport:main binding.write write finished +0ms
  cc-znp:SRSP <-- SYS:osalNvRead, __timeout__ +3s
  zigbee-shepherd:request RSP <-- SYS:osalNvRead +1ms
  zigbee-shepherd:init Coordinator initialize had an error: Error: request timeout
    at CcZnp.<anonymous> (/tmp/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:255:22)
    at Object.onceWrapper (events.js:273:13)
    at CcZnp.emit (events.js:182:13)
    at Timeout.<anonymous> (/tmp/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:234:18)
    at ontimeout (timers.js:424:11)
    at tryOnTimeout (timers.js:288:5)
    at listOnTimeout (timers.js:251:5)
    at Timer.processTimers (timers.js:211:10) +1ms
  zigbee2mqtt:info 1/15/2019, 4:53:28 PM Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
  serialport:main #flush +3s
  serialport:bindings flush +3s
  serialport:main binding.flush finished +0ms
  serialport:main #close +0ms
  serialport:bindings close +1ms
  serialport:poller Stopping poller +3s
  serialport:poller Destroying poller +0ms
  serialport:main binding.read error { Error: Canceled
    at Poller.emitCanceled (/tmp/zigbee2mqtt/node_modules/serialport/lib/bindings/poller.js:102:17)
    at Poller.stop (/tmp/zigbee2mqtt/node_modules/serialport/lib/bindings/poller.js:92:10)
    at close.then (/tmp/zigbee2mqtt/node_modules/serialport/lib/bindings/linux.js:49:21) canceled: true } +1ms
  serialport:main _read queueing _read for after open +0ms
  serialport:main binding.close finished +35s
  cc-znp The serialport /dev/ttyACM0 is closed. +35s
  zigbee2mqtt:info 1/15/2019, 4:54:28 PM Starting zigbee-shepherd
  zigbee-shepherd:init zigbee-shepherd booting... +25s
  serialport:main opening path: /dev/ttyACM0 +25s
  serialport:bindings open +1m
  serialport:main _read queueing _read for after open +0ms
  serialport:poller Creating poller +1m
[... 7 lines cut ...]
  serialport:main _write 8 bytes of data +3ms
  serialport:bindings write 8 bytes +3ms
  serialport:unixWrite Starting write 8 bytes offset 0 bytesToWrite 8 +1m
  serialport:unixRead waiting for readable because of code: EAGAIN +3ms
  serialport:poller Polling for "readable" +5ms
  serialport:unixWrite write returned null 8 +1ms
  serialport:unixWrite wrote 8 bytes +0ms
  serialport:unixWrite Finished writing 8 bytes +0ms
  serialport:main binding.write write finished +2ms
  cc-znp:SRSP <-- SYS:osalNvRead, __timeout__ +3s
  zigbee-shepherd:request RSP <-- SYS:osalNvRead +0ms
  zigbee-shepherd:init Coordinator initialize had an error: Error: request timeout
    at CcZnp.<anonymous> (/tmp/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:255:22)
    at Object.onceWrapper (events.js:273:13)
    at CcZnp.emit (events.js:182:13)
    at Timeout.<anonymous> (/tmp/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:234:18)
    at ontimeout (timers.js:424:11)
    at tryOnTimeout (timers.js:288:5)
    at listOnTimeout (timers.js:251:5)
    at Timer.processTimers (timers.js:211:10) +2ms
  zigbee2mqtt:error 1/15/2019, 4:54:36 PM Error while starting zigbee-shepherd!
  zigbee2mqtt:error 1/15/2019, 4:54:36 PM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 1/15/2019, 4:54:36 PM Failed to start
	{"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/tmp/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:255:22)\n    at Object.onceWrapper (events.js:273:13)\n    at CcZnp.emit (events.js:182:13)\n    at Timeout.<anonymous> (/tmp/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:234:18)\n    at ontimeout (timers.js:424:11)\n    at tryOnTimeout (timers.js:288:5)\n    at listOnTimeout (timers.js:251:5)\n    at Timer.processTimers (timers.js:211:10)"}
```

msadeghz, 2019-01-19:
I got this problem.
but when I removed the IEEE address of the dongle in configuration.yaml, it goes OK.
but in the next time when I want remove the IEEE address from the file, I can't find it.

msadeghz, 2019-01-19:
> So I swapped the USB port and it seems to work now but after rebooting the same issue happens again and I have to shutdown and remove.
> Now none of my devices are registered 👎
> EDIT: Panic over, its just a couple of IKEA tradfri bulbs which have changed ID's for some reason.

I got this problem.
but when I removed the IEEE address of the dongle in configuration.yaml, it goes OK.
but in the next time when I want remove the IEEE address from the file, I can't find it.
edit: so, how can I force zigbee2mqtt to forget the dongle? I remove database and IEEE addresses from configuration.yaml in data directory, but the issue is not solved.

msadeghz, 2019-01-20:
> I got this problem.
> but when I removed the IEEE address of the dongle in configuration.yaml, it goes OK.
> but in the next time when I want remove the IEEE address from the file, I can't find it. [...]

today, I've run the firmware and the problem is not seen!! I don't know the source of the issue.

rcarmo, 2019-03-27:
This is still happening on occasion for me.

Koenkk, 2019-03-28:
@rcarmo does it still happen with the latest dev and max stability firmware? https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator

rcarmo, 2019-03-28:
I will have to update my firmware - I see the last update was 8 days ago, mine was updated a month or so ago.

closed 2019-06-03 by stale[bot]

jl200200, 2020-10-26:
Im having the same problem. After reboot CC2531 is unable to start. When i unplug it and change the usb port Zigbee-Herdsman starts fine.

What can I do? I cant unplug and plug my cc2531 after every reboot.

Thanks.

ybatuto, 2021-03-28:
I suspect it may be a hardware issue of cc2531. I got the same stuff. The stick worked ok for a few weeks. Then it turned out that it could not initialize itself after restarting the server. Then the stick started to turn off spontaneously just after 2-3 hours of operation.
So, I ordered cc2538 and cc2562 to test.

iotinkognito, 2021-07-27:
Having the same issue. Works find until I try to turn on permit and then it says adapter removed...