Koenkk/zigbee2mqtt #417: Visualise ZigBee network

57k, 2018-09-28:
Hi everybody,
in order to understand my current Zigbee Network more easily, I would like to use a simple software, that can visualize my network. Which tools are you using?
Thanks in advance
Jens

57k, 2018-09-28:
just publish via mqtt 

zigbee2mqtt/bridge/networkmap

the value

graphviz

at the same time subscribe to 

zigbee2mqtt/bridge/networkmap/graphviz

my next problem is that I do not get all attached devices.

kimonm, 2018-09-28:
I think it'd be easier to read the graphs if the friendly names were used as labels (vs. the numerical code). Does it make sense?

thomasvt1, 2018-09-29:
> I think it'd be easier to read the graphs if the friendly names were used as labels (vs. the numerical code). Does it make sense?

https://github.com/Koenkk/zigbee2mqtt/commit/845a5cdce89c69dbeb3e1303f7d9496c6e58f295

;)

Koenkk, 2018-09-29:
@kimonm it's already in the dev branch (will be in new zigbee2mqtt release)

hoangtuit, 2018-10-05:
Hi, 
I used this command:

`mosquitto_sub -h $MQTT_SRV -C 1 -t zigbee2mqtt/bridge/networkmap/graphviz | sfdp -Tpng > /tmp/map.$(date +%Y%m%d%H%M%S).png`

It's created a image file in tmp folder, but nothing in there (0 KB).

aidbish, 2018-10-05:
> just publish via mqtt
> 
> zigbee2mqtt/bridge/networkmap
> 
> the value
> 
> graphviz
> 
> at the same time subscribe to
> 
> zigbee2mqtt/bridge/networkmap/graphviz
> 
> my next problem is that I do not get all attached devices.

tried this, but in logfile i get
zigbee2mqtt:warn 2018-10-5 18:05:50 Cannot handle MQTT message with topic 'zigbee2mqtt/bridge/networkmap' and message 'graphviz

do i need to install anything extra, i am using the latest-dev docker image

lolorc, 2018-10-05:
@hoangtuit does the command return immediatly ? you don't have to ctrl-c it do you ?
what's the result of mosquitto_sub -h $MQTT_SRV -C 1 -t zigbee2mqtt/bridge/networkmap/graphviz
when you send the graphviz command ?
it should output graphviz text data that you could either use with graphviz commands or 
http://www.webgraphviz.com/

@aidbish looks like your not running the latest version (1.0.6 atm) sorry but I'm clueless about docker.

aidbish, 2018-10-05:
it was the formatting of my publish command that was incorrect
i now can see a scan starting in the logs, but it finishes within 1 second and nothing is shown if i subscribe to the topic

i have tried the following sub commands

mosquitto_sub -h 10.0.0.18 -C 1 -t zigbee2mqtt/bridge/networkmap/graphviz

mosquitto_sub -h 10.0.0.18 -C 1 -t zigbee2mqtt/bridge/networkmap/graphviz | sfdp -Tpng | display -

i have to ctrl-c as they never seem to complete

hoangtuit, 2018-10-05:
@lolorc the command don't return. When i run  graphviz command

mosquitto_sub -h $MQTT_SRV -C 1 -t zigbee2mqtt/bridge/networkmap/graphviz

Nothing to return. I have 1 coordinator and 10 devices in my network

lolorc, 2018-10-05:
Are you both using dev with the latest commits ?
if so, a bug has be introduced with the latest network map change (display friendlyname).
You might want to check my PR to see if it fixes it https://github.com/Koenkk/zigbee2mqtt/pull/443
or you can also edit https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/extension/networkMap.js#L41
and replace
```
const friendlyName = settings.getDevice(item.ieeeAddr).friendly_name;
```
by
```
const friendlyName = settings.getDevice(item.ieeeAddr) ? settings.getDevice(item.ieeeAddr).friendly_name : "";
```

hoangtuit, 2018-10-05:
I did. But nothing happen.

lolorc, 2018-10-05:
dunno what you did :-) but you might want to join some debug log.

hoangtuit, 2018-10-05:
My stupid, It worked. Tks @lolorc

closed 2018-12-14 by stale[bot]

h4nc, 2018-12-23:
worked great with http://www.webgraphviz.com.

Is there a way to visualize this directly in home assistant. Maybe someone here  already did this?

hassio; zigbee2mqtt addon

Koenkk, 2018-12-24:
yes, see https://github.com/Koenkk/zigbee2mqtt/pull/732

RaveGun, 2019-01-08:
Hi, I have ZigBee2Mqtt running as a docker container on a RPi that is running the latest DietPie.

If I request the raw map I get an answer with some devices. I do not know if it's correct as I am not aware what are all the devices I have anyway.

If I try the graphviz mode I am not getting any answer. The docker logs are also not showing any reaction to the published topic. After this point there is also no response to a raw request anymore. I must restart the docker container to get raw working again.

I did install graphviz with: sudo apt install graphviz but there is still no response.

Does the docker container has the rights to run the graphviz.

Shall I switch to a normal installation as right now the container doesn't start automatically after a reboot.

Thank you.

RaveGun, 2019-01-09:
Nevermind, I was running the release branch, not the Dev flavour of the docker container. Now it's working and the custom ipanel component in HA too.

epicurean70, 2019-06-13:
> Nevermind, I was running the release branch, not the Dev flavour of the docker container. Now it's working and the custom ipanel component in HA too.

HI,Did you also install graphviz as a docker? If so , can you explain how the setup it up to show the zigbee2mqtt network map?

thomasvt1, 2019-06-13:
> > Nevermind, I was running the release branch, not the Dev flavour of the docker container. Now it's working and the custom ipanel component in HA too.
> 
> HI,Did you also install graphviz as a docker? If so , can you explain how the setup it up to show the zigbee2mqtt network map?

Try using [webgraphviz](http://www.webgraphviz.com/) then you don't have to install anything

epicurean70, 2019-06-13:
> Try using [webgraphviz](http://www.webgraphviz.com/) then you don't have to install anything

ok. Then how do you see your own network? what data do you put in that box?

thomasvt1, 2019-06-13:
> ok. Then how do you see your own network? what data do you put in that box?

My suggestion is to read the documentation;
https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html

epicurean70, 2019-06-13:
Sorry, real newbie here. 
What exactly should I be looking for in that documentation? Are those mqtt messages for me to publish,, and view in mqtt fx?