Koenkk/zigbee2mqtt #418: New device: Busch Jaeger RM01 ZLL Switch

WinSCaP, 2018-09-28:
Unsupported device: https://www.busch-jaeger-catalogue.com/2CKA006730A0070,artikel.html

No clue about how to continu.

I've added this to devices.js:

`    // Busch Jaeger
    {
        zigbeeModel: ['RM01'],
        model: 'RM01',
        vendor: 'Busch Jaeger',
        description: 'Busch Jaeger 6735-81',
        supports: 'click',
        fromZigbee: [],
        toZigbee: [],
    },
`

Output from cat database.db | grep RM01
`{"id":25,"type":"Router","ieeeAddr":"0xd85def11a1001e0d","nwkAddr":11997,"manufId":4398,"manufName":"Busch-Jaeger","powerSource":"Mains (single phase)","modelId":"RM01","epList":[10,18],"status":"online","joinTime":1538127902,"endpoints":{"10":{"profId":49246,"epId":10,"devId":2064,"inClusterList":[0,4096],"outClusterList":[3,4,5,6,8,25,768,4096],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":1,"stackVersion":1,"hwVersion":0,"manufacturerName":"Busch-Jaeger","modelId":"RM01","dateCode":"20161209","powerSource":1,"swBuildId":"1.2.0"}},"genIdentify":{"dir":{"value":2},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"lightingColorCtrl":{"dir":{"value":2},"attrs":{}},"lightLink":{"dir":{"value":3},"attrs":{}}}},"18":{"profId":49246,"epId":18,"devId":256,"inClusterList":[0,3,4,5,6,8],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":1,"stackVersion":1,"hwVersion":0,"manufacturerName":"Busch-Jaeger","modelId":"RM01","dateCode":"20161209","powerSource":1,"swBuildId":"1.2.0"}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genGroups":{"dir":{"value":1},"attrs":{"nameSupport":0}},"genScenes":{"dir":{"value":1},"attrs":{"count":0,"currentScene":0,"currentGroup":0,"sceneValid":1,"nameSupport":0}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":0,"globalSceneCtrl":1,"onTime":0,"offWaitTime":0}},"genLevelCtrl":{"dir":{"value":1},"attrs":{"currentLevel":254,"remainingTime":0}}}}},"_id":"W6zjxl2SfmkgLN5q"}
{"id":25,"type":"Router","ieeeAddr":"0xd85def11a1001e0d","nwkAddr":11997,"manufId":4398,"manufName":"Busch-Jaeger","powerSource":"Mains (single phase)","modelId":"RM01","epList":[10,18],"status":"offline","joinTime":null,"endpoints":{"10":{"profId":49246,"epId":10,"devId":2064,"inClusterList":[0,4096],"outClusterList":[3,4,5,6,8,25,768,4096],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":1,"stackVersion":1,"hwVersion":0,"manufacturerName":"Busch-Jaeger","modelId":"RM01","dateCode":"20161209","powerSource":1,"swBuildId":"1.2.0"}},"genIdentify":{"dir":{"value":2},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"lightingColorCtrl":{"dir":{"value":2},"attrs":{}},"lightLink":{"dir":{"value":3},"attrs":{}}}},"18":{"profId":49246,"epId":18,"devId":256,"inClusterList":[0,3,4,5,6,8],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":1,"stackVersion":1,"hwVersion":0,"manufacturerName":"Busch-Jaeger","modelId":"RM01","dateCode":"20161209","powerSource":1,"swBuildId":"1.2.0"}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genGroups":{"dir":{"value":1},"attrs":{"nameSupport":0}},"genScenes":{"dir":{"value":1},"attrs":{"count":0,"currentScene":0,"currentGroup":0,"sceneValid":1,"nameSupport":0}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":0,"globalSceneCtrl":1,"onTime":0,"offWaitTime":0}},"genLevelCtrl":{"dir":{"value":1},"attrs":{"currentLevel":254,"remainingTime":0}}}}},"_id":"W6zjxl2SfmkgLN5q"}
{"id":25,"type":"Router","ieeeAddr":"0xd85def11a1001e0d","nwkAddr":11997,"manufId":4398,"manufName":"Busch-Jaeger","powerSource":"Mains (single phase)","modelId":"RM01","epList":[10,18],"status":"online","joinTime":1538128317,"endpoints":{"10":{"profId":49246,"epId":10,"devId":2064,"inClusterList":[0,4096],"outClusterList":[3,4,5,6,8,25,768,4096],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":1,"stackVersion":1,"hwVersion":0,"manufacturerName":"Busch-Jaeger","modelId":"RM01","dateCode":"20161209","powerSource":1,"swBuildId":"1.2.0"}},"genIdentify":{"dir":{"value":2},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"lightingColorCtrl":{"dir":{"value":2},"attrs":{}},"lightLink":{"dir":{"value":3},"attrs":{}}}},"18":{"profId":49246,"epId":18,"devId":256,"inClusterList":[0,3,4,5,6,8],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":1,"stackVersion":1,"hwVersion":0,"manufacturerName":"Busch-Jaeger","modelId":"RM01","dateCode":"20161209","powerSource":1,"swBuildId":"1.2.0"}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genGroups":{"dir":{"value":1},"attrs":{"nameSupport":0}},"genScenes":{"dir":{"value":1},"attrs":{"count":0,"currentScene":0,"currentGroup":0,"sceneValid":1,"nameSupport":0}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":0,"globalSceneCtrl":1,"onTime":0,"offWaitTime":0}},"genLevelCtrl":{"dir":{"value":1},"attrs":{"currentLevel":254,"remainingTime":0}}}}},"_id":"W6zjxl2SfmkgLN5q"}
{"id":25,"type":"Router","ieeeAddr":"0xd85def11a1001e0d","nwkAddr":11997,"manufId":4398,"manufName":"Busch-Jaeger","powerSource":"Mains (single phase)","modelId":"RM01","epList":[10,18],"status":"online","joinTime":1538128317,"endpoints":{"10":{"profId":49246,"epId":10,"devId":2064,"inClusterList":[0,4096],"outClusterList":[3,4,5,6,8,25,768,4096],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":1,"stackVersion":1,"hwVersion":0,"manufacturerName":"Busch-Jaeger","modelId":"RM01","dateCode":"20161209","powerSource":1,"swBuildId":"1.2.0"}},"genIdentify":{"dir":{"value":2},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"lightingColorCtrl":{"dir":{"value":2},"attrs":{}},"lightLink":{"dir":{"value":3},"attrs":{}}}},"18":{"profId":49246,"epId":18,"devId":256,"inClusterList":[0,3,4,5,6,8],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":1,"stackVersion":1,"hwVersion":0,"manufacturerName":"Busch-Jaeger","modelId":"RM01","dateCode":"20161209","powerSource":1,"swBuildId":"1.2.0"}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genGroups":{"dir":{"value":1},"attrs":{"nameSupport":0}},"genScenes":{"dir":{"value":1},"attrs":{"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":0,"globalSceneCtrl":1,"onTime":0,"offWaitTime":0}},"genLevelCtrl":{"dir":{"value":1},"attrs":{"currentLevel":254,"remainingTime":0}}}}},"_id":"W6zjxl2SfmkgLN5q"}
{"id":25,"type":"Router","ieeeAddr":"0xd85def11a1001e0d","nwkAddr":11997,"manufId":4398,"manufName":"Busch-Jaeger","powerSource":"Mains (single phase)","modelId":"RM01","epList":[10,18],"status":"online","joinTime":1538128346,"endpoints":{"10":{"profId":49246,"epId":10,"devId":2064,"inClusterList":[0,4096],"outClusterList":[3,4,5,6,8,25,768,4096],"clusters":{}},"18":{"profId":49246,"epId":18,"devId":256,"inClusterList":[0,3,4,5,6,8],"outClusterList":[],"clusters":{}}},"_id":"otcvn2dOW9RmMfLw"}
{"id":25,"type":"Router","ieeeAddr":"0xd85def11a1001e0d","nwkAddr":11997,"manufId":4398,"manufName":"Busch-Jaeger","powerSource":"Mains (single phase)","modelId":"RM01","epList":[10,18],"status":"online","joinTime":1538128346,"endpoints":{"10":{"profId":49246,"epId":10,"devId":2064,"inClusterList":[0,4096],"outClusterList":[3,4,5,6,8,25,768,4096],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":1,"stackVersion":1,"hwVersion":0,"manufacturerName":"Busch-Jaeger","modelId":"RM01","dateCode":"20161209","powerSource":1,"swBuildId":"1.2.0"}},"genIdentify":{"dir":{"value":2},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"genOta":{"dir":{"value":2}," [... cut ...] ode":"20161209","powerSource":1,"swBuildId":"1.2.0"}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genGroups":{"dir":{"value":1},"attrs":{"nameSupport":0}},"genScenes":{"dir":{"value":1},"attrs":{"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":0,"globalSceneCtrl":1,"onTime":0,"offWaitTime":0}},"genLevelCtrl":{"dir":{"value":1},"attrs":{"currentLevel":254,"remainingTime":0}}}}},"_id":"otcvn2dOW9RmMfLw"}
`

tb-killa, 2018-09-28:
Hi your Device seems to be a Light link device. Dont know if this will work.
But we could check if it work by usingen binding to specific Clusters.
What you need to know, the Dynamic Button mapping like Busch Jaeger allow will not work.
I will provide some sample code for first Tests later.

tb-killa, 2018-09-28:
Hi @WinSCaP please test this and report logfile with ``DEBUG=zigbee-shepherd* npm start`` ..
it should give you button with on/off and levelctrl (touch / hold / release after hold)
Maybe we could check again with another cluster binding on another endpoint but lets first check what this report :)

```
{
        zigbeeModel: ['RM01'],
        model: 'RM01',
        vendor: 'Busch Jaeger',
        description: 'Busch Jaeger 6735-81',
        supports: '',
        fromZigbee: [],
        toZigbee: [tz.factory_reset],
            configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const ep18 = shepherd.find(ieeeAddr, 18);
			
            const actions = [
                (cb) => ep18.bind('genOnOff', coordinator, cb),
                (cb) => ep18.bind('genLevelCtrl', coordinator, cb),
				
            ];
             execute(ep18, actions, callback);
        },
},
```

WinSCaP, 2018-09-30:
I currently seem to have b0rked it. I will remove the switch from the wall and try again next to the controller, re-adding via multiple routers is not working. Also I can't get a good logfile, the debug messages are not saved, not even when using `DEBUG=zigbee-shepherd* npm start |& tee -a debuglog.txt`

tb-killa, 2018-09-30:
Hi i use this for my personal debugging:
```
DEBUG=zigbee-shepherd* npm start 3> data/"$(date +"%Y_%m_%d_%I_%M")_debug.log"
```
maybe it help ;) ..

WinSCaP, 2018-10-04:
Finally some time. The device is: 0xd85def11a1001e0d
Logs are here: https://pastebin.com/3LTP85DK

WinSCaP, 2018-10-04:
I also added the devices to my Hue bridge, it shows up as dimmable light and is controllable.

Akkunasenbohrer, 2018-11-05:
There is far more magic with this Busch Jäger devices:
They act as routers and are able to scan the network and can learn all device attributes (up too 20 devices).
This switches are available with up to 8 keys. You can programm device states of lights (on/off, color, brightness) to this keys and restore the programmed states with a finger press.
In this mode you can switch as many scenes as your Busch Jäger device has keys (max: 8)

A alternative way of operation boundles two keys together to a dimmable switch to dimm four lightlink compatible devices. In this mode the left key and the matching right one represent one dimmer switch.

Using this switches as ZEDs gives up to 4 binary switches.

The technical documentation is here: https://www.busch-jaeger-katalog.de/files/files_ONLINE/Systemhandbuch_ZigBee_BJE-25-11-2015_EN.pdf
(The german one seems to be better maintained)

This switches are perfect in a HUE/Osram-Lightify/generic_zigbee solution with one zigbee network.

But I'm getting crazy to work with them in a dedicated zigbee network were they should work as ZEDs, not as routers and/or controllers.

closed 2019-01-12 by stale[bot]

yesman82, 2019-03-26:
Where you guys able to implement any Busch Jaeger ZLL switch?

Akkunasenbohrer, 2019-03-26:
It's working here.... was a little tricky but yes, Busch Jäger is working.

yesman82, 2019-03-26:
Strangely I can add the device (6715 U) but I don‘t receive any MQTT messages. Am I missing any device configuration or do I have to do something else? 
PS: I will get the CC debugger etc in a few days and then I’ll update the CC2531 firmware to the latest version and can sniff what is actually coming in.

Akkunasenbohrer, 2019-03-26:
Perhaps we should use the phone and use our mother language to get a better
idea what we're doing with our wall switches?

pinguinpfleger, 2019-06-28:
Somehow this issue has been closed but Busch-Jaeger Switches still not implemented yet.
@Akkunasenbohrer you made it? Can you please share how you get it done.

Boostflow, 2019-07-30:
> Somehow this issue has been closed but Busch-Jaeger Switches still not implemented yet.
> @Akkunasenbohrer you made it? Can you please share how you get it done.

I would be keen on learning the details as well. I've managed to add the devices using tb-killa's suggestion: " zigbee2mqtt:info 7/30/2019, 1:27:46 PM MQTT publish: topic 'zigbee2mqtt/0x00158d000321ac49', payload '{"contact":true,"linkquality":10,"battery":100,"voltage":3045}'"

No idea on how to actually control it. Any input would be appriciated

Daniel0984, 2019-08-19:
I would be intrested as well.

The number of endpoints depends on the model of the switch. There switches with a local relais, and there a switches without.

Has anyone had success in adding the needed bindings?

deconz handles this with some magic : https://github.com/dresden-elektronik/deconz-rest-plugin/blob/0c66676c4e1c0d4d2bdeb1f3b5d30ba5392be12e/bindings.cpp (Lines 2099)

[image]

[image]

Daniel0984, 2019-08-19:
I think thats as far as I can get without help:

https://pastebin.com/YJQdrMLJ

Light and light state does not work, reporting for all other buttons seems to work.
Publishing "ON" to device/state tries to enable the 2nd endpoint for some reason?

xeetsh, 2019-11-07:
Has there been any progress in implementing these? I would love to help as I have 12 of these switches in all varieties but struggle on getting useful information out of what has been postet yet. Is there anyone that would like to talk about what issues there are right now?

Akkunasenbohrer, 2019-11-07:
I use these switches together with Home Assistant in a separate zigbee
network.
Pressing a key triggers event. I don't programm this switches, all the
logic is in Home Assistant.
I put a maximum of 8 actions on each key to get 64 actions with one
switch...

Where's your problem using these switches?

.//. Arne

xeetsh, 2019-11-07:
I actually never reached the point where Home Assistant got any triggers from the switches as I struggle on configuring zigbee2mqtt in a way that the switches got recognized. Could you please post your zigbee-shepherd configuration (like devices.js, from_zigbee.js and to_zigbee.js) for the switches? This would really help me out.

xeetsh, 2019-11-16:
@Akkunasenbohrer Sorry for bothering you but I would love to get the switches set up and right now you seem to be the only one that made sense of that crazy Busch Jäger magic. Would you please consider sharing your configuration? Or if it isn't as easy as sharing the config files I would be more than happy to switch over to mail or talking on the phone about this. Thank you so much for your time.

Daniel0984, 2019-11-18:
I would be also interested. I got them to pair, but (as mentioned) failed to implement the logic for relais / non relais switches.

Akkunasenbohrer, 2019-11-19:
I made this month ago... I try to put all pieces together (and translate my remarks to english):

According to Bellows the Busch Jäger switches are seen as:
#Device:
 NWK: 0xb631
 IEEE: d8:5d:ef:11:a1:00:33:1a
 Endpoints:
   10: profile=0xc05e, device_type=DeviceType.COLOR_SCENE_CONTROLLER
     Input Clusters:
       Basic (0)
       LightLink (4096)
     Output Clusters:
       Identify (3)
       Groups (4)
       Scenes (5)
       On/Off (6)
       Level control (8)
       Ota (25)
       Color Control (768)
       LightLink (4096)
   11: profile=0xc05e, device_type=DeviceType.COLOR_SCENE_CONTROLLER
     Input Clusters:
       Basic (0)
       LightLink (4096)
     Output Clusters:
       Identify (3)
       Groups (4)
       Scenes (5)
       On/Off (6)
       Level control (8)
       Color Control (768)
       LightLink (4096)
   12: profile=0xc05e, device_type=DeviceType.COLOR_SCENE_CONTROLLER
     Input Clusters:
       Basic (0)
       LightLink (4096)
     Output Clusters:
       Identify (3)
       Groups (4)
       Scenes (5)
       On/Off (6)
       Level control (8)
       Color Control (768)
       LightLink (4096)
   13: profile=0xc05e, device_type=DeviceType.COLOR_SCENE_CONTROLLER
     Input Clusters:
       Basic (0)
       LightLink (4096)
     Output Clusters:
       Identify (3)
       Groups (4)
       Scenes (5)
       On/Off (6)
       Level control (8)
       Color Control (768)
       LightLink (4096)

In /srv/homeassistant/lib/python3.5/site-packages/homeassistant/components/zha/const.py
this Device Class is defined as "Binary Sensor" 

Problem:  The 8 Key Busch Jäger Switch looks like  4 ON/OFF switches.
Switched to "ON" there seems to be now way to recognice a second ON key press.
There is no "fire_event: True".

Solution:
I don't use the state. Instead i define a template sensor using the level. This template sensor can work as trigger source.
After triggering a state change event I set the attribut internaly (without sending it to the switch) to 128 (or any other value which is not true or false) so I can see the next ON or OFF key press which will change the level back to true or false.
I found no way to do this directly in Homeassistant. Appdaemon should be able to do this but I was too lazy to examine appdaemon.
I used Homeassistant's REST API and scripts using curl.

The maximum configuration I used together with the Busch Jäger 8 Key switch 6737-84 was
8 keys x 8 actions = 64 actions from one Busch Jäger wall switch.
Pressing key x 4 times means starting the fourth's action for this key.

You need NO configuration in the Switches. Of course, you must add them to your zigbee network but I don't use the LightLink configuration at all.
You can use the night-link function of the switch's LEDs but this has nothing to do with their zigbee interface.

Using the Busch Jäger switches together with the wall power supply is working perfect.
If the Busch Jäger switch replace a normal power switch with the Busch Jäger relais- or dimm lower part some things change:
Keys 1 and 2 will become real ON/OFF or Dimm keys.
These two keys will be visible as zigbee light. So we cannot use them with Homeassistant.
Homeassistant will be not able to read the state of this light reliable which is a real pain in the a...

I'm on the way making an archive with my home assistant configuration for xeetsh (nice phone call! Thank you Max!).

If you need a working home assistant configation drop me an email.

Daniel0984, 2019-11-21:
> If the Busch Jäger switch replace a normal power switch with the Busch Jäger relais- or dimm lower part some things change:
Keys 1 and 2 will become real ON/OFF or Dimm keys.
These two keys will be visible as zigbee light. So we cannot use them with Homeassistant.
Homeassistant will be not able to read the state of this light reliable which is a real pain in the a...

But isn't that exactly the point? I want my Home Automation to know what the light states are.
Looks like I will stick with Deconz for now, as it works and respects the Light Link features.

Akkunasenbohrer, 2019-11-21:
Perhaps I was too short in my answer.
I don't use light link features at all. Everything is controlled from
homeassistant. The Busch Jäger switches have no configuration at all.
I have Philips HUE bridges, Osram lightify gateways, Ikea Tradfri, AVM
fritz and some other "smart home" products at home.
I use home assistant to get them all together under one user interface.
I use Busch Jäger switches to trigger actions in homeassistant. For example
I use one Busch Jäger button to mute/unmute the TV.

Using a relais/dimmer module together with the Busch Jäger switch means
that the first two buttons will become real on/off switches,
Of course home assistant can switch this power line via zigbee control BUT
you cannot use these two buttons for anything else aka you cannot trigger
other actions with them.

I normally use only one button for ON/OFF (toggle) a light but this won't
work in this special case.
So I avoid to use the relais/dimmer module and use only the power supply
lower part so I can freely programm all buttons of the switch.

I have only one relais and one dimmer module which are still in use. When I
switch them directly using the Busch Jäger switch sometimes homeassistant
didn't get the state. I'm sure there is a solution for this problem but I
won't work here because I'll replace the last two relais/dimmer modules
with the power supply modules soon.

oepoemoepoe, 2020-01-06:
How do you pair these switches? I tried by holding down the top two buttons for 5sek, then pressing them again, and/or pressing the top right button. Nothing seems to work.

docbobo, 2020-01-06:
If I recall correctly:
- press top buttons until they stop blinking. Then press top buttons again. This resets the device according to docs. 
- Permit joining of network in Z2M. 

At that point, you should get some log output, that the RM01 device is not currently supported.

oepoemoepoe, 2020-01-06:
For some reason I do not get anything when doing that. I assume the switch is already factory reset now. I do not use Z2M but zigbee2mqtt directly and opened joining beforehand. I tried all things 6.1.1, 6.1.2, 6.1.3, 6.1.4 according to the user manual.

docbobo, 2020-01-06:
Does anybody have some more detail on endpoint 18? I am seeing this during configure:

```
zigbee2mqtt    | Endpoint {
zigbee2mqtt    |   ID: 18,
zigbee2mqtt    |   profileID: 49246,
zigbee2mqtt    |   deviceID: 0,
zigbee2mqtt    |   inputClusters: [ 0, 4, 3, 5, 6, 8 ],
zigbee2mqtt    |   outputClusters: [],
zigbee2mqtt    |   deviceNetworkAddress: 40011,
zigbee2mqtt    |   deviceIeeeAddress: '0xd85def11a10033ac',
zigbee2mqtt    |   clusters: { genBasic: { attributes: [Object] } },
zigbee2mqtt    |   _binds:
zigbee2mqtt    |    [ { cluster: 6,
zigbee2mqtt    |        type: 'endpoint',
zigbee2mqtt    |        deviceIeeeAddress: '0x00124b00194a204f',
zigbee2mqtt    |        endpointID: 1 } ] }
```

Given the absence of anything under outputClusters, I am uncertain on whether the device is going to report the manual trigger. Any thoughts?

docbobo, 2020-01-06:
Okay, according to https://github.com/openhab/org.openhab.binding.zigbee/issues/49, neither the dimmer nor the relais report the status change correctly when the physical switch is being user. Does anybody know how to implement polling of the state?

Daniel0984, 2020-01-07:
It works with deCONZ. The state of the light is reported almost instantly (shorter delay than I would expect with periodic polling)

docbobo, 2020-01-07:
Thanks, @Daniel0984. Do you know the FW version? I also stumbled across https://github.com/dresden-elektronik/deconz-rest-plugin/issues/874#issuecomment-435586949, which also comments that this is a known problem with the BJ relais & dimmer that they are unwilling to fix.

Daniel0984, 2020-01-07:
I am at the most recent beta version. But I've been using those switches for quite a while now. (see here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/368)

Since they reworked their switch editor (https://github.com/dresden-elektronik/deconz-rest-plugin/issues/463) its quite nice and WAF approved :)

oepoemoepoe, 2020-01-07:
When configuring with deconz, are the scenes actually saved on the BJ switch, meaning it works the same as when configured manually (even talking to the lights with the coordinator offline) or does only deconz know about the scenes (RM01 -> coordinator -> deconz -> lights)?

Daniel0984, 2020-01-07:
Not 100% sure for scenes, but groups work without the controller software running. Afaik scenes are saved inside the Zigbee groups, so they should also work.

I could create a scene and test it, if you want this to be confirmed.

oepoemoepoe, 2020-01-09:
> Not 100% sure for scenes, but groups work without the controller software running. Afaik scenes are saved inside the Zigbee groups, so they should also work.
> 
> I could create a scene and test it, if you want this to be confirmed.

Do you know how that works? The only in-clusters are 0 (basic) and 4096 (touchlink). How can deconz assign scenes/groups to these switches?

Daniel0984, 2020-01-10:
It should be done via the ZLL Commissioning Cluster / "Light link" (0x1000) for the non light endpoints. 
See https://github.com/dresden-elektronik/deconz-rest-plugin/blob/0c66676c4e1c0d4d2bdeb1f3b5d30ba5392be12e/general.xml#L2431

For the light itself I am also unsure. My network has several groups with the name of a switch and I assume in the past deCONZ created a group and added the whole device to it - although they stopped doing it like this with the new editor UI.

Maybe @docbobo can give some insights with his new sniffer :)

dreamsinmotion, 2020-01-28:
The buttons work fine for me. I Have the dimmer module 6715 U in combination with the 4 button unit. Setting the state (on/off) is no problem, but is there a way to control and get the dimmer intensity?

docbobo, 2020-01-28:
This is a little bit what I had suspected. It's due to the fact that the number of bindings is severely limited with those devices. There's certainly something that you could try: the configure method in devices.js binds genOnOff for endpoint 0x12. If you change that to genLevelCtrl, maybe (!) both dimmer and the state work. Feedback appreciated.

dreamsinmotion, 2020-01-28:
Thanks for your suggestion. Unfortunately no success by changing genOnOff to genLevelCtrl. I still get this debug message "zigbee2mqtt:error: No converter available for 'brightness' (215)"