Koenkk/zigbee2mqtt #420: Xiaomi MiJia Honeywell smoke detector as a alarm Sierene

dr-apple, 2018-09-28:
Hi ich have the Xiaomi MiJia Honeywell smoke detector and it work fine.

Can i controll via Zigbee the Xiaomi MiJia Honeywell smoke detector for put the alarm Sirene /beep turn on / off ?

cu Danny

JellevdK, 2018-11-29:
Nice idea, this will also create the 'linking' method, just as which is possible with the Xiaomi GW.

HannesG79, 2018-11-29:
Hi, that would be a good Option, if you have a false Alarm. The acknowledge button is small. and you can use it as Alarm as well.

arneeichhorn, 2019-01-16:
I would also love to see this feature.
Would be super nice to "link" the detectors to each other with this feature and also to use them as sirens in use with motion sensors and/or door contacts.
I read that it should also be possible to set sensitivity for the detection with the original GW? Just like its done for the Gas Detector, that supports selftest and sensitivity settings via zigbee2mqtt

Koenkk, 2019-01-28:
In order to support such feature we need to check what the original gateway sends (https://github.com/Koenkk/zigbee2mqtt/blob/dev/docs/how_tos/how_to_sniff_zigbee_traffic.md). Then it's indeed just a matter of adding a new toZigbee converter.

neerdoc, 2019-01-28:
Ok, since I am on a Mac, I could not follow those instructions.
I did find [these](https://github.com/andrewdodd/pyCCSniffer) instructions instead that I followed. Don't know if it gives what we need, so please let me know. Pasted it [here](https://pastebin.com/gCGjBxxK).

Koenkk, 2019-01-29:
@neerdoc I don't see the cluster and command being send (that's what we need in order to support this).

neerdoc, 2019-01-29:
Managed to get the data into WireShark, but there are literary thousands of entries after 60 seconds of data during one join. Exactly which frame to you need?

neerdoc, 2019-01-29:
Reading the Zigbee standard it seems that this should indeed work:

> The IAS WD cluster provides an interface to the functionality of any Warning Device equipment of the IAS system. Using this cluster, a ZigBee enabled CIE device can access a ZigBee enabled IAS WD device and issue alarm warning indications (siren, strobe lighting, etc.) when a system alarm condition is detected (according to [N1])

The following definition is in the standard:
- Cluster Identifier = 0x0502
- Command Identifier
  - Start Warning = 0x00
  - Squak = 0x01

After that it gets very detailed on what to send in the payload. It is all defined in chapter 8.4.1 in the [specification](http://www.zigbee.org/~zigbeeor/wp-content/uploads/2014/10/07-5123-06-zigbee-cluster-library-specification.pdf).

Are there any instructions on how I can trigger a message in the "toZigbee.js" from the command line? (Like pushing a certain button on the keyboard.) If so, maybe I can try to implement this message myself.

Koenkk, 2019-01-30:
@neerdoc the easiest way to try this is by using an MQTT client (e.g. MQTT.fx).

- Add a toZigbee converter with a key (e.g. `key: ['alarm'],`) and the command you want to send.
- Add this toZigbee converter to your device in `devices.js`
- Send an MQTT command to `zigbee2mqtt/[DEVICE_ID]/set` with payload `{"alert":""}`
- This will trigger the code of your toZigbee converter.

neerdoc, 2019-01-30:
Thanks @Koenkk, I got so far that I can trigger the message now! Unfortunately, the message fails to be received it seems. I will continue later, but it is a bit tricky to understand exactly what message is sent through all the layers of npm_modules... It seems very clear what I need to send in the Zigbee standard; it is very _unclear_ how I achieve this.

Current error message:
`  zigbee2mqtt:error 1/30/2019, 9:07:46 PM Zigbee publish to device '0x005043c91f292a8f', ssIasWd - startWarning - {"startwarninginfo":2,"warningduration":2} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 240. MAC transaction expired.
`

neerdoc, 2019-01-31:
Unfortunately, I do not have any gateway except for the CC2531... :)

Debugging some more. After hacking around in the *zcl_packet* lib, I got it to send this message:
`cc-znp:SREQ --> AF:dataRequest, { dstaddr: 10751, destendpoint: 1, srcendpoint: 1, clusterid: 1282, transid: 6, options: 48, radius: 30, len: 8, data: <Buffer 01 07 00 22 02 00 00 00> } +1s`

which I believe is very close to what I need. According to the specification, the iasWD definition (cluster id 1285 (or 0x502)) should be: 00 22 02 00 00 00 for a 2 second alert at high level without strobe. Nothing happens, but as you can see, there are an 01 07 in front of the actual message. The 07 also seems to be random (or maybe counting, a bit hard to determine). I do not know where these are generated (and to be totally honest I don't know if they are the problem either).

I actually get a response from the device on this message:
`  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 1282, srcaddr: 10751, srcendpoint: 1, dstendpoint: 1, wasbroadcast: 0, linkquality: 63, securityuse: 0, timestamp: 10220817, transseqnumber: 0, len: 5, data: <Buffer 18 06 0b 00 00> } +1ms`

Any hints on whether those two bytes are the problem or not is appreciated!

neerdoc, 2019-01-31:
Unfortunately, I only have the CC2531 as a gateway ;)

Debugging some more. After hacking some in the *zcl-packet* library, I managed to send the following message:
`cc-znp:SREQ --> AF:dataRequest, { dstaddr: 10751, destendpoint: 1, srcendpoint: 1, clusterid: 1282, transid: 5, options: 48, radius: 30, len: 8, data: <Buffer 01 06 00 22 02 00 00 00> } +6s`

Now this is _very_ close to the specification. To make the alarm to a high level 2 second siren, I should send the message "00 22 02 00 00 00". As you can see, it is correct except the first two bytes (01 06) which I cannot figure out where they are coming from or how to change. The second byte also seems to change if I repeat the message.

I actually even get a response from the device (though I have not decoded what it means yet...):
`cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 1282, srcaddr: 10751, srcendpoint: 1, dstendpoint: 1, wasbroadcast: 0, linkquality: 63, securityuse: 0, timestamp: 10220817, transseqnumber: 0, len: 5, data: <Buffer 18 06 0b 00 00> } +1ms`

Any ideas on if those two bytes is the problem?

Koenkk, 2019-02-02:
@neerdoc how does the `rawPayload` look like? https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/components/af.js#L30

neerdoc, 2019-02-05:
Been busy for a few days, but now I had the time:
`zigbee-shepherd:af <Buffer 01 02 00 22 02 00 00 00> +0ms`

So it seems to be the same problem. The first "01 02" should not be there (I think).

Koenkk, 2019-02-06:
The frame is constructed here: https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/components/af.js#L346

Can you add `console.log(frameCntl, manufCode, seqNum, cmd, zclData, cId)` there?

sascha777, 2019-02-17:
@neerdoc it’s really great to see that you are working on the alarm message for the Honeywell Mitja siren!!
Did you find out how to adress them in the payload?

Stefan300381, 2019-02-25:
hi all, I joined this game lately and maybe you can help me to help you. If you need the original gateway, I own one :-) Anything I should do?

Stefan300381, 2019-03-14:
hi, me again. From reading the information above, is it correct to say that in case of fire, only ONE smoke detector (the one that detects the fire) will give an alarm (loud sirene) and the others will stay calm? This is something that can be achieved with original GW but not via zigbee2mqtt, correct?

neerdoc, 2019-03-14:
Hi! Sorry for the late reply...

You are correct in the assumption that only the smoke detector that detects the smoke will sound the alarm. The intention is that all sirens should sound, or be able to do it depending on the configuration of course.

I don't know if it is possible with an original gateway, but if it is, it would be very helpful if you can test it and also log the traffic that is sent to the sirene. 

The Zigbee protocol specification together with the capability that is reported by the smoke detector seems to support the notion that it should be possible.

DoctorOctagonapus, 2019-05-14:
> I don't know if it is possible with an original gateway, but if it is, it would be very helpful if you can test it and also log the traffic that is sent to the sirene.

According to the documentation it is possible, though they call it Tandem Mode. It's only supported with the official gateway though, presumably there's some kind of payload or signal the gateway can send to the alarm to make it sound.

Stefan300381, 2019-05-14:
I'm willing to help, but I'm not a zigbee expert. Would need some help and instructions how to sniff. I think, the first challenge will be to flash a specific sniffing firmware, right?

AS137430, 2019-05-16:
[Hubitat driver code](https://raw.githubusercontent.com/veeceeoh/xiaomi-hubitat/master/devicedrivers/xiaomi-honeywell-smoke-detector.src/xiaomi-honeywell-smoke-detector.groovy)

From the code,
value of 3 for zone_status is 'tested':
```
// Parse IAS Zone Status message (0 = clear, 1 = detected, or 3 = tested)
private Map parseZoneStatusMessage(status) {
	def value = ["clear", "detected", "tested"]
	def eventType = ["Clear", "Detected", "Tested"]
	def descText = ["All clear", "Smoke detected", "Completed self-test"]
```
For self-test:
```
def test() {
	// Sends a command to sensor to complete self-test. An alarm beep indicates normal operation.
	return zigbee.writeAttribute(0x0500, 0xFFF1, DataType.UINT32, 0x03010000, [mfgCode: "0x115F"])
}
```
How to convert this to code for toZigbee.js?
```
if (sensLevel) {
		displayInfoLog("Setting smoke sensitivity level to ${sensLevel}")
		zigbee.writeAttribute(0x0500, 0xFFF1, DataType.UINT32, Integer.parseInt(sensitivity), [mfgCode: "0x115F"])
	}
```
For setting sensitivity level, probably goes to toZigbee.js as well?

Koenkk, 2019-05-16:
Try with the following in `toZigbee.js`

```js
JTYJ_GD_01LM_BW_self_test: {
    key: ['self_test'],
    convert: (key, value, message, type, postfix) => {
        if (type === 'set') {
            return [{
                cid: 'ssIasZone',
                cmd: 'write',
                cmdType: 'foundation',
                zclData: [{
                    attrId: 0xFFF1,
                    dataType: 0x23,
                    attrData: 0x03010000,
                }],
                cfg: cfg.xiaomi,
            }];
        }
    },
},
```

add to https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L399 `tz.JTYJ_GD_01LM_BW_self_test` and publish to `zigbee2mqtt/[DEVICE_ID]/set` payload `{"self_test": ""}`

dr-apple, 2019-05-16:
How can i finde the toZigbee.js file ?

dr-apple, 2019-05-16:
Now can we this make for 

HS1SA | HEIMAN Smoke detector (smoke)

?

AS137430, 2019-05-16:
@Koenkk: 

Your suggested changes worked, the alarm siren went off once.
```
5/16/2019, 8:50:49 PM - debug: Received MQTT message on 'zigbee2mqtt/0x00158d00028785ae/set' with data '{"self_test":""}'
5/16/2019, 8:50:49 PM - info: Zigbee publish to device '0x00158d00028785ae', ssIasZone - write - [{"attrId":65521,"dataType":35,"attrData":50397184}] - {"manufSpec":1,"disDefaultRsp":1,"manufCode":4447} - null
```

AS137430, 2019-05-16:
I modified [fromZigbee.js#L1105](https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/fromZigbee.js#L1105) with the following:
```
JTYJGD01LMBW_smoke: {
        cid: 'ssIasZone',
        type: 'statusChange',
        convert: (model, msg, publish, options) => {
            switch (msg.data.zoneStatus){
                case 1: return {smoke: true}; break;
                case 2: return {tested: true}; break;
                default: return {smoke: false}; break;
            }
        },
    },
```
and pressed button for 3 seconds and was able to get this
```
5/16/2019, 9:38:05 PM - debug: Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":2}' of device 'lumi.sensor_smoke' (0x00158d00028785ae) of endpoint 1
```
But the 'tested' status doesn't clear even when zoneStatus is back to 0.
```
5/16/2019, 9:38:07 PM - debug: Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":0}' of device 'lumi.sensor_smoke' (0x00158d00028785ae) of endpoint 1
5/16/2019, 9:38:07 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00028785ae', payload '{"battery":100,"voltage":3205,"linkquality":18,"smoke":false,"tested":true}'
```

AS137430, 2019-05-16:
For sensitivity setting, does this code make sense?

```
JTYJ_GD_01LM_BW_sensitivity: {
        key: ['sensitivity'],
        convert: (key, value, message, type, postfix) => {
            const cid = 'ssIasZone';

            if (type === 'set') {
                const lookup = {
                    'low': 0x04030000,
                    'medium': 0x04020000,
                    'high': 0x04010000,
                };

                if (lookup.hasOwnProperty(value)) {
                    return [{
                        cid: cid,
                        cmd: 'write',
                        cmdType: 'foundation',
                        zclData: [{
                            attrId: 0xFFF1, // presentValue
                            dataType: 0x23, // dataType
                            attrData: lookup[value],
                        }],
                        cfg: cfg.xiaomi,
                    }];
                }
            }
        },
    },
```

Koenkk, 2019-05-17:
With that code you should indeed be able to set the sensitivity. 

Feel free to make a PR to https://github.com/koenkk/zigbee-shepherd-converters once you got stuff working.

Stefan300381, 2019-05-17:
great you made progress, but now I'm confused. the PR will fix what? to set the senitivity for smoke detection? or using the detector as alarm sirene?

AS137430, 2019-05-18:
@Koenkk 

Today's tests after adding the sensitivity code yielded different results:
```
5/18/2019, 3:44:53 PM - info: Zigbee publish to device '0x00158d00028785ae', ssIasZone - write - [{"attrId":65521,"dataType":35,"attrData":50397184}] - {"manufSpec":1,"disDefaultRsp":1,"manufCode":4447} - null
5/18/2019, 3:44:53 PM - error: Zigbee publish to device '0x00158d00028785ae', ssIasZone - write - [{"attrId":65521,"dataType":35,"attrData":50397184}] - {"manufSpec":1,"disDefaultRsp":1,"manufCode":4447} - null failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.
5/18/2019, 3:44:53 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.","meta":{"entity":{"ID":"0x00158d00028785ae","type":"device","friendlyName":"0x00158d00028785ae"},"message":"{\"self_test\":\"\"}"}}'
```
and
```
/18/2019, 3:41:15 PM - info: Zigbee publish to device '0x00158d00028785ae', ssIasZone - write - [{"attrId":65521,"dataType":35,"attrData":67174400}] - {"manufSpec":1,"disDefaultRsp":1,"manufCode":4447} - null
5/18/2019, 3:41:16 PM - error: Zigbee publish to device '0x00158d00028785ae', ssIasZone - write - [{"attrId":65521,"dataType":35,"attrData":67174400}] - {"manufSpec":1,"disDefaultRsp":1,"manufCode":4447} - null failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.
5/18/2019, 3:41:16 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.","meta":{"entity":{"ID":"0x00158d00028785ae","type":"device","friendlyName":"0x00158d00028785ae"},"message":"{\"sensitivity\":\"high\"}"}}'
```

Earlier I had request timeout issues with the CC2531 dongle and to reflash the firmware. I'm not sure if the errors are related to the code changes.

@Stefan300381 : I suppose when things do work out, the PR will do both: alarm siren and set smoke sensitivity.

Koenkk, 2019-05-18:
What firmware are you running?

AS137430, 2019-05-18:
The hex file from CC2531_20190425.zip taken from [here](https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin)

peterforeman, 2019-05-18:
I'm interested in running this code so added it to my zigbee2mqtt 1.4.0 container. The smoke sensor is working/connected and I can receive messages. But same here as @AS137430: "No network route" (code 205).

Also firmware version 20190425.

Koenkk, 2019-05-19:
@AS137430 at the moment that setting sensitivity fails, does the self test work?

AS137430, 2019-05-20:
@Koenkk : Self test doesn't work, sending the MQTT message doesn't sound the alarm siren anymore. And the log shows the following
```
5/20/2019, 9:39:06 AM - info: Zigbee publish to device '0x00158d00028785ae', ssIasZone - write - [{"attrId":65521,"dataType":35,"attrData":50397184}] - {"manufSpec":1,"disDefaultRsp":1,"manufCode":4447} - null
5/20/2019, 9:39:07 AM - error: Zigbee publish to device '0x00158d00028785ae', ssIasZone - write - [{"attrId":65521,"dataType":35,"attrData":50397184}] - {"manufSpec":1,"disDefaultRsp":1,"manufCode":4447} - null failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.
5/20/2019, 9:39:07 AM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.","meta":{"entity":{"ID":"0x00158d00028785ae","type":"device","friendlyName":"0x00158d00028785ae"},"message":"{\"self_test\":\"\"}"}}'
```

Koenkk, 2019-05-20:
@AS137430 I'm not sure what is causing this, why did it work before? Do you perhaps need to wake the device by pressing a button on it?

AS137430, 2019-05-21:
The cc2531 was on an earlier version of firmware when it worked. I will flash to the earlier version to test if it works and report back the results

AS137430, 2019-05-22:
The AF code 205 error is still present with firmware: CC2531_MAX_STABILITY_20190315.zip.

VictorvandenHoven, 2019-05-29:
> Try with the following in `toZigbee.js`
>
> ```js
> JTYJ_GD_01LM_BW_self_test: {
>     key: ['self_test'],
>     convert: (key, value, message, type, postfix) => {
>         if (type === 'set') { [...]

After doing what you suggest, zigbee2mqtt/Rookmelder3/set   json payload: {"self_test": ""} 
I get the following logging:

> May 29 14:50:30 ZIGBEE npm[14419]:   zigbee2mqtt:debug 5/29/2019, 2:50:30 PM Received MQTT message on 'zigbee2mqtt/Rookmelder3/set' with data '{"self_test": ""}'
> May 29 14:50:30 ZIGBEE npm[14419]:   zigbee2mqtt:info 5/29/2019, 2:50:30 PM Zigbee publish to device '0x00158d00025d94f5', undefined - undefined - undefined - {"manufSpec":0,"disDefaultRsp":0} - null
> May 29 14:50:30 ZIGBEE npm[14419]:   ****zigbee2mqtt:error 5/29/2019, 2:50:30 PM Unknown zigbee publish cmdType undefined**

**Any idea what to change to get it working?**

When pushing the "test"-button, I get the following:

> May 29 14:57:32 ZIGBEE npm[14419]:   zigbee2mqtt:debug 5/29/2019, 2:57:32 PM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"65281":{"1":3125,"3":28,"4":5032,"5":8,"6":[0,0],"8":4100,"10":38681,"100":0,"150":0}}}' of device 'lumi.sensor_smoke' (0x00158d00025d94f5) of endpoint 1
> May 29 14:57:32 ZIGBEE npm[14419]:   zigbee2mqtt:info 5/29/2019, 2:57:32 PM MQTT publish: topic 'zigbee2mqtt/**Rookmelder3**', payload '{"smoke":false,"linkquality":39,"battery":100,"voltage":3125}'
> May 29 14:57:32 ZIGBEE npm[14419]:   zigbee2mqtt:debug 5/29/2019, 2:57:32 PM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"modelId":"lumi.sensor_smoke"}}' of device 'lumi.sensor_smoke' (**0x00158d00025d94f5**) of endpoint 1
> May 29 14:57:33 ZIGBEE npm[14419]:   zigbee2mqtt:debug 5/29/2019, 2:57:33 PM Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":0}' of device 'lumi.sensor_smoke' (0x00158d00025d94f5) of endpoint 1
> May 29 14:57:33 ZIGBEE npm[14419]:   zigbee2mqtt:info 5/29/2019, 2:57:33 PM MQTT publish: topic 'zigbee2mqtt/Rookmelder3', payload '{"smoke":false,"linkquality":39,"battery":100,"voltage":3125}'
> May 29 14:57:33 ZIGBEE npm[14419]:   zigbee2mqtt:debug 5/29/2019, 2:57:33 PM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"65281":{"1":3125,"3":27,"4":5032,"5":8,"6":[0,1],"8":4100,"10":38681,"100":0,"150":0}}}' of device 'lumi.sensor_smoke' (0x00158d00025d94f5) of endpoint 1
> May 29 14:57:33 ZIGBEE npm[14419]:   zigbee2mqtt:info 5/29/2019, 2:57:33 PM MQTT publish: topic 'zigbee2mqtt/Rookmelder3', payload '{"smoke":false,"linkquality":36,"battery":100,"voltage":3125}'
> 
> 
> 
> 
>

VictorvandenHoven, 2019-05-29:
> After doing what you suggest, zigbee2mqtt/Rookmelder3/set json payload: {"self_test": ""}
> I get the following logging:
>
>
> **Any idea what to change to get it working?**
>
> When pushing the [...]

After updating zigbee2mqtt to 1.4.0 I now get the following:

> 
> May 29 18:26:20 ZIGBEE npm[15706]:   zigbee2mqtt:debug 5/29/2019, 6:26:20 PM Received MQTT message on 'zigbee2mqtt/0x00158d00025d94f5/set' with data '{"self_test": ""}'
> May 29 18:26:20 ZIGBEE npm[15706]:   zigbee2mqtt:info 5/29/2019, 6:26:20 PM Zigbee publish to device '0x00158d00025d94f5', ssIasZone - write - [{"attrId":65521,"dataType":35,"attrData":50397184}] - {"manufSpec":1,"disDefaultRsp":1,"manufCode":4447} - null
> May 29 18:26:26 ZIGBEE npm[15706]:   zigbee2mqtt:error 5/29/2019, 6:26:26 PM Zigbee publish to device '0x00158d00025d94f5', ssIasZone - write - [{"attrId":65521,"dataType":35,"attrData":50397184}] - {"manufSpec":1,"disDefaultRsp":1,"manufCode":4447} - null failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.
> May 29 18:26:26 ZIGBEE npm[15706]:   zigbee2mqtt:info 5/29/2019, 6:26:26 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.","meta":{"entity":{"ID":"0x00158d00025d94f5","type":"device","friendlyName":"Rookmelder3"},"message":"{\"self_test\": \"\"}"}}'

When pushing the test-button by hand, I do receive messages. So I suspect the message "Please confirm that the device has (re)joined the network"  is not valid..........

What else to try to get self_test working?

VictorvandenHoven, 2019-05-29:
After a couple of hours the alarm beep went off after setting the "self_test" message on the queue.
Did not change anything, but it seems to work now!

Next thing to try is the sensitivity.

lovethat777, 2019-06-12:
Any progress ?

klitscheIO, 2019-07-11:
I could also trigger the test via MQTT. But the single beep the detector fires cannot be used as siren. It is short and not very loud. We have to find the message the original gateway sends to a second smoke detector in "tandem mode" to trigger the alarm.

> According to the documentation it is possible, though they call it Tandem Mode. It's only supported with the official gateway though, presumably there's some kind of payload or signal the gateway [...]

closed 2019-09-16 by stale[bot]

gonzasmts, 2019-10-10:
Hello, 
any news about this?

it would be nice if we are able to:
- set the sensitivity (i want to add one in the kitchen)
- execute self-test
- Tandem mode (not required for my case)

also i don't know if possible, but is possible to mute the alarm after it get’s triggered, without need to click the button?

kovaga, 2019-10-31:
> I could also trigger the test via MQTT. But the single beep the detector fires cannot be used as siren. It is short and not very loud. We have to find the message the original gateway sends to a [...]

I dont think it is possible to remotely trigger the second detector to sound the alarm. I have two  connected smoke detectors, but no option in the original Xiaomi map to start the 2nd alarm when the 1st detects smoke. The "Linkage hub alarm" option allows only the Hub to sound the alarm.

And thats triggered by the     ZoneStatus: 0x0001, Alarm 1

ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x19)
    Sequence Number: 8
    Command: Zone Status Change Notification (0x00)
    ZoneStatus: 0x0001, Alarm 1
        .... .... .... ...1 = Alarm 1: Opened or alarmed
        .... .... .... ..0. = Alarm 2: Closed or not alarmed
        .... .... .... .0.. = Tamper: Not tampered
        .... .... .... 0... = Battery: Battery OK
        .... .... ...0 .... = Supervision Reports: Does not report
        .... .... ..0. .... = Restore Reports: Does not report restore
        .... .... .0.. .... = Trouble: OK
        .... .... 0... .... = AC (mains): AC/Mains OK
    Extended Status: 0x00
    Zone ID: 0xff
    Delay (in quarterseconds): 0

danielo515, 2021-10-30:
Thanks or the answer. But will that disable the siren or just stop the one that is already running?

kovaga, 2021-10-31:
> Thanks or the answer. But will that disable the siren or just stop the one that is already running?

Not quite sure what you mean here, but basically the siren is always connected to the network.
When you set a duration and a signal to start producing a noise, there are two ways how it will stop this noise, either wait for the duration to expire or send an "off" signal. This will stop the running noise until a new command will arrivve to initiate a new one.

danielo515, 2021-11-03:
Maybe if I explain the scenario I have in mind it is easier to understand.
I have the Honeywell smoke detector in my kitchen, on the ceiling. When the
siren starts I'm the only one on the family tall enough to hit the button
that stops it. For that reason, I want to create an automation to send the
stop command to the smoke detector. But of course, if there is smoke again
I want it to work, I don't want to disable it permanently. Hope my
explanation makes sense now.

Regards

alanieska, 2023-03-25:
hello, has anyone solved this? it happened to me that it triggered a false alarm (I looked through the camera that nothing was on fire) but unfortunately it beeped the whole time until I got home from work. Is there a way to turn this off in m2qtt? (sorry for the english, google translate)