Koenkk/zigbee2mqtt #421: WXKG11LM - Xiaomi Aqara wireless switch

thehaxxa, 2018-09-28:
The new model of the Xiaomi Aqara wireless switch (WXKG11LM) has a new model ID: "lumi.remote.b1acn01" . 

When adding the device I get this warning: 

`zigbee2mqtt:warn 9/28/2018, 12:55:16 PM Device with modelID 'lumi.remote.b1acn01      ' is not supported.
`

I added the device to devices.js using the existing code, with the name changed:

`    {
        zigbeeModel: ['lumi.remote.b1acn01'],
        model: 'WXKG11LM',
        vendor: 'Xiaomi',
        description: 'Aqara wireless switch',
        supports: 'single, double, triple, quadruple click',
        fromZigbee: [fz.xiaomi_battery_3v, fz.WXKG11LM_click, fz.ignore_onoff_change, fz.ignore_basic_change],
        toZigbee: [],
    },`

Despite this it still warns its not supported:

`zigbee2mqtt:warn 9/28/2018, 12:58:19 PM Device with modelID 'lumi.remote.b1acn01      ' is not supported.
`

thehaxxa, 2018-09-28:
Turns out the model name had 6 Null unicode characters after its ID; I have managed to get it working and submitted a Pull Request: https://github.com/Koenkk/zigbee-shepherd-converters/pull/89 .

closed 2018-09-28 by thehaxxa

elmexdechileen, 2018-10-03:
Thanks @thehaxxa, couldn't figure this out as the logging trimmed the white spaces I think.