Koenkk/zigbee2mqtt #422: RWL021 new Version ??

arteck, 2018-09-28:
i have here 2 dimmer 1 year old .. it works .. no probelem..
today i bought a new one with little Problem

```
 zigbee2mqtt:error 2018-9-28 16:10:55 Cannot get the Node Descriptor of the Device: 0x0017880103caf394 (Error: Timed out after 10000 ms)
  zigbee2mqtt:error 2018-9-28 16:11:15 Cannot get the Node Descriptor of the Device: 0x0017880103caf394 (Error: Timed out after 10000 ms)
```

any ideas ??

```

{
   "id":42,
   "type":"EndDevice",
   "ieeeAddr":"0x0017880103caf394",
   "nwkAddr":3480,
   "manufId":4107,
   "manufName":"Philips",
   "powerSource":"Battery",
   "modelId":"RWL021",
   "epList":[
      1,
      2
   ],
   "status":"online",
   "joinTime":1538143108,
   "endpoints":{
      "1":{
         "profId":49246,
         "epId":1,
         "devId":2096,
         "inClusterList":[
            0
         ],
         "outClusterList":[
            0,
            3,
            4,
            5,
            6,
            8
         ],
         "clusters":{
            "genBasic":{
               "dir":{
                  "value":3
               },
               "attrs":{
                  "zclVersion":1,
                  "appVersion":2,
                  "stackVersion":1,
                  "hwVersion":1,
                  "manufacturerName":"Philips",
                  "modelId":"RWL021",
                  "dateCode":"20160302",
                  "powerSource":3,
                  "swBuildId":"5.45.1.17846"
               }
            },
            "genIdentify":{
               "dir":{
                  "value":2
               },
               "attrs":{

               }
            },
            "genGroups":{
               "dir":{
                  "value":2
               },
               "attrs":{

               }
            },
            "genScenes":{
               "dir":{
                  "value":2
               },
               "attrs":{

               }
            },
            "genOnOff":{
               "dir":{
                  "value":2
               },
               "attrs":{

               }
            },
            "genLevelCtrl":{
               "dir":{
                  "value":2
               },
               "attrs":{

               }
            }
         }
      },
      "2":{
         "profId":260,
         "epId":2,
         "devId":12,
         "inClusterList":[
            0,
            1,
            3,
            15,
            64512
         ],
         "outClusterList":[
            25
         ],
         "clusters":{
            "64512":{
               "dir":{
                  "value":1
               },
               "attrs":{

               }
            },
            "genBasic":{
               "dir":{
                  "value":1
               },
               "attrs":{

               }
            },
            "genPowerCfg":{
               "dir":{
                  "value":1
               },
               "attrs":{
                  "batteryVoltage":29,
                  "batteryPercentageRemaining":200
               }
            },
            "genIdentify":{
               "dir":{
                  "value":1
               },
               "attrs":{

               }
            },
            "genBinaryInput":{
               "dir":{
                  "value":1
               },
               "attrs":{

               }
            },
            "genOta":{
               "dir":{
                  "value":2
               },
               "attrs":{

               }
            }
         }
      }
   },
   "_id":"ogMtQUjhhU5AnNYD"
}
```

Koenkk, 2018-09-28:
If I understand correctly, the problem is that you cannot get it to pair, right?

arteck, 2018-09-28:
yes i can but after that the button not works..

arteck, 2018-09-28:
here one that works

```
{
  "id": 22,
  "type": "EndDevice",
  "ieeeAddr": "0x0017880103caa673",
  "nwkAddr": 32159,
  "manufId": 4107,
  "manufName": "Philips",
  "powerSource": "Battery",
  "modelId": "RWL021",
  "epList": [
    1,
    2
  ],
  "status": "offline",
  "joinTime": null,
  "endpoints": {
    "1": {
      "profId": 49246,
      "epId": 1,
      "devId": 2096,
      "inClusterList": [
        0
      ],
      "outClusterList": [
        0,
        3,
        4,
        5,
        6,
        8
      ],
      "clusters": {
        "genBasic": {
          "dir": {
            "value": 3
          },
          "attrs": {
            "zclVersion": 1,
            "appVersion": 2,
            "stackVersion": 1,
            "hwVersion": 1,
            "manufacturerName": "Philips",
            "modelId": "RWL021",
            "dateCode": "20160302",
            "powerSource": 3,
            "swBuildId": "5.45.1.17846"
          }
        },
        "genIdentify": {
          "dir": {
            "value": 2
          },
          "attrs": {}
        },
        "genGroups": {
          "dir": {
            "value": 2
          },
          "attrs": {}
        },
        "genScenes": {
          "dir": {
            "value": 2
          },
          "attrs": {}
        },
        "genOnOff": {
          "dir": {
            "value": 2
          },
          "attrs": {}
        },
        "genLevelCtrl": {
          "dir": {
            "value": 2
          },
          "attrs": {}
        }
      }
    },
    "2": {
      "profId": 260,
      "epId": 2,
      "devId": 12,
      "inClusterList": [
        0,
        1,
        3,
        15,
        64512
      ],
      "outClusterList": [
        25
      ],
      "clusters": {
        "64512": {
          "dir": {
            "value": 1
          },
          "attrs": {}
        },
        "genBasic": {
          "dir": {
            "value": 1
          },
          "attrs": {
            "zclVersion": 1,
            "appVersion": 2,
            "stackVersion": 1,
            "hwVersion": 1,
            "manufacturerName": "Philips",
            "modelId": "RWL021",
            "dateCode": "20160302",
            "powerSource": 3,
            "swBuildId": "5.45.1.17846"
          }
        },
        "genPowerCfg": {
          "dir": {
            "value": 1
          },
          "attrs": {
            "batteryVoltage": 28,
            "batteryPercentageRemaining": 200
          }
        },
        "genIdentify": {
          "dir": {
            "value": 1
          },
          "attrs": {}
        },
        "genBinaryInput": {
          "dir": {
            "value": 1
          },
          "attrs": {}
        },
        "genOta": {
          "dir": {
            "value": 2
          },
          "attrs": {}
        }
      }
    }
  },
  "_id": "VQ5VN6Xei2eAPBMM"
```

arteck, 2018-09-29:
```
zigbee-shepherd:request REQ --> ZDO:mgmtPermitJoinReq +393ms
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 1 +40ms
  zigbee2mqtt:info 2018-9-29 08:48:20 zigbee-shepherd ready
  zigbee2mqtt:info 2018-9-29 08:48:20 Connected to MQTT server
  zigbee2mqtt:info 2018-9-29 08:48:20 MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
  zigbee2mqtt:info 2018-9-29 08:48:20 MQTT publish, topic: 'zigbee2mqtt/0x00158d000232aa65', payload: '{"battery":"100.00","voltage":3055,"linkquality":5,"contact":true}'
  zigbee-shepherd:msgHdlr IND <-- ZDO:permitJoinInd +377ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:mgmtPermitJoinRsp +6ms
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +14ms
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +4s
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +4ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object] +14s
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 1 +8ms
  zigbee-shepherd-converters:devices Configured '(cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb)' with result 'Error: AF data request fails, status code: 183. APS no ack.' +0ms
  zigbee2mqtt:error 2018-9-29 08:48:38 Failed to configure 0x0017880103caf394 0x0017880103caf394
```

arteck, 2018-09-29:
ok i found the problem
i reinstall the zigbee2mqtt and now works.... strange ...very strange

closed 2018-09-29 by arteck