Koenkk/zigbee2mqtt #425: Nue Double Power Outlet - Device Support Help

mmnpkf, 2018-09-30:
Hi,
Let me start by saying that I'm a complete novice and I'm just fumbling my way through. I'm working through "how to support new devices" to add the Nue Double Power Outlet:

[https://www.amazon.com.au/Approved-Standard-Nue-SmartThings-Automation/dp/B07GX943VP/ref=sr_1_15?ie=UTF8&qid=1538269827&sr=8-15&keywords=nue+zigbee](url)

I've managed to successfully pair it with the CC2531 stick.  However, it is showing up as just one device.  This sort of makes sense since it is one box with two power outlets.  I'm trying to figure out how I can control each outlet.  When I write the zigbee message to the console when manually turning each outlet on or off I get the following:

Right Outlet:
`{ type: 'devChange',
  endpoints: 
   [ Endpoint {
       isLocal: [Function],
       device: [Object],
       profId: 49246,
       epId: 11,
       devId: 0,
       inClusterList: [Array],
       outClusterList: [Array],
       clusters: [Object],
       onAfDataConfirm: null,
       onAfReflectError: null,
       onAfIncomingMsg: null,
       onAfIncomingMsgExt: null,
       onZclFoundation: null,
       onZclFunctional: null,
       foundation: [Function],
       functional: [Function],
       bind: [Function],
       unbind: [Function],
       read: [Function],
       write: [Function],
       report: [Function] } ],
  data: { cid: 'genOnOff', data: { onOff: 1 } } }
`

Left Outlet:
`{ type: 'devChange',
  endpoints: 
   [ Endpoint {
       isLocal: [Function],
       device: [Object],
       profId: 49246,
       epId: 12,
       devId: 0,
       inClusterList: [Array],
       outClusterList: [Array],
       clusters: [Object],
       onAfDataConfirm: null,
       onAfReflectError: null,
       onAfIncomingMsg: null,
       onAfIncomingMsgExt: null,
       onZclFoundation: null,
       onZclFunctional: null,
       foundation: [Function],
       functional: [Function],
       bind: [Function],
       unbind: [Function],
       read: [Function],
       write: [Function],
       report: [Function] } ],
  data: { cid: 'genOnOff', data: { onOff: 1 } } }
`
The difference appears to be just the "epid" which is 12 for one outlet and 11 for the other.

Can anyone advise me how I use the epid to expose this device as two devices and control them independently?
Thanks.

mmnpkf, 2018-09-30:
Seems I was too quick to post my question.  I found some examples in the fromZigbee file using epId and I think I've now got it working - just need to do more testing.

closed 2018-09-30 by mmnpkf

Koenkk, 2018-09-30:
Could you make a PR if you get it working?

mmnpkf, 2018-10-01:
Mmmm... getting the dreaded "cannot get node descriptor" error.  The switches operate just fine so not sre why I'm getting this error.  Will try re-pairing...

mmnpkf, 2018-10-02:
Sorry, I’ve managed to break it again. I’m not sure how. The code I posted above doesn’t actually work. I don’t know why it was working and now it doesn’t. I’ll continue fiddling and wait until I’ve got it working reliably for a while before declaring that it works.