Koenkk/zigbee2mqtt #429: does pairing through router actually work ?

lolorc, 2018-09-30:
using dev branch, commit https://github.com/Koenkk/zigbee2mqtt/commit/bf68f4575f591c5ef6ecb500acf6ec056c1eee97

In the process of repairing all my devices, it takes ages, pairing with cc2531/z stack is so difficult compared to zigate/JN5168

I started with my osram plugs and my cc2531 router, they all are few meters away from the coordinator.
I've been struggling to pair my xiaomi device when all the routers on, nearly impossible.
I've just unplugged them all and its looks it's slightly less difficult.

I've seen https://github.com/Koenkk/zigbee2mqtt/commit/71d12c3c711ea989b8b65d7005e09f0c0163cb1c
what's the current status ?

subzero79, 2018-09-30:
Doesn’t work

https://github.com/Koenkk/zigbee2mqtt/issues/331#issuecomment-424253806

lolorc, 2018-09-30:
;-) thanks missed the answer

lolorc, 2018-10-02:
renamed it as a feature request then :-)

closed 2018-12-08 by stale[bot]