Koenkk/zigbee2mqtt #430: Issue with RGB HUE Bulbs with FHEM

mfhotdog, 2018-09-30:
Hello all, 

I have a problem regarding the HUE RGB Lights with FHEM. The device is correctly paired and connected. I am able to setup the WEBCMD for the different white colortemperatures, dimmer and toggle on/off. But if I setup the colorpicker, the bulb turns once blue and thats it. I noticed that FHEM is sending the color in HEX like: 60FF26 (green) Is there still a way how zigbee can convert these values to the xy room? Or is someone using the same constellation with FHEM and has a walkaround for different RGB Values send by FHEM? 
Thank you. 
PS: What do you think to increase the wiki about the setup for common home automation software? I could send the settings for FHEM if you want.

Koenkk, 2018-10-01:
Feel free to create a new page called: Integrating with FHEM under https://github.com/Koenkk/zigbee2mqtt/wiki#customization.

Could you provide your debug log?

closed 2018-12-07 by stale[bot]

Evildime, 2018-12-29:
@mfhotdog @Koenkk 
I fixed this issue in PR https://github.com/Koenkk/zigbee-shepherd-converters/pull/180

This allows setting the color with the FHEM color picker.

FHEM setList attribute:
color:colorpicker,RGB zigbee2mqtt/[DEVICE_ID]/set {"$EVTPART0": {"hex":"$EVTPART1"}}