Koenkk/zigbee2mqtt #434: Cannot get the Node Descriptor of the device

kaelsaes, 2018-10-01:
I bought 2 new Aqara switches a few weeks ago. Pairing was no problem. But the payload of the MQTT message doesn't contain the click value.

During pairing process the message "Cannot get the Node Descriptor of the device" appears. It seems that the system doesn't know the new/old (?) Aqara switch.

It's a WXKG11LM

Thanks for your help.

Koenkk, 2018-10-01:
That means you haven't paired it correctly, please follow: https://github.com/Koenkk/zigbee2mqtt/wiki/Pairing-devices#xiaomi

phplego, 2019-03-28:
I got the same error (Cannot get the Node Descriptor of the device) when paring Trust Remote control (ZYCT-202).

The way I do it

> Factory reset the remote by holding the 0 (off) button for +-20 seconds. To establish a connection keep the remote within 2 meters from the hub. Press and hold the smart group button (button with two bulbs) and wait until the lights, below the channels, flash.

Other devices as

* WXKG02LM (Xiaomi Aqara double key)
* QBKG12LM (Xiaomi Aqara double key wired)
* WSDCGQ01LM (Xiaomi MiJia temperature & humidity sensor)

work fine.