Koenkk/zigbee2mqtt #436: Question regarding cache_state and sensor mqtt message

kimonm, 2018-10-01:
With configuration "cache_state: true," the mqtt message includes the last state event for sensors, even if not updated. (I understand why.) However, with the DJT11LM motion sensor, this means the last event (e.g., vibration) is sent when the sensor is just updating the battery status. I don't see how to identify that this isn't a new event.

Is the best practice to have fromZigbee.js entry define a state such as "idle" if the data field isn't set? Or is there a better way? I could change the code for DJT11LM to be something like:

if (msg.data.data['85']) {
const data = msg.data.data['85'];
result.event = vibrationLookup[data];
} else {
result.event = 'idle';
}

Koenkk, 2018-10-02:
The [`action`](https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/fromZigbee.js#L664) key is not cached due to https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/controller.js#L327

Could you provide your log?

kimonm, 2018-10-02:
Thanks for the explanation. I found the error in my code - it was an older version using event key rather than action. Btw, really excellent documentation. The work is appreciated.

closed 2018-10-02 by kimonm