Koenkk/zigbee2mqtt #437: Error at startup

kaelsaes, 2018-10-01:
Okt 01 21:34:37 Raspberry-MQTT npm[346]:   zigbee2mqtt:info 2018-10-1 21:34:37 Starting zigbee-shepherd
Okt 01 21:34:40 Raspberry-MQTT npm[346]:   zigbee2mqtt:info 2018-10-1 21:34:40 Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)

What does this error mean? After the 60 seconds zigbee-shepherd starts normal.

Koenkk, 2018-10-02:
This means your CC2531 is in a bad state. Could be due to a power outage or not correctly stopping zigbee2mqtt.

lolorc, 2018-10-02:
it also happen when the cc2531 has just been plugged, doesn't it ?
I think the firmware scans for existing networks when it start, so that I doesn't reuse an existing combination of channel/panid. (might be wrong)

Koenkk, 2018-10-02:
Yes, that seems to be the case.

kaelsaes, 2018-10-02:
@Koenkk 

What does this mean........."or not correctly stopping zigbee2mqtt" ?

Koenkk, 2018-10-02:
Not correctly stopped -> e.g. systeem restart by pulling the power.

Koenkk, 2018-10-02:
How do you run zigbee2mqtt?

kaelsaes, 2018-10-02:
It's running as a service. I have installed everything like explained in the wiki.

Koenkk, 2018-10-02:
Does it also appear when you do

```
sudo systemctl stop zigbee2mqtt
sudo systemctl start zigbee2mqtt
```

kaelsaes, 2018-10-02:
No.

Okt 02 18:49:34 Raspberry-MQTT systemd[1]: Stopping zigbee2mqtt...
Okt 02 18:49:34 Raspberry-MQTT systemd[1]: Stopped zigbee2mqtt.
Okt 02 18:49:48 Raspberry-MQTT systemd[1]: Started zigbee2mqtt.
Okt 02 18:49:49 Raspberry-MQTT npm[917]: > zigbee2mqtt@0.1.6 start /opt/zigbee2mqtt
Okt 02 18:49:49 Raspberry-MQTT npm[917]: > node index.js
Okt 02 18:49:50 Raspberry-MQTT npm[917]:   zigbee2mqtt:info 2018-10-2 18:49:50 Logging to directory: '/opt/zigbee2mqtt/data/log/2018-10-02.18-49-50'
Okt 02 18:49:51 Raspberry-MQTT npm[917]:   zigbee2mqtt:info 2018-10-2 18:49:51 Starting zigbee2mqtt version 0.1.6 (commit #5ed29d0)
Okt 02 18:49:51 Raspberry-MQTT npm[917]:   zigbee2mqtt:info 2018-10-2 18:49:51 Starting zigbee-shepherd
Okt 02 18:49:52 Raspberry-MQTT npm[917]:   zigbee2mqtt:info 2018-10-2 18:49:52 zigbee-shepherd started
Okt 02 18:49:52 Raspberry-MQTT npm[917]:   zigbee2mqtt:info 2018-10-2 18:49:52 Coordinator firmware version: '20180815'
Okt 02 18:49:52 Raspberry-MQTT npm[917]:   zigbee2mqtt:info 2018-10-2 18:49:52 Currently 25 devices are joined:

Koenkk, 2018-10-03:
Then somehow the process is killed on a reboot, I however don't know why.

kaelsaes, 2018-10-04:
Ok, thanks.

closed 2018-10-04 by kaelsaes