Koenkk/zigbee2mqtt #438: connect cc2531(coordinator) to cc2531(router or end-device)

leejaeuk, 2018-10-02:
Hello, @Koenkk 
Thank you for your effort. 
I already have raspberry pi 3B and i have plan to purchase three cc2531 usb dongle, xiaomi home kit.
I have several question about zigbee2mqtt service.

1.  if i don't use Home-Assistant integration, how can i see the device status or debug log?
( homeassistant: false - state. if i just go to 'mqtt://localhost', can i see the device state? )

2. how to connect cc2531 dongles? i watching the (https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices ) page so learn about other device connect. but i doubt about between cc2531 to cc2531 connection. If i just flashing to each cc2531 module coordinator and router, then router cc2531 device is recognized automatically?

ptvoinfo, 2018-10-02:
@leejaeuk You need to set `permit_join: true` in the zigbee2mqtt settings and a router will connect automatically after flashing. Just a note: a CC2531 router has a small coverage.

hoangtuit, 2018-10-02:
Hello @ptvoinfo 
Once router connect to coordinator, can i set permit_join: false?

ptvoinfo, 2018-10-02:
@hoangtuit Yes, of course. This is required to prevent attacks.

leejaeuk, 2018-10-02:
@ptvoinfo Thank you for fast response.

then i wondered about data transfer way.
from this URL https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices 
new zigbee cc2531 usb module also same process compare to other Xiaomi(adding new device)?

leejaeuk, 2018-10-02:
@ptvoinfo Oh...forgive my overlooking.

This is My goal = Using more than more cc2531 usb, i want to control cc2531 chip gpio pin.
If i can control cc2531 chip gpio, i connected gpio pin to relay.  So i want to make my own smart light switch. is it possible? (Communication way : USB or UART or SPI)

hub(RPi+CC2531) <----> light switch(Relay+cc2531)

In my opinion, cc2531 usb module needed end-device setting.. but there are no end-device firmware..
give me some advice.

Thank you!

leejaeuk, 2018-10-02:
@ptvoinfo additionally, 
i done successfully connected this way hub(RPi3B + Xbee s2c + usb module) <----->light switch (Xbee s2c+Relay)

kaelsaes, 2018-10-02:
@ptvoinfo 

I'm wondering if the range of the CC2531 with your new firmware is lower than with your old firmware!?!

All of my 3 CC2531 routers have problems pairing with each other and the coordinator.
When I place them near each other the pairing process works fine.

57k, 2018-10-02:
Just my personal status update ... maybe somebody is facing the same issues ... 

I tried to extend my zigbee mesh with further routers - but with the new firmware (UPDATE 2018-09-26 PTVO) I am not able to join any CC2531 or I also tried CC2530 - also the USB firmware for CC2531 did not work as it was recommended in another forum thread. 
Further new Xiaomi or tradfri sensors or lamps can join the network easily.
My CC2531 coordinator has still a firmware which I would say is about two-three month old ( I do not have the firmware number). If I create the network map via mqtt I only get around 10 devices (all routers plus some devices) showing up. All of them are connected to the coordinator (star). Nevertheless I receive sensor updates from routers which are one or two floors away - even from my garage which is next to my house. I am afraid if I put a new firmware to my coordinator that I have to reconnect all my +30 zigbee devices which I would like to avoid. 

Nevertheless I would like to thank you for your efforts within this project. 
Jens

ptvoinfo, 2018-10-02:
@leejaeuk A custom firmware is required for your usage scenario. Currently, you cannot control digital pins.
@kaelsaes The range is the same. Nothing changed. CC2531 has a small antenna and therefore CC2530 with an external antenna works much better.

ptvoinfo, 2018-10-02:
@57k My latest firmware for CC2531 with USB dumps information about found networks to a virtual COM port (while pairing). You may connect a terminal software to this COM port and check what is wrong.

leejaeuk, 2018-10-02:
@ptvoinfo 

How do you write your own firmware? This maybe a bit rude question. I'm sorry.
I also know the Z-stack.
I want to control end-device(CC2531 or another chipset) Gpio pin

DNCKATSW001 (from https://github.com/Koenkk/zigbee2mqtt/wiki/Supported-devices )
maybe become a solution about these problem?

lolorc, 2018-10-02:
from my experiments with the current firmwares, I've not been able to tell a cc2530-cc2591+antenna works better than the small cc2531. Which is very very weird.

tb-killa, 2018-10-03:
@ptvoinfo would you share your router firmware sources ?

kaelsaes, 2018-10-04:
I was not able to build up a network with the new router firmware. No idea why but the routers don't connect proper so the mesh network isn't working.

Went back to the "old" firmware and everything works fine again.

I'm using CC2531 for coordinator and 3 routers.

hoangtuit, 2018-10-09:
Hello @ptvoinfo 
How to disable led on routers? I use cc2531 router and the green led alway on.

ptvoinfo, 2018-10-09:
@hoangtuit Hi, just press the S1 button.

hoangtuit, 2018-10-09:
@ptvoinfo Thank you.

TurboTronix, 2019-03-21:
> @ptvoinfo additionally,
> i done successfully connected this way hub(RPi3B + Xbee s2c + usb module) <----->light switch (Xbee s2c+Relay)

Which Xbee s2c are you using?

CaptainNeil, 2021-01-15:
> @leejaeuk zigbee2mqtt already supports routers with my firmware (https://github.com/Koenkk/zigbee2mqtt/wiki/Supported-devices). A router starts the pairing process automatically after flashing. Note: CC2531 will be detected as CC2530.ROUTER. This is not a mistake.
Where can I get the CC2531 coordinator source code.