Koenkk/zigbee2mqtt #439: alternate configuration.yaml / mv configuration.yaml configuration.yaml.sample + gitignore

lolorc, 2018-10-02:
Hi

it's a bit of a hassle to upgrade zigbee2mqtt and think about recopying configuration.yaml back (which is versioned)
Especially when you know the panid will change when you happen to forget to copy back the conf.
This is the 2nd time in few days I have to repair my 30-ish devices because of my mistake, and it takes AGES to pair them :-(
This is PITA, see #318 

Would you agree to:
- add an option or an environnement VAR to specify the path of configuration.yaml
- git move data/configuration.yaml data/configuration.yaml.sample + data/configuration.yaml in .gitignore + update doc so users copy the sample to data/configuration.yaml the first time they start zigbee2mqtt

Koenkk, 2018-10-02:
The environment variable is already in place, you can set the data directory with `ZIGBEE2MQTT_DATA`.

lolorc, 2018-10-03:
oops my mistake should have seen that, enough for me, closing

closed 2018-10-03 by lolorc