Koenkk/zigbee2mqtt #440: CC2531 Alternative Firmware Flashing issue

Donderda, 2018-10-02:
Hi everybody,

I'm trying to flash my CC2531 Stick via the alternative flashing method. Currently, I'm trying to use an NodeMCU. I changed the Sketch according to my wiring to this:

```c
// Pinout configuration (Configured for Arduino Leonardo)
int CC_RST   = D1;
int CC_DC    = D2;
int CC_DD_I  = D6;
int CC_DD_O  = D5;
```

Due to the fact that the NodeMCU is running on 3.3V, I directly connected the pins with my CC2531. But I can't flash the device :(. I just get the message: `ERROR: The chip is not responding. Check your connection and/or wiring!`. Do I have to press any button to make this work or am I doing something wrong?

Donderda, 2018-10-02:
Has anyone knowledge using the CCLoader sketch? Or has anyone a .bin-version of the needed firmware?

greeny2612, 2018-10-03:
I used a Wemos D1 successfully with the following pins.
int CC_RST   = D8;
int CC_DC    = D7;
int CC_DD_I  = D6;
int CC_DD_O  = D5; 

I think it's the default pin state that causes issues.

Also change the setup baud rate in the setup.

 // Initialize serial port
  Serial.begin(9600);

Donderda, 2018-10-07:
I just flashed my CC2531 with a CC-Debugger. Thank you for your time. For me the alternative flashing method didn't work out :(

closed 2018-10-07 by Donderda