Koenkk/zigbee2mqtt #441: Xiaomi 'mains' devices send 'alive' information. Currently all hardcoded to 'online'.

BenNeumeister, 2018-10-02:
Not a bug per-say, more feature correction, but it seems that all of the Xiaomi mains powered devices send a timed CID which could be used for 'online' monitoring.

In the code you have it hardcoded to not look for 'awake' frames from any Xiaomi device.

> //Set all Xiaomi devices (manufId === 4151) to be online, so shepherd won't try
        // to query info from devices (which would fail because they go tosleep).
        // Xiaomi lumi.plug has manufId === 4447 and can be in the sleep mode too
        const devices = this.getAllClients();
        devices.forEach((d) => {
            if ((d.manufId === 4151) || (d.manufId === 4447)) {
                const device = this.shepherd.find(d.ieeeAddr, 1);
                if (device) {
                    device.getDevice().update({
                        status: 'online',
                        joinTime: Math.floor(Date.now() / 1000),
                    });
                }
            }
        }); 

For example I have a QBKG03LM and ZNCZ02LM, and both send an 'online' messages every 6-8mins

Frame looks to be the same 'base' spec for all items.
I.e. GenBasic CID with AttID '65281' with an array item included pf varying size.
If you receive this CID (aswell as the 'trigger' ones) within say a 10minute window, mark the device as online. Otherwise it's offline after 10minute counter has elapsed.

`zigbee2mqtt:debug 2018-10-2 20:57:27 Recieved zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"65281":{"3":27,"5":8,"7":"0x0000000000000000","8":4886,"9":512,"100":1,"149":0.42732903361320496,"150":2304,"152":0.3319999873638153,"154":16}}}' of device 'lumi.plug' (0x00158d000xxxxxxxx)`

`zigbee2mqtt:debug 2018-10-2 20:58:04 Recieved zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"65281":{"3":21,"5":11,"7":"0x0000000000000000","8":4383,"9":512,"100":0,"101":0,"149":0.0008930352632887661,"152":0,"154":16}}}' of device 'lumi.ctrl_ln2.aq1' (0x00158d000xxxxxx)`

Would be cool if you could add in checks for the mains devices, probably via ID, so they could be tracked.
P.s. Looks like **Received** in the debug message has a spelling typo :-)

Let me know if you need any more info!

Thanks!
Ben

Koenkk, 2018-10-03:
hi ben! thansk for reporting, could you make a PR for this?

BenNeumeister, 2018-10-03:
Guess I could. Haven't looked at your code too much in detail, but suppose can have a look.

Koenkk, 2018-10-03:
Ok thanks! Be sure to fork from the dev branch and not the master branch.

ryanbeaton, 2018-10-04:
@ptvoinfo did “Added polling for routers (prevents deep sleep mode of Xiaomi routers)” in https://github.com/Koenkk/zigbee2mqtt/pull/50 so the 4447 could be removed I suggest.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

ryanbeaton, 2018-10-04:
Hmmm pull 50 introduced the 4447 into the if statement. Weird!

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

BenNeumeister, 2018-10-04:
For info, mine's still going strong after a week being plugged in and not powered.

`  zigbee2mqtt:debug 2018-10-4 21:58:39 Recieved zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"65281":{"3":23,"5":8,"7":"0x0000000000000000","8":4886,"9":256,"100":0,"149":0.4279957413673401,"150":2304,"152":0,"154":16}}}' of device 'lumi.plug' (0x00158d000xxxxxx)`

So yeah, I'd say that's the easiest way and use the reports from the sensor. Again, you'd need to limit this to only 'powered' devices, else if you simply don't report attReport (i.e. battery powered sensor) it just assumes your online again.

You could check the 'powerSource' attribute, i.e.
`"powerSource":"Battery"` vs `"powerSource":"Mains (single phase)"`

closed 2018-12-10 by stale[bot]