Koenkk/zigbee2mqtt #442: Esp8266 as zigbee coordinator

lokomotionn, 2018-10-03:
Hello, it is possible to use the serial port in esp8266 to connect cc2531 and use it as a coordinator to send data to the mqtt gate. I ask because the server (Raspberry) I have in the basement and I would like the coordinator to be on the floor to get a better range. It would be a nice solution because I have wifi access in the whole house.

Koenkk, 2018-10-03:
No, but you could you use another raspberry pi instead of the esp8266.

kirovilya, 2018-10-03:
@lokomotionn I think it is possible 😉 I was doing an experiment yesterday...
[image]
The main idea is to make a virtual serialport that works on tcp with esp8266.
How will progress - I will write here. I also dream that the coordinator was separate from the server.

further, you can also transfer processing logic to esp8266 later.

BenNeumeister, 2018-10-03:
Why not just setup a router on your ground floor as act as the 'center' and then relay the info to the co-ordinator on the pi?

kirovilya, 2018-10-03:
Success! https://www.youtube.com/watch?v=HI4AkD0QGgM

kirovilya, 2018-10-03:
esp8266 (wemos d1 mini) flashing with Esp-link https://github.com/jeelabs/esp-link or Ser2Net from EspEasy https://github.com/letscontrolit/ESPEasy

Wiring:
ESP8266  -- CC2530
----------------------
3v3  -- VCC
GND -- GND
TX -- P02
RX -- P03
GND -- P20
GND -- P04
GND -- P05

For windows virtual serialport https://www.hw-group.com/software/hw-vsp3-virtual-serial-port
For linux http://tibbo.com/support/downloads/vspdl.html (but I am not test)

JLeandroFreitas, 2018-10-04:
Ser2net so i think this must be possible.
I think this very nice i vote for adding function for 
configuration.yaml.
serial:
  host: ipaddress of the ESP8266 
  port: TCP port you added in the devices tab of the ESP8266

Press the Flashmode button when connection to the computer

Downlad the lastest https://github.com/letscontrolit/ESPEasy/releases
Extract the ESPMega zip file, and open ‘FlashESP8266.exe’
Select the right COM port and select the right Firmware (ESP_Easy_mega-XXXXXXXX_normal_ESP8266_4096.bin)

When it is ready close the flashing tool
The ESP8266 will now emit a wifi signal, connect with it with the following password: ‘configesp’ (more information at https://www.letscontrolit.com/wiki/index.php/ESPEasy#Introduction)
After connection a screen opens in which you can let the ESP8266 connect to your WIFI, if succesfull its new IP address will be shown.
Go to this IP address in your browser and click on devices
or Go to http://192.168.4.1/setup
Click on "Devices" Edit of the first task and select ‘Communication - Serial Server’ from the dropdown list
Fill in the form as following:
a.	Name: ZIGBEE2MQTT
b.	Enabled: checked
c.	TCP Port: a number between 1000 and 9999 "1775"
d.	Baud Rate: 115200
e.	Data bits: 8
f.	Parity: No Parity
g.	Stop bits: 1
h.	Reset target after boot: - none –
i.	RX receive timeout: 0
j.	Event processing: Generic
Press Submit

Then its complete the device will get devicename ESP-Easy-0 check your router for the IP or you will get directed after the setup of the Wifi the first time you connected to the device.

You can also try to use ser2net on Linux then use that port.
https://www.letscontrolit.com/wiki/index.php/Ser2Net

JLeandroFreitas, 2018-10-18:
> Ser2net has a virtual port client?
> How to configure the configuration.yaml file for a virtual serial port?
> 
> homeassistant: false
> permit_join: true
> mqtt:
>    base_topic: zigbee2mqtt
>    server: 'mqtt: // localhost'
> series:
>    **port: / dev / ttyUSB1**

Try this http://techtinkering.com/2013/04/02/connecting-to-a-remote-serial-port-over-tcpip/

oriolism, 2018-11-15:
@lokomotionn Do you have this setup working? If so, its stable? Any downside?

JLFN, 2018-12-10:
@lokomotionn  and @oriolism check this out https://github.com/Koenkk/zigbee2mqtt/issues/665#issuecomment-445722057

randybb, 2019-01-10:
Probably would be nicer to do the whole zigbee to mqtt translation on esp8266. It doesn't make sense to use it just as a wireless (virtual) serial port.

JLFN, 2019-01-11:
> Probably would be nicer to do the whole zigbee to mqtt translation on esp8266. It doesn't make sense to use it just as a wireless (virtual) serial port.

@randybb  You probably should try to make it your self if believe its possible with the low resource of esp8266.

JLFN, 2019-02-01:
Solved :)

JLFN, 2019-02-06:
close this see docs!

Koenkk, 2019-02-06:
https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_esp8266_with_cc2530.html thanks to @JLFN

closed 2019-02-06 by Koenkk

enomam, 2019-02-12:
Quick question about the IP used in the `socat` command - is it the ip of the ESP8266? I'm assuming so, because any other ip doesn't seem to make sense in this use case.

Koenkk, 2019-02-12:
@enomam yes

2016for, 2019-03-28:
Wow @kirovilya @JLFN @Koenkk. Was in need for something exactly like this!

This basically should act as multiple xiaomi hubs and can be placed anywhere in the house.

How has been your experience with this? How stable are they?
Any interference b/w zigbee of cc2530 and wifi of esp8266?

How's the lag compared to having a router cc2531/0 instead?

Thanks in anticipation

JLFN, 2019-03-28:
@2016for i get lag when i controll 3 lamps the same time but i think controlling those as group will maybe solve that. Else its fast response.

2016for, 2019-03-28:
@JLFN Thanks for the reply mate. Are you using any motion/door sensors?
My current Xiaomi hub setup gives me instantaneous feedback/status of these sensors, really want to be able to replicate that.

JunShine, 2019-09-27:
Can I replace ESPEasy with ESPHome(https://github.com/esphome/esphome)? How to fill in the ESPHome configuration entry？

hdo, 2019-10-25:
Unfortunately this does not work with Docker :-(

As i understand Docker can't deal with symlinks.

h4nc, 2019-11-23:
@koenkk is there a reason why the tutorial isn't online any more?

https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_esp8266_with_cc2530.html

Koenkk, 2019-11-23:
It has been moved to https://www.zigbee2mqtt.io/information/connecting_cc2530.html

tlc76, 2020-02-26:
Hello,

Meanwhile, the serial server configuration interface has changed. A new field called "TX Enable Pin" was added.
The values in the drop down list are listed in the second screenshot.

Can you please advise how to configure this setting?

Thanks

[image]
[image]

tlc76, 2020-02-27:
And second question: is the configuration mentioned in [this post](https://github.com/Koenkk/zigbee2mqtt/issues/442#issuecomment-426994483), applicable also to the CC2530+CC2591 version? Or only to the standalone CC2530?

codmpm, 2020-09-12:
Is there an interest for building a decent hardware solution (open-source)?
Just sprung to my head, that I have everything on hand needed to built this.

gtkow, 2020-11-30:
Hello Everybody, 

I have connected an esp8266 (wemos d1) to a CC2530 + CC2591. I have followed this instruction : https://www.zigbee2mqtt.io/information/connecting_cc2530.html But it doesn't work... When I launch zigbee2mqtt I have this message : 

`Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)`

When I connect my CC2530 + CC2591 to my Pi0 with these instruction  : http://www.handverdrahtet.org/2019/12/setting-up-zigbee2mqtt-gateway.html it's working and I have this message :

`MQTT publish: topic 'zigbee2mqtt2/bridge/state', payload 'online'
Zigbee2MQTT:info  2020-11-30 15:54:39: MQTT publish: topic 'zigbee2mqtt2/bridge/config', payload '{"commit":"cb9699e","coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"},"log_level":"info","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6757},"permit_join":true,"version":"1.16.1"}'`

I think the problem is the espeasy but I have followed instruction... How I can debug this?
Does someone have an idea?

Thank's for your help

Farfar, 2021-05-11:
> I have connected an esp8266 (wemos d1) to a CC2530 + CC2591. I have followed this instruction : https://www.zigbee2mqtt.io/information/connecting_cc2530.html But it doesn't work.

I am trying the same setup. Did you make it work?

Farfar, 2021-05-12:
At first I didn't succeed but after a few retries I got my Wemos D1 working as a serial to wifi bridge and zigbee2mqtt up and running. I must have done something wrong first try.
Followed instructions in documentation to the letter and all seems to be working fine.