Koenkk/zigbee2mqtt #445: Bitron smoke detector

kaelsaes, 2018-10-04:
I would like to use my Bitron smoke detector with zigbee2mqtt. Is there a converter available?

https://www.conrad.de/de/bitron-video-funk-rauchmelder-av201024a-90201024a-1507095.html?WT.mc_id=google_pla&WT.srch=1&ef_id=W7WurwAAAJ9VJ3Id:20181004112653:s&gclid=EAIaIQobChMIuI6Kq9bs3QIVE8ayCh380ggNEAQYBCABEgLDqfD_BwE&hk=SEM&insert_kz=VQ&s_kwcid=AL!222!3!254339639498!!!g!!

tb-killa, 2018-10-04:
Hi i´m use some Bitron Devices => [The big "Bitron" Thread](https://github.com/Koenkk/zigbee2mqtt/issues/348).
Nice to see that we could implement another type of device.

Actually there is no converter available, but we could work on them and implement the needing stuff.

If you in touch with one of this device and in touch with your zigbee2mqtt instance via ssh or another tool to manipulate files too .. lets start and pair the device.

after that i need your ``database.db`` content and maybe we could start on building converter code :)

closed 2018-10-17 by kaelsaes

Mouky66, 2019-01-14:
Any updates on this? I am also looking for a converter.
Thank you :)