Koenkk/zigbee2mqtt #446: Can't pair any additional device

pepp86, 2018-10-04:
Hi Team,

I am running the latest zigbee2mqtt release and the latest dev branch. I've got some Osram+ Plugs working fine, but not able to connect new Plugs or light bulbs.

Pairing seems to be fine, but the server is not sending anything to mqtt about the new devices. Any Ideas?

Log: https://pastebin.com/mW1M06h5

ben423423n32j14e, 2018-10-04:
Did you read the wiki?

https://github.com/Koenkk/zigbee2mqtt/wiki/ZigBee-network

pepp86, 2018-10-05:
Yes of course. 

Actually my network consists of 5 Osram plugs which are routers and one Philips hue motion sensor. 

Far away from the Zigbee network limit.

pepp86, 2018-10-05:
Ok. Found a solution. zigbee2mqtt is not creating the inital topic on the mqtt server.
After creating the topic with a mqtt client, there is date in it.

Any idea how to solve it? Any input required?

lolorc, 2018-10-05:
this initial topic issue sounds weird to me.
But I have had issues pairing devices surrounded by my osram plugs acting as router. Had to pair very close to the coordinator or shut the routers down.

closed 2018-12-11 by stale[bot]