Koenkk/zigbee2mqtt #447: Not connected to MQTT server!

spysky, 2018-10-04:
Hi,

I've followed all the instructions, but when i try to start, it fails to connect to mqtt server.
I've tested the mqtt server is up and running (homeassistant is using it, and I've used an external test tool on the same machine as the zigbee2mqtt to test the connection.)

Log output:
https://pastebin.com/gy4nQ72R

The configuration.yaml is the default one, I've just uncomented the user and password as my mqtt server  (mosquitto) has authentication enabled.

AnhDuc85, 2018-10-04:
what's your environment?

spysky, 2018-10-04:
I'm running linux mint
I've already tested with the IP, but got the same issue.

kimonm, 2018-10-05:
Are you including the port in the server specification? E.g., 
server: 'mqtt://localhost:1883'

spysky, 2018-10-05:
Still the same.
Tried it using: 'mqtt://localhost:1883' , 'mqtt://127.0.0.1:1883' ( I know it's the same, but at this point it clouldn't harm to check), and using my IP.

spysky, 2018-10-07:
> Basic questions:
> 
> * Which mqtt broker?
> * got auth at your broker?
> * does  [mosquitto_pub -h localhost -t test -m "hello world"] work? (You maybe have to add user and auth)
> 
> I would bet my money on it that you're using the built in broker from Home Assistant and can't publish messages therefore.
> But only a guess.

Which mqtt broker?
-> Eclipse Mosquitto

got auth at your broker?
-> Yes, As I said in the #1 post, the only config I changed in configuration.yaml was to set the user & password.

does [mosquitto_pub -h localhost -t test -m "hello world"] work? (You maybe have to add user and auth)
-> That command outputs an error due to auth. 
However this one with authentication works like a charm [mosquitto_pub -h localhost -u username -P password -t test -m "hello world"]

Thanks

spysky, 2018-10-09:
OK, so I made some tests, and seems like the issue is related to authentication.
I created a new environment and installed Mosquitto on it.
Tried to connect the zigbee2mqtt to it and voila,  it connected without any issues.

After this, I enabled authentication and again tried to connect.
This time i got the same error, "Not connected to MQTT server!"

Any new ideas on how to approach this (without having to disable authentication)?

Thanks

bk8190, 2019-01-18:
I fixed it by specifying a user/password.  Weird because my MQTT server doesn't require one from other applications.

thanhdotien278, 2019-07-05:
> Well this was kind of a dumb thing, but at least I fond the issue.
> 
> In case anyone has the same problem, username and password must be quoted, example:
> 
> mqtt:
> base_topic: zigbee2mqtt
> server: 'mqtt://localhost:1883'
> user: 'user'
> password: 'password'
> reject_unauthorized: false
> serial:
> port: /dev/ttyACM0
> 
> advanced:
> log_level: debug

You just saved my day!

Loook11, 2020-02-17:
By default docker image container works as "bridge" network type, him need to change as "host" type, then can connect with parameters (configuration.yaml):
...
mqtt:
base_topic: zigbee2mqtt
server: 'mqtt://127.0.0.1:1883'
.....
or if used as "bridge" type need use correct mqtt server ip address in your network.

Have a good day!